markusressel/fan2go #9: detect throws error

Brianpickfordd, 2021-02-27:
Hi, thanks for sharing. I've not used go before, I think I have the dependencies installed but I suspect that I'm missing somthing ovbious. I'm running Arch with Gnome and get this error:
`$ fan2go detect
panic: runtime error: makeslice: len out of range

goroutine 1 [running]:
github.com/asecurityteam/rolling.NewWindow(...)
	/go/pkg/mod/github.com/asecurityteam/rolling@v2.0.4+incompatible/window.go:11
github.com/markusressel/fan2go/internal.createFans(0xc0000b2600, 0x2e, 0x2e, 0x0, 0x0)
	/github/workspace/internal/backend.go:489 +0x25d
github.com/markusressel/fan2go/internal.FindControllers(0x8c1d00, 0xc0000a1860, 0x9571ab, 0x7, 0xc0000a3a90)
	/github/workspace/internal/backend.go:451 +0x293
github.com/markusressel/fan2go/cmd.glob..func2(0xd2fd60, 0xd55270, 0x0, 0x0)
	/github/workspace/cmd/root.go:37 +0x37
github.com/spf13/cobra.(*Command).execute(0xd2fd60, 0xd55270, 0x0, 0x0, 0xd2fd60, 0xd55270)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:854 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xd2fac0, 0xd55088, 0x9571ab, 0x7)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:958 +0x349
github.com/spf13/cobra.(*Command).Execute(...)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:895
github.com/markusressel/fan2go/cmd.Execute()
	/github/workspace/cmd/root.go:145 +0x27f
main.main()
	/github/workspace/main.go:25 +0x20
`

markusressel, 2021-02-28:
With go you don't need to install any dependencies. I get the same error on my end, so this is "just" a bug.

markusressel, 2021-02-28:
Should be fixed in 0.0.14

closed 2021-02-28 by markusressel

Brianpickfordd, 2021-02-28:
Working well now, thanks very much. This seems to be a much better approach than fancontroll