markusressel/fan2go #15: Make an AUR package?

odcpw, 2021-06-02:
Hi, 

after struggling setting up fans on new Thinkpad T14 Gen to with a new (and newbie) Arch installation (hwmon index changing on reboot for thinkfan as well as fancontrol), I am very happy to have found fan2go!

Could you continue using your skills to make an AUR package and proposing it to the Arch Wiki?

markusressel, 2021-06-02:
Glad fan2go works for you 👍 

I definitely thought about creating an AUR package. However, I am not sure I have the time to maintain it in an efficient way. If someone can provide a sample for using a Github Action to release an AUR package in a fully automated way I would certainly think about it again 😄 I would also be happy to see someone else creating a AUR package off of this repo and maintain it independently.

manvari, 2021-11-04:
I created an AUR package and will try to maintain it.

https://aur.archlinux.org/packages/fan2go-git/

salvoc81, 2021-11-17:
Hello @manvari, thanks a lot for trying to build the AUR package.  
Sadly it could not be installed on my system.  
```
Preparing...
Checking fan2go-git dependencies...
Resolving dependencies...
Checking inter-conflicts...
Download of go (2:1.17.3-1) started
Download of go (2:1.17.3-1) finished
Checking keyring...
Checking integrity...
Loading package files...
Checking file conflicts...
Checking available disc space...
Installing go (2:1.17.3-1)...
Cloning fan2go-git build files...
Generating fan2go-git information...

Building fan2go-git...
==> Making package: fan2go-git 0.0.18.r6.g0719fb2-2 (2021年11月17日 15時55分48秒)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning fan2go git repo...
Cloning into bare repository '/var/tmp/pamac-build-salvocos/fan2go-git/fan2go'...
  -> Found fan2go.service
==> Validating source files with sha256sums...
    fan2go ... Skipped
    fan2go.service ... FAILED
==> ERROR: One or more files did not pass the validity check!
Failed to build fan2go-git
```
Also thanks to @markusressel  for creating this very nice tool.  
Will try to compile it for myself...

manvari, 2021-11-17:
Hi Salvatore, thank you for letting me know about the problem. I updated the systemd unit file but forgot to change its SHA256 sum in PKGBUILD. I will push the fix ASAP.

markusressel, 2021-11-17:
Also: There is no need to compile it for youself. Release binaries can be found on their release page of this repo.

markusressel, 2021-11-30:
I just released `0.1.0` which contains a major break to the configuration layout.
Please upgrade your AUR packages with caution.

markusressel, 2021-12-03:
@manvari seems like the hash sum problem still exists (or came up again?).
Could you have another look?

olielvewen, 2022-01-08:
@markusressel 
I confirm this issue that I have had today trying fan2go than I've just discovered reading the wiki arch

I have passed this issue editing the PKBUILD file and after marking the sum control (the second ) to SKIP.
Now, I'm trying to understand how to fan2go works.
I have been able to run the fan2go service but that's all for the moment.

`  FATAL   Error reading config file, Config File "fan2go" Not Found in "[/home/manjaro /etc/fan2go]"
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
        github.com/pterm/pterm@v0.12.33/prefix_printer.go:340
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0x56524a5a0dc0, {0x56524a096371, 0x1d}, {0xc0000dbd60, 0x1, 0x1})
        github.com/pterm/pterm@v0.12.33/prefix_printer.go:282 +0x1d5
github.com/markusressel/fan2go/internal/ui.Fatal(...)
        github.com/markusressel/fan2go/internal/ui/logging.go:36
github.com/markusressel/fan2go/internal/configuration.ReadConfigFile()
        github.com/markusressel/fan2go/internal/configuration/config.go:79 +0x65
github.com/markusressel/fan2go/cmd.glob..func3(0x56524a5a3b80, {0x56524a088886, 0x0, 0x0})
        github.com/markusressel/fan2go/cmd/root.go:33 +0x65
github.com/spf13/cobra.(*Command).execute(0x56524a5a3b80, {0xc000020230, 0x0, 0x0})
        github.com/spf13/cobra@v1.3.0/command.go:860 +0x5f8
github.com/spf13/cobra.(*Command).ExecuteC(0x56524a5a3b80)
        github.com/spf13/cobra@v1.3.0/command.go:974 +0x3bc
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.3.0/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
        github.com/markusressel/fan2go/cmd/root.go:78 +0x115
main.main()
        ./main.go:25 +0x17
    ~  systemclt status fan2go                                                                      2 ✘ 
zsh: correct 'systemclt' to 'systemctl' [nyae]? y
○ fan2go.service - Advanced Fan Control program
     Loaded: loaded (/usr/lib/systemd/system/fan2go.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
    ~  systemctl start fan2go                                                                       3 ✘ 
    ~  systemctl status fan2go                                                                        ✔ 
● fan2go.service - Advanced Fan Control program
     Loaded: loaded (/usr/lib/systemd/system/fan2go.service; disabled; vendor preset: disabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2022-01-08 17:43:06 UTC; 354ms ago
    Process: 71756 ExecStart=/usr/bin/fan2go -c /etc/fan2go/fan2go.yaml --no-style (code=exited, status=2)
   Main PID: 71756 (code=exited, status=2)
        CPU: 8ms
`

markusressel, 2022-01-09:
@olielvewen Please follow the instructions on the front page of this repository (the `README.md`). The service won't do anything without you creating a configuration first. If you have a suggestion on how to improve the instructions in the `README`, feel free to open a new issue or better yet a PR with your suggested changes.

olielvewen, 2022-01-09:
@markusressel I have seen that...... after. I will open a PR/new issue for so you can validate my config file for my gpu.

Gelmo, 2022-01-09:
I've just submitted an Orphan request for the git package at AUR. @markusressel three weeks have passed since your comment there, so it may be time to reach out to an admin.

markusressel, 2022-01-10:
I already submitted a deletion request, but it was rejected:

```
please file an orphan request (+ giving the actual maintainer the time
to acknowledge) or ask the maintainer to add you as package co-
maintainer
```

Since you already submitted an orphan request, and I already asked here and in the AUR comments for him to add me as a co-maintainer, I guess we have to wait some more time...

markusressel, 2022-02-03:
The package has now been disowned. Not sure how to proceed from here...

Gelmo, 2022-02-05:
@markusressel I would suggest canceling the deletion request and adopting the package instead. You can adopt it with the "Adopt Package" link in the right-side menu

markusressel, 2022-02-05:
@Gelmo Thx for the tip! I already looked for this exact button, but couldn't find it, now I did! :laughing: 

I have taken over the ownership and updated the package. Works for me!
Let me know in the AUR comment section if you have any issues.

Also, I am not sure if I can cancel the deletion request (I wouldn't know how?), but maybe the reader will be smart enough to figure out that I already found a better solution to the problem than deletion of the package.

I will close this issue now.
If you have any issues, again, please use the AUR comment section. Thx!

closed 2022-02-05 by markusressel

olielvewen, 2022-02-05:
@markusressel 

Great news. You should ask in the Aur forum [section Aur Issues....,](https://bbs.archlinux.org/viewforum.php?id=38) you'll find quickly somebody to help you I think.