markusressel/fan2go #20: panic: runtime error: invalid memory address or nil pointer dereference

levecml, 2021-07-26:
First of all, Dankeschön, for all the hard work 

Running on Linux Mint 20.2 (basically a Ubuntu 21.04)
Board is a quite old Intel DQ61-EP 
lm-sensors is already the newest version (1:3.6.0-2ubuntu1).

```
master@tenshi ~ $ fan2go detect
Detected Devices:
amdgpu
  pwm1 (1): RPM: 998 PWM: 61 Auto: false
  temp1_input (1): 45000
coretemp
  temp1_input (1): 50000
  temp2_input (2): 54000
  temp3_input (3): 48000
  temp4_input (4): 47000
  temp5_input (5): 47000
nct6775
  pwm1 (1): RPM: 0 PWM: 255 Auto: false
  pwm2 (2): RPM: 153 PWM: 12 Auto: false
  pwm3 (3): RPM: 958 PWM: 102 Auto: false
  temp1_input (1): 44000
  temp2_input (2): 48000
  temp3_input (3): 50000
  temp6_input (6): 72000
  temp8_input (8): 0
master@tenshi ~ $ 
```
setup and initial run for the very first Time
```
master@tenshi ~ $ sudo fan2go
2021/07/26 20:58:14 Ignoring unconfigured sensor amdgpu/temp1_input
2021/07/26 20:58:14 Ignoring unconfigured sensor coretemp/temp2_input
2021/07/26 20:58:14 Ignoring unconfigured sensor coretemp/temp3_input
2021/07/26 20:58:14 Ignoring unconfigured sensor coretemp/temp4_input
2021/07/26 20:58:14 Ignoring unconfigured sensor coretemp/temp5_input
2021/07/26 20:58:14 Ignoring unconfigured sensor nct6775/temp1_input
2021/07/26 20:58:14 Ignoring unconfigured sensor nct6775/temp2_input
2021/07/26 20:58:14 Ignoring unconfigured sensor nct6775/temp6_input
2021/07/26 20:58:14 Ignoring unconfigured sensor nct6775/temp8_input
2021/07/26 20:58:14 Ignoring unconfigured fan amdgpu/pwm1
2021/07/26 20:58:14 Ignoring unconfigured fan nct6775/pwm1
2021/07/26 20:58:14 Ignoring unconfigured fan nct6775/pwm3
2021/07/26 20:58:14 Gathering data...
2021/07/26 20:58:17 Loading fan curve data for fan 'cpu_cooler'...
2021/07/26 20:58:17 No fan curve data found for fan 'cpu_cooler', starting initialization sequence...
2021/07/26 20:58:22 Measuring RPM of cpu_cooler (pwm2) at PWM: 0
2021/07/26 20:58:24 Measuring RPM of cpu_cooler (pwm2) at PWM: 1
2021/07/26 20:58:26 Measuring RPM of cpu_cooler (pwm2) at PWM: 2
2021/07/26 20:58:28 Measuring RPM of cpu_cooler (pwm2) at PWM: 3
....
2021/07/26 21:06:50 Measuring RPM of cpu_cooler (pwm2) at PWM: 254
2021/07/26 21:06:50 Start PWM of cpu_cooler (pwm2): 0
2021/07/26 21:06:50 Max PWM of cpu_cooler (pwm2): 235
2021/07/26 21:06:50 Starting controller loop for fan 'cpu_cooler'
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x55e6c0]

goroutine 21 [running]:
github.com/markusressel/fan2go/internal.calculateTargetSpeed(...)
	/github/workspace/internal/backend.go:411
github.com/markusressel/fan2go/internal.setOptimalFanSpeed(0xc00010be30, 0xc0000e3e48, 0x2)
	/github/workspace/internal/backend.go:404 +0x60
github.com/markusressel/fan2go/internal.fanController(0xa135e0, 0xc000069e00, 0xc0000dac00, 0xc00010be30, 0xc00041e300, 0x0, 0x0)
	/github/workspace/internal/backend.go:318 +0x23c
github.com/markusressel/fan2go/internal.Run.func5(0x0, 0x0)
	/github/workspace/internal/backend.go:99 +0xc9
github.com/oklog/run.(*Group).Run.func1(0xc0003ea000, 0xc0003a82a0, 0xc000113140)
	/go/pkg/mod/github.com/oklog/run@v1.1.0/group.go:38 +0x27
created by github.com/oklog/run.(*Group).Run
	/go/pkg/mod/github.com/oklog/run@v1.1.0/group.go:37 +0xbb
master@tenshi ~ $
master@tenshi ~ $
```

yaml:
```
# The path of the database file.
dbPath: "/etc/fan2go/fan2go.db"
# The rate to poll temperature sensors at.
tempSensorPollingRate: 200ms
# The number of sensor items to keep in a rolling window array.
rollingWindowSize: 50
# The rate to poll fan RPM input sensors at.
rpmPollingRate: 1s
# The rate to update fan speed targets at.
controllerAdjustmentTickRate: 200ms

# A list of sensors to monitor.
sensors:
  # A user defined ID, which is used to reference a
  # a sensor in a fan configuration (see below)
  - id: cpu_package
    # The controller platform as displayed by `fan2go detect`, f.ex.:
    # "nouveau", "coretemp" or "it8620" etc.
    platform: coretemp
    # The index of this sensor as displayed by `fan2go detect`.
    index: 1
    # The minimum target temp for this sensor.
    # If the sensor falls below this value, all fans referencing it
    # will run at minimum PWM value.
    min: 40
    # The maximum target temp for this sensor.
    # If the sensor is above this value, all fans referencing it
    # will run at maximum PWM value.
    max: 75

  - id: mainboard
    platform: nct6775
    index: 3
    min: 40
    max: 60

# A list of fans to control.
fans:
  # An user defined ID.
  # Used for logging only.
  - id: cpu_cooler
    # The platform of the controller which is
    # connected to this fan (see sensor.platform above).
    platform: nct6775
    # The index of this fan as displayed by `fan2go detect`.
    fan: 2
    # Indicates whether this fan is allowed to fully stop.
    neverStop: yes
    # The sensor ID (defined above) that should be used to determine the
    # speed of this fan.
    sensor: coretemp
```
I cannot see where I am wrong ...

markusressel, 2021-07-26:
The config looks fine to me too.
Does this also happen after the fan curve has been detected?

levecml, 2021-07-27:
Markus, you made my day, you are absolutely right. I actually chose the the sensor "chip class", not the id - my fault. It does not crash anymore and instead seems to run smoothly. Thanks a lot.

closed 2021-07-27 by levecml