markusressel/fan2go #28: Increase spindown time on inital measurement

Schlaefer, 2021-09-25:
If without active fan-control a fan spins at full speed (BIOS emergency mode) it may take a considerable time to spin down to the minimum speed (or stop). The current time window of 3 seconds is to small for that to happen.

The initial waiting time should be increased to at least 30 seconds, up to a minute would be preferable.

markusressel, 2021-09-26:
Thx for the suggestion.

I also noticed the spindown taking a different amount of time depending on the hardware involved.
The main reason I didn't choose a much bigger delay for the initial spindown was to avoid the system from running a long duration without any fan activity, since this could potentially be dangerous for system components.

What we could do:
* add a configuration option for the initial spindown delay (but still default to 3 seconds)
* avoid measuring all fans at the same time to retain at least _some_ airflow within the system during measurements

Schlaefer, 2021-09-26:
Thanks for the answer. Maybe I'll look into it myself, this seems like a fun project to learn some Go.

Is there a particular reason to start the measurement with zero instead with full speed? Starting at full speed would allow for a waiting period no matter the current speed without any temperature harm.

markusressel, 2021-09-26:
Yes. 

For most fans there is a difference between
* the min pwm at which the fan starts to rotate from a stand still
* and the max pwm at which the fan still rotates when it was rotating previously
where the latter is smaller than the former in most cases.

To make the "never stop" option work, we need to know the min pwm at which the fan will still spin 100% of the time, independent of whether it was spinning before or not. Approaching it from the bottom is more reliable for this (although also not perfect, I think).

The system admin should take precautions for the measurement phase anyway. Its impossible to fully measure the fans without turning them off, we can just try to minimize the risk.

markusressel, 2021-09-26:
The `minPWM` vs `startPWM` currently is not a concept in fan2go, but I also thought about integrating it at some point.

markusressel, 2021-09-28:
@Schlaefer I have worked on this a bit in #32, do you think the algorithm I added will solve your problem?

Schlaefer, 2021-09-29:
Looks good overall. Here are my measurements with a desktop-PC and two headers (case and CPU).

## Old code at 98280cda11989319c89aee841b28f0331ca3eca4

```
Start PWM   66   
Max PWM     252  

1424 ┼                                                                                                ╭──
1329 ┤                                                                                        ╭╮╭─────╯
1234 ┤                                                                                  ╭─────╯╰╯
1139 ┤                                                                         ╭────────╯
1044 ┤                                                                  ╭──────╯
 949 ┤                                                           ╭──────╯
 854 ┤                                                     ╭─────╯
 759 ┤                                              ╭──────╯
 665 ┤                                       ╭──────╯
 570 ┤                                  ╭────╯
 475 ┤                             ╭────╯
 380 ┤                         ╭───╯
 285 ┤                         │
 190 ┤                         │
  95 ┤                         │
   0 ┼─────────────────────────╯
RPM / PWM

f71808a -> pwm2

Start PWM   0
Max PWM     255

1265 ┼                                                                                                  ╭
1181 ┼╮                                                                                               ╭─╯
1096 ┤╰╮                                                                                     ╭────────╯
1012 ┤ ╰╮                                                                          ╭─────────╯
 928 ┤  │                                                                  ╭───────╯
 843 ┤  ╰╮                                                         ╭───────╯
 759 ┤   ╰╮                                                ╭───────╯
 675 ┤    │                                        ╭───────╯
 590 ┤    ╰╮                                ╭──────╯
 506 ┤     │                            ╭───╯
 422 ┤     │                            │
 337 ┤     │                            │
 253 ┤     ╰╮                          ╭╯
 169 ┤      │                          │
  84 ┤      │                          │
   0 ┤      ╰──────────────────────────╯
RPM / PWM
```

## Original with longer wait times

This is my measurement with the original code but a) have it wait for 30 seconds to settle down and b) wait 10 seconds between each measurement step. This is "expected", since I can confirm these values by manually echoing into hwmon.

```
Start PWM   67   
Max PWM     251  

1407 ┼                                                                                              ╭╮╭─╮
1313 ┤                                                                                       ╭──────╯╰╯ │
1220 ┤                                                                                ╭──────╯          │
1126 ┤                                                                        ╭───────╯                 │
1032 ┤                                                                  ╭─────╯                         │
 938 ┤                                                           ╭──────╯                               │
 844 ┤                                                   ╭───────╯                                      │
 750 ┤                                              ╭────╯                                              │
 657 ┤                                       ╭──────╯                                                   │
 563 ┤                                 ╭─────╯                                                          │
 469 ┤                            ╭────╯                                                                │
 375 ┤                         ╭──╯                                                                     │
[... 166 lines cut ...]
Start PWM   78
Max PWM     255

1212 ┼                                                                                                 ╭─
1131 ┤                                                                                           ╭─────╯
1050 ┤                                                                                  ╭────────╯
 970 ┤                                                                          ╭───────╯
 889 ┤                                                                  ╭───────╯
 808 ┤                                                          ╭───────╯
 727 ┤                                                   ╭──────╯
 646 ┤                                           ╭───────╯
 566 ┤                                     ╭─────╯
 485 ┤                               ╭─────╯
 404 ┤                              ╭╯
 323 ┤                              │
 242 ┤                              │
 162 ┤                              │
  81 ┤                             ╭╯
   0 ┼─────────────────────────────╯
RPM / PWM
```

Pwm1 shows a little bump (also confirmed by a second run).  The issue is that the fan at pwm1 (a Noctua redux PWM 140mm 1200rpm) starts spinning around 22 PWM, but the shitty motherboard doesn't detect these low RPMs. So what happens is the fan gets its initial somewhat faster startup kick but settles to below the board-threshold again (at around 68 PWM) and then isn't picked up until it reaches 68 PWM.

Technically the startup-kick shouldn't be detected - maybe, but on the other hand the fan is actually spinning. For me in my current setup this solution is fine. 👍

markusressel, 2021-09-29:
Well thats an interesting edge case there for sure :smile: 
Not sure if we could or even should account for this...

> This is my measurement with the original code but a) have it wait for 30 seconds to settle down and b) wait 10 seconds between each measurement step.

Regarding the 10 second delay between measurements: I think we could apply the same "settle" logic I implemented for the spindown to the  other measuring steps as well. However, this would lead to drastically longer measuring phases, since we would have to take a couple measurements (which are only updated every second) to even be able to detect if the fan has settled. I think just assuming it has within a fixed amount of time is the better option here. Worst case scenario is that the fans are actually spinning a little bit faster than measured, which imo is acceptable.

> 021/09/29 15:48:33 File reading error open /sys/devices/platform/f71882fg.656/pwm3_enable: no such file or directory
Not sure where this comes from, maybe I am logging too much?
Is the fan control working despite this error?

Schlaefer, 2021-09-29:
> Not sure if we could or even should account for this...

Agreed. If fan2go becomes more popular there could be a desire to address that issues, but until then the current implementation seems totally fine to me. :+1: 

> Regarding the 10 second delay between measurements 

I wouldn't consider the 10 seconds a recommendation either. It was a "I know this doesn't harm my setup, I know what I'm doing, I prefer slow but accurate and don't care if it takes an hour" situation. I was just throwing it in as datapoint. No call for action. :)

> Not sure where this comes from, maybe I am logging too much? Is the fan control working despite this error?

This is supposed to be an always-on at 100% AIO liquid cooler header and it can't be controlled by the Linux driver. I forgot to delete it from the output. Fancontrol can't control it either. Don't worry, this is fine.(tm)

markusressel, 2021-09-29:
I will close this issue for now.
Before making a new release I will probably put some additional work in, so feel to use the version from `master` (or the old one with the database from the `master` version).

closed 2021-09-29 by markusressel