markusressel/fan2go #30: Distinguish between startPwm and minPwm

markusressel, 2021-09-28:
For most fans there is a difference between
* the min pwm at which the fan starts to rotate from a stand still
* and the max pwm at which the fan still rotates when it was rotating previously

where the latter is smaller than the former in most cases.

To be able to securely make a fan start up, we need to know the `startPwm` value.
To also be able to let the fan run at a lower speed than `startPwm` (which might be possible or not) we also need to know the `minPwm` value.

Alternatively we could simply "jumpstart" fans by setting them to 100% for a brief moment, but this would also require knowledge about how long to do this without inducing annoying "ramping" sounds.

markusressel, 2025-05-29:
Yes it exists, and it can be set manually, but detecting it automatically is more tricky than one would think because real live physics is messy and in addition to that fans and even controllers act in wildly different ways. The whole initialization algorithm is not really optimal at the moment, it takes forever and still provides suboptimal results. Good enough for a first guess though. There were already proposals for improving the RPM curve measurement but it was not s priority to implement it yet. If you are interested in taking a look that would be awesome! 😎

DanielGibson, 2025-05-29:
Not sure yet, I might eventually, let's get the nvidia thing (and potential related refactorings) done first ;)

By the way: What exactly is the initialization (or its results) currently used for?

If no Min/Start/MaxPWM values are set in the config, those are calculated/measured
- But if those values are set in the config then IIRC those are shown at the end of init instead of the measured ones?
- .. but weirdly, the init process still does not *use* MaxPWM if configured, but goes up all to 255, even if I know more than 100 won't make a difference (=> takes 2.55 times as long as needed)
- I observed that on my laptop the fan started spinning at PWM **11** (and `fan2go fan -i myfan -v init` printed that it runs at 950RPM or whatever at PWM 11), but at the end it reported MinPWM/StartPWM as **12**.

Another potential usecase would be allowing the user to set a target RPM and the data from initialization is then used to look up the PWM value that yields the RPM closest to the desired value - but as far as I can tell, that feature doesn't exist, fan curves use PWM values, not RPM. Which, to be clear, is fine and totally works for me, just wondering if I'm missing something here.

Is it used for anything else?

BTW, can the measured data be displayed in a way where one can see any details? That printed curve looks pretty and everything, but is not *that* useful as the X-axis isn't labeled. Would be useful if one could optionally print the PWM vs RPM values as a list (maybe with configurable step size, i.e. only every 5th value if I want that or otherwise every single value)

> because real live physics is messy

Not sure if this is what you mean, but it can of course happen that a fan currently starts at PWM 11, but in a year or two when it's weared down a bit, or just dirtier, it might need some higher value (12? 13? 15? who knows) - and probably the MinPWM-value can similarly change over time.
Not sure what to do about that though, except maybe (if RPM-measurement is available) to increase PWM until the fan starts spinning if a PWM-value >= StartPWM is set (no idea if this maybe already is done).

markusressel, 2025-05-29:
> What exactly is the initialization (or its results) currently used for?

If I remember correctly:
* min/start and max pwm
* setPwmToGetPwm map
* pwmMap
* rpm curve (for display only)

> => takes 2.55 times as long as needed)

Well, as I said, the code for this is not really optimal, to say the least. I have in the past considered multiple times to get rid of it entirely, until a much better implementation could be made, but it hasn't happened yet.

> Another potential usecase would be allowing the user to set a target RPM and the data from initialization is then used to look up the PWM value that yields the RPM closest to the desired value

Thats the exact thought I had that made me start the development of fan2go. And yet, many years later, I still haven't had the time or need to actually implement it 😅 But yes, this would be very cool indeed and a valid use of the data that we (mostly) already have.

See also #152 

> BTW, can the measured data be displayed in a way where one can see any details?

Not via CLI, but you can get the full data via the API.
See: https://github.com/markusressel/fan2go?tab=readme-ov-file#api

> can similarly change over time.

Yes, thats precisely what I mean. The goal of fan2go is to set a configuration once, and never have to think about it again. It should just work. This started out with the correct operation after a system suspend (because fancontrol was not capable of that and the workarounds suggested were atrocious) and I would want to expand it to every other aspect of fan2go as well.

> increase PWM until the fan starts spinning if a PWM-value >= StartPWM is set

Not really, but kind of. The idea to expand this to StartPWM is a very good one.
If `neverStop: true` is set, the RPM of a fan will be checked, and the MinPWM will be automatically increased until it spins. This logic also isn't very optimal though, as it is prone to overshooting a lot. But at least it keeps the system from dying 😄