markusressel/fan2go #34: fan2go doesn't run initial measurement without specifying the config file as argument

Schlaefer, 2021-09-29:
A configuration file `/etc/fan2go/fan2go.yaml` exists. 

Expected: Running `fan2go` should pick up the configuration file and start the initial measurement.

Actual: Error "no valid fan configuration" unless the configuration file path is passed as argument.

Example:

```
❯ ls /etc/fan2go/
total 76
drwxr-xr-x 1 root  root     90 29. Sep 17:28 .
drwxr-xr-x 1 root  root   4958 29. Sep 15:17 ..
-rw-r--r-- 1 siezi siezi  1754 26. Sep 20:59 fan2go.yaml

❯ sudo fan2go
2021/09/29 17:29:16 File reading error open /sys/devices/platform/f71882fg.656/pwm3_enable: no such file or directory
2021/09/29 17:29:16 Ignoring unconfigured sensor acpitz/temp2_input
2021/09/29 17:29:16 Ignoring unconfigured sensor coretemp/temp2_input
2021/09/29 17:29:16 Ignoring unconfigured sensor coretemp/temp3_input
2021/09/29 17:29:16 Ignoring unconfigured sensor coretemp/temp4_input
2021/09/29 17:29:16 Ignoring unconfigured sensor coretemp/temp5_input
2021/09/29 17:29:16 Ignoring unconfigured sensor f71808a/temp1_input
2021/09/29 17:29:16 Ignoring unconfigured sensor f71808a/temp2_input
2021/09/29 17:29:16 Ignoring unconfigured sensor amdgpu/temp1_input
2021/09/29 17:29:16 Ignoring unconfigured sensor amdgpu/temp2_input
2021/09/29 17:29:16 Ignoring unconfigured sensor amdgpu/temp3_input
2021/09/29 17:29:16 Ignoring unconfigured fan f71808a/pwm1 (f71882fg.656)
2021/09/29 17:29:16 Ignoring unconfigured fan f71808a/pwm2 (f71882fg.656)
2021/09/29 17:29:16 Ignoring unconfigured fan f71808a/pwm3 (f71882fg.656)
2021/09/29 17:29:16 Ignoring unconfigured fan amdgpu/pwm1 (hwmon3)
2021/09/29 17:29:16 No valid fan configurations, exiting.

❯ sudo fan2go -c /etc/fan2go/fan2go.yaml
2021/09/29 17:29:20 File reading error open /sys/devices/platform/f71882fg.656/pwm3_enable: no such file or directory
2021/09/29 17:29:20 Ignoring unconfigured sensor acpitz/temp1_input
2021/09/29 17:29:20 Ignoring unconfigured sensor acpitz/temp2_input
2021/09/29 17:29:20 Ignoring unconfigured sensor coretemp/temp2_input
2021/09/29 17:29:20 Ignoring unconfigured sensor coretemp/temp3_input
2021/09/29 17:29:20 Ignoring unconfigured sensor coretemp/temp4_input
2021/09/29 17:29:20 Ignoring unconfigured sensor coretemp/temp5_input
2021/09/29 17:29:20 Ignoring unconfigured sensor f71808a/temp2_input
2021/09/29 17:29:20 Ignoring unconfigured sensor amdgpu/temp1_input
2021/09/29 17:29:20 Ignoring unconfigured sensor amdgpu/temp2_input
2021/09/29 17:29:20 Ignoring unconfigured sensor amdgpu/temp3_input
2021/09/29 17:29:20 Ignoring unconfigured fan f71808a/pwm3 (f71882fg.656)
2021/09/29 17:29:20 Ignoring unconfigured fan amdgpu/pwm1 (hwmon3)
2021/09/29 17:29:20 Gathering data...
2021/09/29 17:29:20 Gathering data...
...
```

markusressel, 2021-09-29:
What is the content of `/etc/fan2go/fan2go.yaml`?

Did you run this from the checked out git repository? Because `fan2go` also looks for a configuration file in `./` and the repository has the example configuration file.

Schlaefer, 2021-09-29:
> What is the content of /etc/fan2go/fan2go.yaml?

https://pastebin.com/jf17pzrS

Ignore the comments, I didn't change them from the template.

I'm not a Linux guru, so maybe I'm missing an important part here. Downloading

```
curl -L -o fan2go https://github.com/markusressel/fan2go/releases/latest/download/fan2go-linux-amd64
chmod +x fan2go
sudo cp ./fan2go /usr/bin/fan2go
sudo fan2go
...
```

works fine.

But if I check-out from source and build I get:

```
❯ git checkout --force 43ac3190c33e0e9070a725a7a6c288d05c543ee2
Previous HEAD position was 98280cd Merge pull request #27 from markusressel/dependabot/go_modules/github.com/spf13/viper-1.9.0
HEAD is now at 43ac319 v0.0.17

❯ go build

❯ chmod +x fan2go

❯ sudo cp ./fan2go /usr/bin/fan2go

❯ sudo fan2go
2021/09/29 21:45:39 File reading error open /sys/devices/platform/f71882fg.656/pwm3_enable: no such file or directory
2021/09/29 21:45:39 Ignoring unconfigured sensor acpitz/temp2_input
2021/09/29 21:45:39 Ignoring unconfigured sensor coretemp/temp2_input
2021/09/29 21:45:39 Ignoring unconfigured sensor coretemp/temp3_input
2021/09/29 21:45:39 Ignoring unconfigured sensor coretemp/temp4_input
2021/09/29 21:45:39 Ignoring unconfigured sensor coretemp/temp5_input
2021/09/29 21:45:39 Ignoring unconfigured sensor f71808a/temp1_input
2021/09/29 21:45:39 Ignoring unconfigured sensor f71808a/temp2_input
2021/09/29 21:45:39 Ignoring unconfigured sensor amdgpu/temp1_input
2021/09/29 21:45:39 Ignoring unconfigured sensor amdgpu/temp2_input
2021/09/29 21:45:39 Ignoring unconfigured sensor amdgpu/temp3_input
2021/09/29 21:45:39 Ignoring unconfigured fan f71808a/pwm1
2021/09/29 21:45:39 Ignoring unconfigured fan f71808a/pwm2
2021/09/29 21:45:39 Ignoring unconfigured fan f71808a/pwm3
2021/09/29 21:45:39 Ignoring unconfigured fan amdgpu/pwm1
2021/09/29 21:45:39 No valid fan configurations, exiting.
```

It's not important but I appreciate any hint by a sensai, because I'm clueless. ;)

markusressel, 2021-09-29:
> There's only one yaml config file at /etc, but neither at /root nor ./.

Yes, yes there is :nerd_face: The "template" you mentioned is inside the root folder of this repository.
If you `git checkout` and `cd fan2go`, this template will be at `./fan2go.yaml` and preferred over any other config file, since it is located in your working directory.

[image]

Schlaefer, 2021-09-29:
> There's only one yaml config file at /etc, but neither at /root nor ./.

And of course I'm a filthy liar. There's the fan2go.yaml in the git root. And I always worked in that directory. 🤦‍♂️

closed 2021-09-29 by Schlaefer