markusressel/fan2go #47: Error in `detect` (thinkpad x1 yoga gen 6)

salvoc81, 2021-11-17:
Hello and thank you for creating this software.

I am getting an error when running `fan2go detect`, any suggestions on the possible problem?

I am attaching the output of lm_sensors followed by that of `detect`  

```
ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +33.0°C  

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
fan2:           0 RPM
CPU:          +36.0°C  
GPU:              N/A  
temp3:         +0.0°C  
temp4:         +0.0°C  
temp5:         +0.0°C  
temp6:         +0.0°C  
temp7:         +0.0°C  
temp8:            N/A  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +31.9°C  (low  =  -0.1°C, high = +79.8°C)
                       (crit = +81.8°C)
Sensor 1:     +30.9°C  (low  = -273.1°C, high = +65261.8°C)

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +35.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +32.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          15.62 V  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +36.0°C  (crit = +128.0°C)
```

Output of the detect command:
```
Detected Devices:
acpitz
  1: hwmon1 (temp1_input): 35000
nvme
  1: Composite (temp1_input): 32850
  2: Sensor 1 (temp2_input): 31850
thinkpad
  1: hwmon4 (pwm1): RPM: 0 PWM: 255 Auto: false
  1: CPU (temp1_input): 35000
  ERROR   File reading error: read /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon4/temp2_input: no such device or address
  2: GPU (temp2_input): -1
  3: hwmon4 (temp3_input): 0
  4: hwmon4 (temp4_input): 0
  5: hwmon4 (temp5_input): 0
  6: hwmon4 (temp6_input): 0
  7: hwmon4 (temp7_input): 0
  ERROR   File reading error: read /sys/devices/platform/thinkpad_hwmon/hwmon/hwmon4/temp8_input: no such device or address
  8: hwmon4 (temp8_input): -1
coretemp
  1: Package id 0 (temp1_input): 34000
  2: Core 0 (temp2_input): 33000
  3: Core 1 (temp3_input): 33000
  4: Core 2 (temp4_input): 31000
  5: Core 3 (temp5_input): 31000
iwlwifi_1
  1: hwmon6 (temp1_input): 32000 
```

Also, please notice the problem with the the repeated index=1 under the `thinkpad` platform

markusressel, 2021-11-17:
Not sure whats going on with the errors, but the index is not repeating. The index is counting for sensors and fans, one is a sensor, one is a fan.

salvoc81, 2021-11-18:
Hello @markusressel basically the errors are related to those files have no value.  
The real problem I am having is that I require the AUR module [thinkfan](https://wiki.archlinux.org/title/Fan_speed_control#ThinkPad_laptops) in order for `fan2go` to work.  
Note that [thinkfan is also not working](https://github.com/vmatare/thinkfan/issues/165).  

`fan2go` is working, but I am getting this warning:
```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Ignoring unconfigured sensor acpitz/temp1_input
 INFO  Ignoring unconfigured sensor nvme/temp1_input
 INFO  Ignoring unconfigured sensor nvme/temp2_input
 INFO  Ignoring unconfigured sensor thinkpad/temp1_input
 INFO  Ignoring unconfigured sensor thinkpad/temp2_input
 INFO  Ignoring unconfigured sensor thinkpad/temp3_input
 INFO  Ignoring unconfigured sensor thinkpad/temp4_input
 INFO  Ignoring unconfigured sensor thinkpad/temp5_input
 INFO  Ignoring unconfigured sensor thinkpad/temp6_input
 INFO  Ignoring unconfigured sensor thinkpad/temp7_input
 INFO  Ignoring unconfigured sensor thinkpad/temp8_input
 INFO  Ignoring unconfigured sensor iwlwifi_1/temp1_input
 INFO  Ignoring unconfigured sensor coretemp/temp2_input
 INFO  Ignoring unconfigured sensor coretemp/temp3_input
 INFO  Ignoring unconfigured sensor coretemp/temp4_input
 INFO  Ignoring unconfigured sensor coretemp/temp5_input
 INFO  Gathering sensor data for fan1...
 INFO  Loading fan curve data for fan 'fan1'...
 INFO  Start PWM of fan1 (hwmon4, pwm1): 1
 INFO  Max PWM of fan1 (hwmon4, pwm1): 255
 INFO  Starting controller loop for fan 'fan1' (hwmon4, pwm1)
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 1 but is now: 0
 WARNING  WARNING: Increasing startPWM of fan1 from 1 to 2, which is supposed to never stop, but RPM is 0.000000
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 2 but is now: 0
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 2 but is now: 0
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 2 but is now: 0
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 2 but is now: 0
...
```

I believe `fan2go` is still the best option for me, so I will try enable the service and see how it goes.

markusressel, 2021-11-18:
fan2go can only work if lm-sensors works. fan2go does nothing to directly interact with fans or sensors, it only interacts with the hwmon filesystem provided by lm-sensors.

salvoc81, 2021-11-18:
I was able to let fa2go work without [thinkfan](https://wiki.archlinux.org/title/Fan_speed_control#ThinkPad_laptops) by simply enabling the kernel module for `thinkpad_acpi` as described in the arch wiki page.  
The only issue right now is that the warning are a bit verbose.  
Do you think it would safe to remove such warning?  
If you direct me to the relevant source files I might such a feature myself (e.g., `--suppress-warning`)  

Thanks a lot.
Will let you know if it works well

markusressel, 2021-11-18:
Which warnings do you refer to? This one?:

```
 WARNING  PWM of fan1 was changed by third party! Last set PWM value was: 2 but is now: 0
```

Not sure about adding a toggle for this... if you get this, your system is not configured correctly and two different applications (or the UEFI/BIOS) are trying to control your fans. This will most probably cause some form of issue.

If you want to ignore the logs, I guess you can also change the systemd unit file to not store any logs.

markusressel, 2021-11-18:
I just realized that `sensors` outputs those inputs as `N/A`...
fan2go probably should be able to handle those cases more gracefully as well.

eldorel, 2021-12-03:
I'd like to followup on the "changed by third party" errors. 

I'm seeing them on all of the fans on my corsair commander pro, and the values are always within 2 points of the values set by fan2go. 

No other fan control software is running, and the bios does not affect those fans at all.

markusressel, 2021-12-03:
@eldorel please create a new issue for your scenario, since it is not related to this one.

closed 2022-04-04 by markusressel