markusressel/fan2go #48: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5a878d]

lbutler112233, 2021-11-17:
ERROR   File reading error: open /sys/devices/virtual/hwmon/hwmon4/pwm1_enable: no such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5a878d]

goroutine 1 [running]:
github.com/markusressel/fan2go/internal.createFans(0xc0000ac810, 0x21, 0x21, 0x0, 0x0)
	/github/workspace/internal/backend.go:598 +0x4cd
github.com/markusressel/fan2go/internal.FindControllers(0xc000082900, 0x8f2da0, 0xe185e0, 0x0, 0x0)
	/github/workspace/internal/backend.go:545 +0x293
github.com/markusressel/fan2go/cmd.glob..func2(0xe0b280, 0xe35450, 0x0, 0x0)
	/github/workspace/cmd/root.go:56 +0xe4
github.com/spf13/cobra.(*Command).execute(0xe0b280, 0xe35450, 0x0, 0x0, 0xe0b280, 0xe35450)
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xe0b000, 0xe3524f, 0x9cc8fe, 0x7)
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x35a
github.com/spf13/cobra.(*Command).Execute(...)
	/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
	/github/workspace/cmd/root.go:248 +0x382
main.main()
	/github/workspace/main.go:25 +0x20

Cryotize, 2021-11-17:
same here:

<pre><font color="#26A269"><b>$</b></font> sudo fan2go detect -v
<span style="background-color:#F66151"><font color="#171421">  ERROR  </font></span> <font color="#F66151">File reading error: open /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-5/1-5.2/1-5.2:1.0/0003:1B1C:0C10.0007/hwmon/hwmon5/pwm1_enable: no such file or directory</font>
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x563857b0eda1]

goroutine 1 [running]:
github.com/markusressel/fan2go/internal.createFans({0xc0001e4240, 0x85})
	github.com/markusressel/fan2go/internal/backend.go:600 +0x3c1
github.com/markusressel/fan2go/internal.FindControllers()
	github.com/markusressel/fan2go/internal/backend.go:547 +0x248
github.com/markusressel/fan2go/cmd.glob..func2(0x563857f681a0, {0x563857c886fd, 0x1, 0x1})
	github.com/markusressel/fan2go/cmd/root.go:56 +0x9b
github.com/spf13/cobra.(*Command).execute(0x563857f681a0, {0xc000119510, 0x1, 0x1})
	github.com/spf13/cobra@v1.2.1/command.go:860 +0x5f8
github.com/spf13/cobra.(*Command).ExecuteC(0x563857f67f20)
	github.com/spf13/cobra@v1.2.1/command.go:974 +0x3bc
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
	github.com/markusressel/fan2go/cmd/root.go:248 +0x2a6
main.main()
	./main.go:25 +0x17</pre>

markusressel, 2021-11-18:
Seems like the `pwmX_enable` file doesn't always exist. Not sure how to handle those, can they still be controlled?

Cryotize, 2021-11-18:
Fancontrol manages to control them, i can give you my config, maybe this helps:

<pre># Configuration file generated by pwmconfig, changes will be lost
INTERVAL=10
DEVPATH=hwmon2=devices/pci0000:00/0000:00:18.3 hwmon5=devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-5/1-5.2/1-5.2:1.0/0003:1B1C:0C10.0008
DEVNAME=hwmon2=k10temp hwmon5=corsaircpro
FCTEMPS=hwmon5/pwm6=hwmon2/temp1_input hwmon5/pwm5=hwmon2/temp1_input hwmon5/pwm4=hwmon2/temp1_input hwmon5/pwm3=hwmon2/temp1_input hwmon5/pwm2=hwmon2/temp1_input hwmon5/pwm1=hwmon2/temp1_input
FCFANS=hwmon5/pwm6=hwmon5/fan6_input hwmon5/pwm5=hwmon5/fan5_input hwmon5/pwm4=hwmon5/fan4_input hwmon5/pwm3=hwmon5/fan3_input hwmon5/pwm2=hwmon5/fan2_input hwmon5/pwm1=hwmon5/fan1_input
MINTEMP=hwmon5/pwm6=35 hwmon5/pwm5=35 hwmon5/pwm4=35 hwmon5/pwm3=35 hwmon5/pwm2=35 hwmon5/pwm1=35
MAXTEMP=hwmon5/pwm6=70 hwmon5/pwm5=70 hwmon5/pwm4=70 hwmon5/pwm3=70 hwmon5/pwm2=70 hwmon5/pwm1=70
MINSTART=hwmon5/pwm6=2 hwmon5/pwm5=2 hwmon5/pwm4=2 hwmon5/pwm3=2 hwmon5/pwm2=2 hwmon5/pwm1=2
MINSTOP=hwmon5/pwm6=0 hwmon5/pwm5=0 hwmon5/pwm4=0 hwmon5/pwm3=0 hwmon5/pwm2=0 hwmon5/pwm1=0
MAXPWM=hwmon5/pwm6=180 hwmon5/pwm5=180 hwmon5/pwm4=180 hwmon5/pwm3=180 hwmon5/pwm2=180 hwmon5/pwm1=180</pre>

markusressel, 2021-11-30:
Please try again with version `0.1.0`.

markusressel, 2021-12-01:
OK maybe `0.2.0` does the trick?

Cryotize, 2021-12-02:
It does, works like a charm!

Thanks a lot, appreciate your work :heart:

closed 2021-12-02 by markusressel