markusressel/fan2go #49: Fan stop is detected at differrent PWM level than with fancontrol

Shished, 2021-11-18:
So I switched from `fancontrol` to this app and noticed that detected stopping PWM levels for case fans are way higher than before.
`fancontrol` detected stop level is 30 for both fans while in `fan2go` it is 169 for the front fan and 151 for the back fan.
Why is there such discrepancy? It is the same for a CPU fan.

`pwmconfig` script which is used to gather the curve data for fancontrol checks the speed from Max to Min while `fan2go` does it in the opposite way. IMO, this is the source of problem because it is hard for a fan to start spinning from idle at low PWM levels, when the average fan voltage is too low.

markusressel, 2021-11-18:
It is intentional that we measure from the bottom, to make sure the fan is able to start at a given PWM level 100% of the time, otherwise the `neverStop` config option can not be guaranteed. fan2go **does not** detect the "stop" speed, it only detects the "start" speed.

The difference between 30 and 169 doesn't look right though.
Is this a big fan?
Is it (from a stand still) able to start  at PWM 30? 
Or is it really only able to start at around 169 (nice)?

I have thought about adding a "kickstart" PWM speed to improve the situation, which should apply a higher PWM value for a short amount of time when the current RPM is 0 and/or the current PWM value is below a certain threshold.

markusressel, 2022-01-31:
closing because of inactivity

closed 2022-01-31 by markusressel