markusressel/fan2go #51: Cannot run fan2go detect

tamimmahmudkhan, 2021-11-29:
Whenever I try I get the following output. I can setup fancontrol just fine. But every reboot the hwmon paths change which causes problems.
```
fan2go detect
  ERROR   File reading error: open /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon4/pwm1_enable: no such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x5a878d]

goroutine 1 [running]:
github.com/markusressel/fan2go/internal.createFans(0xc000026880, 0x31, 0x31, 0x0, 0x0)
        /github/workspace/internal/backend.go:598 +0x4cd
github.com/markusressel/fan2go/internal.FindControllers(0xc000001b00, 0x8f2da0, 0xe185e0, 0x0, 0x0)
        /github/workspace/internal/backend.go:545 +0x293
github.com/markusressel/fan2go/cmd.glob..func2(0xe0b280, 0xe35450, 0x0, 0x0)
        /github/workspace/cmd/root.go:56 +0xe4
github.com/spf13/cobra.(*Command).execute(0xe0b280, 0xe35450, 0x0, 0x0, 0xe0b280, 0xe35450)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xe0b000, 0xe3524f, 0x9cc8fe, 0x7)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x35a
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
        /github/workspace/cmd/root.go:248 +0x382
main.main()
        /github/workspace/main.go:25 +0x20
```

markusressel, 2021-11-30:
Should be fixed in `0.1.0`, please let me know.

markusressel, 2021-11-30:
Same as #48, closing.

closed 2021-11-30 by markusressel