markusressel/fan2go #53: Sensor detection issue on 0.1.0

tamimmahmudkhan, 2021-12-01:
This is the new error I get when trying to detect sensors with `fan2go detect`.
```
igneel@G3-3590 ~/Downloads> fan2go detect
> acpitz
 Sensors   Index   Label    Value  
           1       hwmon1   25000  

> nvme-pci-0200
 Sensors   Index   Label       Value  
           1       Composite   63850  

  ERROR   File reading error: open /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon4/pwm1_enable: no such file or directory
  FATAL   Cannot read pwm_enable value of 
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:340
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0xe210c0, 0x9e8958, 0x22, 0xc0000b7248, 0x1, 0x1, 0x3d)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:282 +0x23f
github.com/markusressel/fan2go/internal/ui.Fatal(...)
        /github/workspace/internal/ui/logging.go:36
github.com/markusressel/fan2go/cmd.createFans(0xc000026ac0, 0x31, 0x0, 0x0, 0x0)
        /github/workspace/cmd/detect.go:162 +0x75d
github.com/markusressel/fan2go/cmd.glob..func2(0xe23560, 0xe4d770, 0x0, 0x0)
        /github/workspace/cmd/detect.go:51 +0x38d
github.com/spf13/cobra.(*Command).execute(0xe23560, 0xe4d770, 0x0, 0x0, 0xe23560, 0xe4d770)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xe237e0, 0xe4d56f, 0x9d923b, 0x7)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x35a
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
        /github/workspace/cmd/root.go:71 +0x2bf
main.main()
        /github/workspace/main.go:25 +0x20
igneel@G3-3590 ~/Downloads [2]> sudo fan2go detect
> acpitz
 Sensors   Index   Label    Value  
           1       hwmon1   25000  

> nvme-pci-0200
 Sensors   Index   Label       Value  
           1       Composite   63850  

  ERROR   File reading error: open /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon4/pwm1_enable: no such file or directory
  FATAL   Cannot read pwm_enable value of 
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:340
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0xe210c0, 0x9e8958, 0x22, 0xc0000c3248, 0x1, 0x1, 0x3d)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:282 +0x23f
github.com/markusressel/fan2go/internal/ui.Fatal(...)
        /github/workspace/internal/ui/logging.go:36
github.com/markusressel/fan2go/cmd.createFans(0xc0000d0940, 0x31, 0x0, 0x0, 0x0)
        /github/workspace/cmd/detect.go:162 +0x75d
github.com/markusressel/fan2go/cmd.glob..func2(0xe23560, 0xe4d770, 0x0, 0x0)
        /github/workspace/cmd/detect.go:51 +0x38d
github.com/spf13/cobra.(*Command).execute(0xe23560, 0xe4d770, 0x0, 0x0, 0xe23560, 0xe4d770)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xe237e0, 0xe4d56f, 0x9d923b, 0x7)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x35a
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
        /github/workspace/cmd/root.go:71 +0x2bf
main.main()
        /github/workspace/main.go:25 +0x20
igneel@G3-3590 ~/Downloads [2]> sudo fan2go detect
> acpitz
 Sensors   Index   Label    Value  
           1       hwmon1   25000  

> nvme-pci-0200
 Sensors   Index   Label       Value  
           1       Composite   63850  

  ERROR   File reading error: open /sys/devices/platform/dell_smm_hwmon/hwmon/hwmon4/pwm1_enable: no such file or directory
  FATAL   Cannot read pwm_enable value of 
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:340
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0xe210c0, 0x9e8958, 0x22, 0xc0000c3248, 0x1, 0x1, 0x3d)
        /go/pkg/mod/github.com/pterm/pterm@v0.12.33/prefix_printer.go:282 +0x23f
github.com/markusressel/fan2go/internal/ui.Fatal(...)
        /github/workspace/internal/ui/logging.go:36
github.com/markusressel/fan2go/cmd.createFans(0xc0000d0940, 0x31, 0x0, 0x0, 0x0)
        /github/workspace/cmd/detect.go:162 +0x75d
github.com/markusressel/fan2go/cmd.glob..func2(0xe23560, 0xe4d770, 0x0, 0x0)
        /github/workspace/cmd/detect.go:51 +0x38d
github.com/spf13/cobra.(*Command).execute(0xe23560, 0xe4d770, 0x0, 0x0, 0xe23560, 0xe4d770)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:860 +0x2aa
github.com/spf13/cobra.(*Command).ExecuteC(0xe237e0, 0xe4d56f, 0x9d923b, 0x7)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x35a
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:902
github.com/markusressel/fan2go/cmd.Execute()
        /github/workspace/cmd/root.go:71 +0x2bf
main.main()
        /github/workspace/main.go:25 +0x20
```

markusressel, 2021-12-01:
Yeah sorry, I think I didn't test that one right.
I have released 0.2.0 just now, please have a look, thx.

markusressel, 2021-12-01:
Closing as duplicate of #48

closed 2021-12-01 by markusressel