markusressel/fan2go #62: Is it possible to archive the release artifacts?

elais, 2021-12-03:
I know it's adding a bit of a burden but it would be easier for me to package for my distro if the release artifacts were tarred. Feel free to say no I'm just asking for a hand.

markusressel, 2021-12-03:
You mean instead of just uploading the binary directly?
I guess that would be possible, but I am not sure what it would add to this repository.

Just using the official binary in a package isn't the best option either, since [some seem to have problems with different versions of dynamically linked libraries](https://github.com/markusressel/fan2go/discussions/55). Therefore the fan2go (and the package) should be built on the target system either way, just like the AUR package does.

elais, 2021-12-03:
Actually I figured how to build from source literally two minutes ago. Sorry for bothering you. And thanks for the reply.

closed 2021-12-03 by elais

markusressel, 2021-12-03:
No problem, glad it is working :)
Feel free to give me a hint when your package is available, so we can list in in the README :nerd_face: