markusressel/fan2go #63: Not reading PWM values on corsair commander pro fan controller.

elais, 2021-12-03:
Hi. Fan2go i having trouble reading the pwm_enable flag for the corsair commander pro, which I think is leading to it not being able to enable the fan control on the fans connected to the controller. Here's the output of my run.

```bash
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 WARNING  Cannot read pwm_enable value of back_top
 WARNING  Cannot read pwm_enable value of back_bottom
 INFO  Gathering sensor data for back_top...
 INFO  Gathering sensor data for back_bottom...
 WARNING  Cannot read pwm_enable value of front_bottom
 INFO  Gathering sensor data for front_bottom...
 WARNING  Cannot read pwm_enable value of front_top
 INFO  Gathering sensor data for front_top...
 INFO  Loading fan curve data for fan 'front_top'...
 INFO  Loading fan curve data for fan 'back_top'...
 INFO  Loading fan curve data for fan 'back_bottom'...
 INFO  Loading fan curve data for fan 'front_bottom'...
 WARNING  No fan curve data found for fan 'front_top', starting initialization sequence...
 WARNING  No fan curve data found for fan 'back_bottom', starting initialization sequence...
 WARNING  Could not enable fan control on front_top, trying to continue anyway...
 WARNING  No fan curve data found for fan 'back_top', starting initialization sequence...
 WARNING  No fan curve data found for fan 'front_bottom', starting initialization sequence...
```

And the output of Fan2go detect

```bash
> amdgpu-pci-0d00
 Fans      Index   Label    RPM   PWM   Auto
           1       hwmon8   0     0     true
 Sensors   Index   Label                    Value
           1       edge (temp1_input)       26000
           2       junction (temp2_input)   29000
           3       mem (temp3_input)        30000

> gigabyte_wmi-virtual-0
 Sensors   Index   Label                  Value
           1       hwmon4 (temp1_input)   41000
           2       hwmon4 (temp2_input)   38000
           3       hwmon4 (temp3_input)   31000
           4       hwmon4 (temp4_input)   20000
           5       hwmon4 (temp5_input)   37000
           6       hwmon4 (temp6_input)   42000

> nvme-pci-0100
 Sensors   Index   Label                     Value
           1       Composite (temp1_input)   52850
           2       Sensor 1 (temp2_input)    52850
           3       Sensor 2 (temp3_input)    70850

> acpitz-acpi-0
 Sensors   Index   Label                  Value
           1       hwmon0 (temp1_input)   16800

> corsaircpro-hid-3-7
 Fans      Index   Label       RPM   PWM   Auto
           1       fan1 4pin   456   10    false
           2       fan2 4pin   834   0     false
           3       fan3 4pin   573   48    false
           4       fan4 4pin   838   0     false
 Sensors   Index   Label                  Value
           1       hwmon7 (temp1_input)   27080

> iwlwifi_1-virtual-0
 Sensors   Index   Label                  Value
           1       hwmon5 (temp1_input)   50000

> k10temp-pci-0c3
 Sensors   Index   Label                 Value
           1       Tctl (temp1_input)    31750
           2       Tccd1 (temp3_input)   31000

> nvme-pci-0800
 Sensors   Index   Label                     Value
           1       Composite (temp1_input)   42850
           2       Sensor 1 (temp2_input)    42850
           3       Sensor 2 (temp3_input)    45850
```

is there another way to check for a pwm signal?

markusressel, 2021-12-03:
fan2go just works with everything lmsensors has (well all fans and temp sensors). If lmsensors doesnt support your controller, fan2go wont work either. 

Can you post the output of `sensors`?

elais, 2021-12-04:
Sure thing

```bash
amdgpu-pci-0d00
Adapter: PCI adapter
vddgfx:      775.00 mV
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +31.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +33.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +36.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
power1:       19.00 W  (cap = 255.00 W)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +48.0°C

gigabyte_wmi-virtual-0
Adapter: Virtual device
temp1:        +42.0°C
temp2:        +42.0°C
temp3:        +47.0°C
temp4:        +20.0°C
temp5:        +41.0°C
temp6:        +44.0°C

nvme-pci-0100
Adapter: PCI adapter
Composite:    +53.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +68.8°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

ucsi_source_psy_3_00081-i2c-3-08
Adapter: Synopsys DesignWare I2C adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

corsaircpro-hid-3-7
Adapter: HID adapter
in0:          12.03 V
in1:           5.01 V
in2:           3.42 V
fan1 4pin:    899 RPM
fan2 4pin:    902 RPM
fan3 4pin:    899 RPM
fan4 4pin:    900 RPM
temp1:        +32.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +47.1°C
Tccd1:        +35.8°C

nvme-pci-0800
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +49.9°C  (low  = -273.1°C, high = +65261.8°C)
```

LM sensors supports it, and again `pwmconfig` picks up the fans' pwm signals.

markusressel, 2021-12-04:
Hmm not sure what to do, `fan2go` displays all columns correctly and even has values, f.ex. here:

```
> corsaircpro-hid-3-7
Fans      Index   Label       RPM   PWM   Auto
          1       fan1 4pin   456   10    false
```

Since it has PWM and RPM values, and `Auto` is set to false, fan2go should just work.

Did you make sure the fan speeds really don't change during the initialization phase?

You can try to run `fan2go` with the `--verbose` flag to get more output during initialization (and generally).

eldorel, 2021-12-05:
@elais  By any chance have you tried running fan2go detect as root?  

I have the commander pro, and other than a repeating error message it is working fine for me as long as I use the platform name matching. 
However, fan2go can't see the corsair's fans on my system unless it's run as root.

Here's the config for one of my fans: 

> fans:
>   - id: Front-01
>     hwmon:
>       platform: corsaircpro-hid-3-5
>       index: 1
>     neverStop: no
>     curve: cpu-package-curve-front
> 
> 
>

elais, 2021-12-05:
I’ll give that a try. So far I’ve run. It as sudo. I was doing maintenance on my system yesterday and it’s still powered down so I’ll give this a go when I’m back online.

elais, 2021-12-06:
@eldorel was right. It does work but also spits out the error he describes in issue #64. During maintenance I noticed two of my fan cables frayed so I replaced their heads. Perhaps they weren't getting the signal as well as they should have or something.

Closing this issue now and thanks for your patience.

closed 2021-12-06 by elais