markusressel/fan2go #70: Fail2go Exits when a single fan or sensor is missing

eldorel, 2021-12-11:
**Describe the bug**

Fan2go  _completely_ fails to start with the error "FATAL: Couldn't find hwmon device with platform " as soon as it reaches the first fan (or sensor) with an incorrect or missing platform.  

**To Reproduce**

1. Setup fan2go with working config with multiple fans, sensors, and curves. 
2. Modify the 'platform' for the first fan configuration to an incorrect value (add a digit to the end).
3. Save config
4. Restart Fan2go
5. monitor log/journal
6. Fan2go will exit with "FATAL: Couldn't find hwmon device with platform " despite the other fans and sensors being correctly configured

**Expected behavior**

Fan2go should alert the user to the broken configuration, but still continue to check {and control} any properly configured fans...

**Screenshots**

[image]

**Desktop (please complete the following information):**

- Distro: [Arch Linux/EndeavourOS]
- `uname -a`: 5.15.7-arch1-1 #1 SMP PREEMPT Wed, 08 Dec 2021 14:33:16 +0000 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.2.2

**Additional context**

So in my case, the USB controller is only running 6 fans out of 12 in this device, and Fan2go is controlling 6 of those. 
However Fan2go _can_ control all of them. 

Having Fan2go fail to activate half of the fans is a problem, but it failing to control the others when it fully could have done so could have unwanted side effects. 

**Additional Additional context**
Having fan2go ignore and alert on badly configured fans would have provided a quick workaround to a device that keeps alternating between two VERY similar Platform IDs, by duplicating the fan config with both platforms and simply allowing it to control using the platform ID that the module order loaded with while the _other 'incorrect'_ platform failed. 

@elais You also might want to double check this on your setup, since you have the same hardware.

eldorel, 2021-12-11:
Clarifying Note: this *also* affects the sensor declaration the same way. If the temperature sensor on the USB hid device is misnamed, fan2go exits instead of trying to work around it. 

In that case, this also affects curves, so it may be a more difficult issue to address. 

Additionally, the 'file' sensor does gracefully handle read failures, so the logic may already be in place to account for this without much effort.

markusressel, 2021-12-11:
Failing early when the configuration cannot be used was intentional.

The reasoning behind this was that if fan2go fails/dies, the fan is either reset to full speed, or controlled by some other entity. Also, a failing service is more of a critical failure and hopefully more noticeable than a simple error log (both are not great).

So the fundamental issue is, that it is not trivial for the enduser to notice fan2go failing to start (for whatever reason). Maybe issuing a system notification would be an option.

eldorel, 2021-12-11:
> The reasoning behind this was that if fan2go fails/dies, the fan is either reset to full speed, or controlled by some other entity. 

The failure I was seeing earlier was happening before fan2go finished reading the PWM values, so it didn't set the fans back to 100% on exit (and the previous release wasn't setting them to full on exit, so they were still sitting in idle). 

The current release is setting the PWM on exit again, so the only risk I see now is for the fans to have been stopped by another program prior to reboot.  
Unfortunately, the windows corsair device control software seems to park the fans before shutdown, so dual booting still has a risk of the fans being uncontrolled _and disabled_ on restart.  

>Also, a failing service is more of a critical failure and hopefully more noticeable than a simple error log (both are not great).

SystemD does not appear to be configured to issue any type of "service failed" notifications on a default install of Arch or EndeavorOS; so it would be *very* easy to miss unless the fans are doing something noticeably wrong or the user is specifically paying attention to them.
(in fact, I *did* miss this for about an hour because I was wearing headphones and didn't notice the lack of noise at first.)

markusressel, 2021-12-17:
> The current release is setting the PWM on exit again, so the only risk I see now is for the fans to have been stopped by another program prior to reboot. Unfortunately, the windows corsair device [...]

fan2go is not responsible for the behaviour of some other tool. If your fans are parked by default, then fan2go will restore them to this exact state when exiting.

If fan2go cannot fully work with the provided configuration, then it will crash. I know, never say never, but I am pretty sure that this design decision will not change in the foreseeable future.

> SystemD does not appear to be configured to issue any type of "service failed" notifications on a default install of Arch or EndeavorOS; so it would be _very_ easy to miss unless the fans are doing something noticeably wrong or the user is specifically paying attention to them.

Correct. That's why I proposed the introduction of a notification feature, although even that would require the user to have a notification daemon installed and running, which cannot taken for granted.

markusressel, 2021-12-20:
Closing in favor of #79

closed 2021-12-20 by markusressel