markusressel/fan2go #71: [Feature Request] Allow Fan and Sensor to utilise hwmonX/name file value for persistence

eldorel, 2021-12-11:
**Is your feature request related to a problem? Please describe.**

Every few reboots, a few of my LM_Sensors/hwmon devices shuffle around thanks to module load order. 
In most cases, using 'platform' works fine, but for USB HID devices, the platform data includes the port number and loading order. 

For example: My Corsair comander pro alternates between being identified as "corsaircpro-hid-3-2" and "corsaircpro-hid-3-13"

In this instance, fan2go fails to start *completely* with the error "FATAL: Couldn't find hwmon device with platform " as soon as it reaches the first fan or sensor with an incorrect or missing platform.  (I'm opening a [separate issue](https://github.com/markusressel/fan2go/issues/70#issue-1077494681) for that, since it probably should alert, but still continue to check {and control} the other, correctly configured, fans...)

**Describe the solution you'd like**
The HWMON spec includes a single file "name" (which is incidentally the *only* mandatory file in the folders), which should contain the name of the device without the port identifiers. 

I would appreciate the ability to set Name as an optional/additional way for Fan2go to identify the correct device.
 Since *most* users only have a single instance of their devices, the name should usually be sufficient, and *platform* is present for advanced users who have duplicate hardware *and* need to control those independently. 

**Describe alternatives you've considered**

I actually attempted to create a second FAN id: using the other platform identifier, expecting fan2go to gracefully move past any fan that it failed to load, which *should* result in only one actual instance of the fan loading. 

That did not work.

eldorel, 2021-12-11:
I realized I didn't provide a reference to the hwmon spec for my statement about 'name' being the only mandatory file. 

https://www.kernel.org/doc/html/latest/hwmon/sysfs-interface.html#attribute-access

> All entries (except name) are optional, and should only be created in a given driver if the chip has the feature.

markusressel, 2021-12-11:
Please don't mix the description of separate issues. It just causes confusion when reading both of your issues. I have edited your post accordingly.

The identifier shuffling around seems to be specific to USB HID devices, maybe we can get a more unique identifier from the USB device itself :thinking: The current "fix" was implemented by @elais and allows to use a regex for the platform config argument. So you can just use `corsaircpro*` if you just have one device.

I am not sure what your understanding of the "platform" is, but it is a generated string that is already built using the `name` file exposed by hwmon. It just has additional stuff attached to it to be able to distinguish multiple instances with the same name (and to be as similar to the `sensors` command as possible).

So, although it might be a little more intuitive to use `name` instead of `platform`, adding a `name:` config option would be pretty redundant.

It might be a good idea to create an FAQ section or Wiki where such cases are described in more detail.

elais, 2021-12-11:
@eldorel try the latest 0.2.4 build of fan2go and simply put "corsaircpro" in the platform fields. It should pick up your commander every time.

eldorel, 2021-12-11:
> @eldorel try the latest 0.2.4 build of fan2go and simply put "corsaircpro" in the platform fields. It should pick up your commander every time.

Perfect, thank you very much.

closed 2021-12-11 by eldorel