markusressel/fan2go #75: Not working after standby

Tibladar, 2021-12-16:
**Describe the bug**
After a standby/hibernation the fans go back to automatic control and aren't controlled by fan2go anymore
This can be seen in 100% fan usage and logs get spammed with
`WARNING: PWM of fan1 was changed by third party! Last set PWM value was: 50 but is now: 250`

When fan2go is manually restarted its fine again

**To Reproduce**
Steps to reproduce the behavior:

1. Setup fan2go
2. Hibernate/standby
3. See logs complaining about not being able to set the fan speed

**Expected behavior**
fan2go should be able to set the fan speeds after standby without having to manually restart the process

**Desktop:**

- Distro: Ubuntu 20.04
- `uname -a`: Linux user 5.13.0-22-generic # 22~20.04.1-Ubuntu SMP Tue Nov 9 15:07:24 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: 0.2.5

**Additional context**
I'm currently trying fan2go (before I used fancontrol) and both seem to suffer from the same issue

For fancontrol there's a working fix mentioned in this german wiki: https://wiki.ubuntuusers.de/L%C3%BCftersteuerung/#Luefter-werden-nach-Ruhezustand-Energiesparmodus-nicht-mehr-angesteuert
> In each [loop] interval, the current time is compared with that of the last cycle. If the last cycle was longer than interval * 5 seconds ago, it is assumed that the system was paused and the fans are reinitialized.

Is there a way to implement such a loop or something similar?

markusressel, 2021-12-17:
fan2go should support this, at least it used to. Maybe it got lost in the recent rewrite, but it is definitely a feature that should work without any user intervention or hack. I will check on my system.

markusressel, 2021-12-17:
I have found the issue, it will be fixed with #77.

PR #77 (2021-12-19): Feature/#64 support rounding hwmon drivers
fixes #64 
by computing a list of supported PWM values per fan, and using the value that is closest to the target value dictated by the curve.

fixes #75 
by always making sure the manual PWM mode is enabled when updating the target PWM of a fan.
closed 2021-12-19 by markusressel