markusressel/fan2go #81: error while loading shared libraries

Frambooisier, 2022-01-11:
**Describe the bug**
I'm not sure if its a bug or just a user error, but I copied the fan2go executable to /bin and while trying to use it this error is printed:
```fan2go: error while loading shared libraries: libsensors.so.5: cannot open shared object file: No such file or directory```

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Desktop (please complete the following information):**

- Distro: OpenSuse Tumbleweed KDE
- `uname -a`: Linux SFX 5.15.12-1-default #1 SMP Wed Dec 29 14:50:16 UTC 2021 (375fcb8) x86_64 x86_64 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: 0.4.0

**Additional context**
Add any other context about the problem here.

markusressel, 2022-01-12:
Just a guess:
Maybe you need to install `libsensors4`?

Frambooisier, 2022-01-12:
[image]

markusressel, 2022-01-12:
Did you try?

Frambooisier, 2022-01-12:
it is/was installed

markusressel, 2022-01-12:
Did you try compiling fan2go yourself on the target system? The gosensors integration probably needs a very very exact match of system libraries to work across different distros... only a "real" package could fix this alltogether. I will create an issue in the gosensors project, maybe there is a way to improve this.

Frambooisier, 2022-01-16:
I did this originally
[image]

now, following your suggestion, I tried to build it following instructions but I get an error.
```flap@SFX:~/fan2go> make build
go build -o bin/fan2go main.go
make: go: No such file or directory
make: *** [Makefile:5: build] Error 127
```

markusressel, 2022-01-16:
To build a go program from source, you need to have the go compiler installed on your system. How to get this is different for every system so I didn't include it in the instructions, but there is most definitely a go package for your system.

Frambooisier, 2022-01-16:
Sucess!
[image]

markusressel, 2022-02-03:
Note that while this does fix the issue, you should be aware that **the install might break because of system updates**.
The only way to really fix this is with a "real" package that declares a dependency to the `libsensors` package and can be automatically compiled every time it is updated.

closed 2023-05-11 by markusressel