markusressel/fan2go #88: Runtime Error/sigsegv

dclark16, 2022-02-06:
**Describe the bug**
Just after the initialisation phase fan2go crashes with a sigsegv error.

 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for top_rad...
 INFO  Gathering sensor data for front_rad_bottom...
 INFO  Gathering sensor data for front_rad_top...
 INFO  Gathering sensor data for rear_fan...
 INFO  Gathering sensor data for pump...
 INFO  Loading fan curve data for fan 'top_rad'...
 INFO  Loading fan curve data for fan 'front_rad_bottom'...
 INFO  Loading fan curve data for fan 'front_rad_top'...
 INFO  Loading fan curve data for fan 'pump'...
 INFO  Loading fan curve data for fan 'rear_fan'...
 WARNING  Fan 'front_rad_bottom' has not yet been analyzed, starting initialization sequence...
 WARNING  Fan 'pump' has not yet been analyzed, starting initialization sequence...
 WARNING  Fan 'front_rad_top' has not yet been analyzed, starting initialization sequence...
 WARNING  Fan 'top_rad' has not yet been analyzed, starting initialization sequence...
 WARNING  Fan 'rear_fan' has not yet been analyzed, starting initialization sequence...
 INFO  Start PWM of front_rad_top: 0
 INFO  Max PWM of front_rad_top: 251
 INFO  Starting controller loop for fan 'front_rad_top'
 INFO  Start PWM of top_rad: 0
 INFO  Max PWM of top_rad: 253
 INFO  Starting controller loop for fan 'top_rad'
 INFO  Start PWM of front_rad_bottom: 0
 INFO  Max PWM of front_rad_bottom: 252
 INFO  Starting controller loop for fan 'front_rad_bottom'
 INFO  Start PWM of rear_fan: 0
 INFO  Max PWM of rear_fan: 251
 INFO  Starting controller loop for fan 'rear_fan'
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x55e06ceec744]

goroutine 65 [running]:
github.com/markusressel/fan2go/internal/controller.(*fanController).calculateTargetPwm(0xc000388ee0)
        github.com/markusressel/fan2go/internal/controller/controller.go:284 +0x64
github.com/markusressel/fan2go/internal/controller.(*fanController).UpdateFanSpeed(0xc000388ee0)
        github.com/markusressel/fan2go/internal/controller/controller.go:160 +0x37
github.com/markusressel/fan2go/internal/controller.(*fanController).Run.func3()
        github.com/markusressel/fan2go/internal/controller/controller.go:139 +0xfe
github.com/oklog/run.(*Group).Run.func1({0xc000614080, 0x55e06d198910})
        github.com/oklog/run@v1.1.0/group.go:38 +0x2f
created by github.com/oklog/run.(*Group).Run
        github.com/oklog/run@v1.1.0/group.go:37 +0x22f

**To Reproduce**
Steps to reproduce the behavior:
Remove database file and start fan2go, wait for initialisation, above error will happen. After initialisation fan2go will always produce the above. 

**Expected behavior**
fan2go should start, and take over fan control

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- Distro: Manjaro
- `uname -a`: Linux  5.16.5-1-MANJARO #1 SMP PREEMPT Tue Feb 1 16:57:14 UTC 2022 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.4.1

**Additional context**
Not sure if this is a problem with my motherboard or not as the reason I started looking for another fan controller to replace fancontrol(-gui) was a persistent error when trying to set PWM. Will remove 1 fan at a time and see if I can narrow down to anything specific.

markusressel, 2022-02-06:
Did you possibly set a curve id on fan configuration that does not exist?
I think this case is not (yet) validated so it might cause issues.

dclark16, 2022-02-06:
*Egg on face* 
That's what you get when you try creating fan profiles at 1 in the morning!

Yeah, I found (as you said) a fan with a curve that had a spelling mistake *facepalm*. Being in IT you'd think I would have learnt to check these things, wouldn't you!

Works perfectly now! Thanks for pointing out my stupid mistake Mark!

closed 2022-02-06 by dclark16

markusressel, 2022-02-06:
No worries, I have also added some code to validate this case, as well as some tests for the config validation logic itself.