markusressel/fan2go #100: Error "open .: is a directory"

h-larsson, 2022-03-18:
I try to start `fan2go` with a configuration similar to the default configuration, but just with cpu fan. 
`fan2go` gives me the following output:
```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 WARNING  Cannot read pwm_enable value of cpu
 INFO  Gathering sensor data for cpu...
 INFO  Loading fan curve data for fan 'cpu'...
 WARNING  Fan 'cpu' has not yet been analyzed, starting initialization sequence...
  ERROR   Unable to run initialization sequence on cpu: open .: is a directory
 INFO  Fan controller for fan cpu stopped.
panic: open .: is a directory

goroutine 20 [running]:
github.com/markusressel/fan2go/internal.RunDaemon.func5()
	github.com/markusressel/fan2go/internal/backend.go:118 +0xc6
github.com/oklog/run.(*Group).Run.func1({0xc0000b5380?, 0x559b52daafc8?})
	github.com/oklog/run@v1.1.0/group.go:38 +0x2f
created by github.com/oklog/run.(*Group).Run
	github.com/oklog/run@v1.1.0/group.go:37 +0x22a

```

Configuration file:
```
# https://github.com/markusressel/fan2go
# The path of the database file
dbPath: "/etc/fan2go/fan2go.db"

# Allow the fan initialization sequence to run in parallel for all configured fans
runFanInitializationInParallel: false
# The maximum difference between consecutive RPM measurements to
# consider a fan speed "settled"
maxRpmDiffForSettledFan: 10

# The rate to poll temperature sensors at
tempSensorPollingRate: 200ms
# The number of temp sensor values to keep in a rolling window array
tempRollingWindowSize: 10

# The rate to poll fan RPM input sensors at
rpmPollingRate: 1s
# The number of rpm sensor values to keep in a rolling window array
rpmRollingWindowSize: 10

# The rate to update fan speed targets at
controllerAdjustmentTickRate: 200ms

# A list of fans to control
fans:
  # A user defined ID.
  # Used for logging only
  - id: cpu
    # The type of fan configuration
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: coretemp-isa-0
      # The index of this fan as displayed by `fan2go detect`
      index: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: yes
    # The curve ID (defined above) that should be used to determine the
    # speed of this fan
    curve: cpu_curve
    # (Optional) Override for the lowest PWM value at which the
    # fan will still be able to start rotating.
    # Note: Settings this to a value that is too small
    #       may damage your fans. Use at your own risk!
    startPwm: 150

# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration
    hwmon:
      # The controller platform as displayed by `fan2go detect`, f.ex.:
      # "nouveau", "coretemp" or "it8620" etc.
      platform: coretemp
      # The index of this sensor as displayed by `fan2go detect`
      index: 1

# A list of control curves which can be utilized by fans
# or other curves
curves:
  # A user defined ID, which is used to reference
  # a curve in a fan configuration (see above)
  - id: cpu_curve
    # The type of curve configuration, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Steps to define a section-wise defined speed curve function
      steps:
        # Sensor value -> Speed (0-255)
        - 33: 150
        - 40: 180
        - 85: 255

statistics:
  # Whether to enable the prometheus exporter or not
  enabled: false
  # The port to expose the exporter on
  port: 9000

```

Output of `fan2go detect`:
```
> amdgpu-pci-0d500
  Fans     Index  Label   RPM   PWM  Auto
           1      hwmon4  2008  49   true
  Sensors  Index  Label               Value
           1      edge (temp1_input)  59000

> pch_lewisburg-virtual-0
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  35000

> nvme-pci-010100
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  38850
           2      Sensor 1 (temp2_input)   38850

> dell_smm-isa-0
  Fans     Index  Label   RPM  PWM  Auto 
           1      hwmon3  0    255  false
           2      hwmon3  0    255  false
           3      hwmon3  698  255  false
> coretemp-isa-0
  Sensors  Index  Label                       Value
           1      Package id 0 (temp1_input)  44000
           2      Core 0 (temp2_input)        40000
           3      Core 1 (temp3_input)        36000
           4      Core 2 (temp4_input)        40000
           5      Core 6 (temp8_input)        38000
           6      Core 12 (temp14_input)      38000
           7      Core 13 (temp15_input)      41000
           8      Core 16 (temp18_input)      37000
           9      Core 18 (temp20_input)      38000
           10     Core 19 (temp21_input)      38000
           11     Core 20 (temp22_input)      38000
           12     Core 21 (temp23_input)      38000
           13     Core 24 (temp26_input)      40000
           14     Core 25 (temp27_input)      35000
           15     Core 26 (temp28_input)      38000
           16     Core 27 (temp29_input)      44000
           17     Core 29 (temp31_input)      39000
```

**Desktop **

- Distro: Arch Linux
- `uname -a`: `Linux  5.16.15-arch1-1 #1 SMP PREEMPT Thu, 17 Mar 2022 00:30:09 +0000 x86_64 GNU/Linux`
- `sensors -v`:sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`:0.5.0

markusressel, 2022-03-19:
Hmm, so this error basically says that fan2go tried to measure your fan curve, but failed immediately, because it was unable to set a PWM value for the fan.

I see that you have defined your fan to be platform `coretemp-isa-0`:

```
fans:
  - id: cpu
    hwmon:
      platform: coretemp-isa-0
      index: 1
```

but `fan2go detect` doesn't output any fan inputs for this platform.
You probably should use one of the `dell_smm-isa-0` fan inputs instead.

Nonetheless this is an interesting edge case that is not yet detected, and we should add a test for this to prevent cryptic error messages.

h-larsson, 2022-03-19:
Ah, my bad. Sorry, I messed up `sensor` and `fan`. I'll check on Monday whether it will work after adjusting the configuration.
It may help to check in ` fan2go config validate` whether the `id` corresponds to `Sensors` and/or `Fans` in `fan2go detect`. 
In your example you give in the readme, the `id` has both `Sensors` and `Fans` but here this is not the case, which I missed when I blindly copied your example file ;)

markusressel, 2022-03-19:
Yes thats what I thought too. It might be tricky to validate this correctly though, since it can only work on the target system. This would mean that the successful validation of a config only applies to a given system, but I guess that makes total sense anyway.

h-larsson, 2022-03-21:
Thanks again, now `fan2go` seem to work.
 (I noticed a different issue, unrelated to `fan2go`, namely that the fan can be either at 0 or at max speed and no value in between is accepted right now; seems to be a hardware issue. Thus I cannot fully test `fan2go` right now).

closed 2022-03-21 by h-larsson