markusressel/fan2go #103: Support reading temperature sensors from nvidia graphics cards

peppizza, 2022-04-02:
I would like to control case fans based on the temperature of my graphics card

markusressel, 2022-04-02:
Hey @peppizza,

currently it is unlikely that direct support for reading GPU temps will be included into fan2go.
Although, if lm-sensors supports your GPU, it will of course work out of the box (like with the RX 6700 XT).

You can still "import" data from external sources using either the `file` sensor, or the newly introduced `cmd` sensor.

DanielGibson, 2025-05-23:
I think this feature would be feasible with https://github.com/NVIDIA/go-nvml

Using that should be more efficient than calling `nvidia-settings -t -q GPUCoreTemp` all the time.
It should even allow to control the GPU fans, though I'm not sure how desirable that really is; I personally care most about controlling the case fans based on GPU temperature.

markusressel, 2025-05-23:
@DanielGibson sounds interesting indeed. If this doesn't bring a boatload of system dependencies with it, feel free to give it a shot to implement this, probably as a new type of platform next to `hwmon`, `file` etc.

DanielGibson, 2025-05-26:
I have trouble understanding the fan2go code, like what all the parts do and how they work together and what I need to change for nvml (seems like there's backend-specific code in lots of places)

As far as I understand, so far the actual reading of sensors and setting of fan speeds is file-based - gosensors is only used to figure out the paths, apparently.

Some things that are different/special with NVML:
* It does not use 0-255 PWM values for setting fan speed, but 0-100 percent values.  
  Of course that can be converted easily with `pwm = (perc*255)/100` and `perc = (pwm*100)/255`, but that conversion isn't lossless, due to rounding
* You can read the current speed, it will also be in percent. If you set a speed, it will take some time until the fan actually reports that speed in percent.  
  Despite what the nvml documentation says, it's the current speed in percent, not the configured speed. That's another factor (besides the rounding losses) that could lead to trouble in case fan2go expects the PWM value it gets to be exactly the same as the one that was set.
* Setting the fanspeed is only supported with newer cards (from "Maxwell" on, that was the Geforce 9xx generation from 2014/2015), I think reading it and the temperature is also supported on older cards
* With newer drivers (v565+) one can also get the fan speed in RPM. go-nvml doesn't expose that (yet?) but in my nvml testprogram I hacked some cgo code to retrieve it, if it's available.
* As mentioned before, a device handle is needed to request speed etc. fan2go would have to open the device and get that handle at some point, and then store it somewhere where the NvidiaFan and NvidiaSensor can access it.  
  If I understand it correctly, with the current (hwmon/file) code having a stable handle is no issue because it only stores the sysfs-paths anyway and opens those files for each get/set.

This is my little test program: [go-nvml-test.zip](https://github.com/user-attachments/files/20433841/go-nvml-test.zip)

markusressel, 2025-05-26:
@DanielGibson 

> As far as I understand, so far the actual reading of sensors and setting of fan speeds is file-based - gosensors is only used to figure out the paths, apparently.

Yes, mostly, but thats not a hard requirement. F.ex. the `cmd` fan implementation does not require any file path. The config can be unique for each platform, so as long as there is a way to uniquely identify the device handle on each boot, that could be part of the `nvmi` fan config instead of a file path.

> * It does not use 0-255 PWM values for setting fan speed, but 0-100 percent values.
>   Of course that can be converted easily with `pwm = (perc*255)/100` and `perc = (pwm*100)/255`, but that conversion isn't lossless, due to rounding

We already had this issue in the past, and the `pwmMap` was used to get around that. However, if we implement a "native" nvmi fan, we can simply code this into its implementation. The lossyness of that conversion needs to be considered, but should be workable.

> * You can read the current speed, it will also be in percent. If you set a speed, it will take some time until the fan actually reports that speed in percent.

Yeah this also has already led to problems in the past, because fan2go's sanity checks are thrown off by that. I have considered implementing a system to disable these sanity checks for certain platforms in the past already, but there has not been the time or need to actually do it (yet).

>   Despite what the nvml documentation says, it's the current speed in percent, not the configured speed. That's another factor (besides the rounding losses) that could lead to trouble in case fan2go expects the PWM value it gets to be exactly the same as the one that was set.

Yes, see the previous answer.

> * Setting the fanspeed is only supported with newer cards (from "Maxwell" on, that was the Geforce 9xx generation from 2014/2015), I think reading it and the temperature is also supported on older cards

That should also be fine. There is a "capability" system already present in fan2go, which allows you to specify if a fan can be set or is read only. I have never experimented with a "read-only" fan though - as that somewhat defeats the point of fan2go - so there will probably be bugs.

> * With newer drivers (v565+) one can also get the fan speed in RPM. go-nvml doesn't expose that (yet?) but in my nvml testprogram I hacked some cgo code to retrieve it, if it's available.

Again, the capabilities described above should be used to indicate this to fan2go.

> * As mentioned before, a device handle is needed to request speed etc. fan2go would have to open the device and get that handle at some point, and then store it somewhere where the NvidiaFan and NvidiaSensor can access it.

I think this can be done inside of the NvmiFan implementation, we can figure out the details in a draft PR.

>   If I understand it correctly, with the current (hwmon/file) code having a stable handle is no issue because it only stores the sysfs-paths anyway and opens those files for each get/set.

I am not sure I fully understand what the problem is. The paths that are present in the `HwMonFan` are determined on each start of fan2go based on the configuration of a fan. These paths are not changed afterwards, so if the path of a fan would change during runtime, it would probably cause a lot of problems. I have not seen this happen in the real world though.

PR #373 (2025-06-03): Support nvidia GPUs using go-nvml
I think most of the work is done, but one crucial part is missing: I'm not sure how to get nvml.Device into NvidiaFan and NvidiaSensor

Fixes #103
closed 2025-06-03 by markusressel