markusressel/fan2go #110: How to set you up?

z36r2850o, 2022-04-29:
Hello. I can't figure out how to set up fan2go. I am facing an error when I try to execute the `sudo fan2go. Here is what I get in response:

```
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for cpu...
 INFO  Loading fan curve data for fan 'cpu'...
 WARNING  Fan 'cpu' has not yet been analyzed, starting initialization sequence...
 WARNING  Cannot read RPM value of fan cpu: read /sys/class/hwmon/hwmon2/fan1_input: no such device or address
 WARNING  Cannot read RPM value of fan cpu: read /sys/class/hwmon/hwmon2/fan1_input: no such device or address
 WARNING  Cannot read RPM value of fan cpu: read /sys/class/hwmon/hwmon2/fan1_input: no such device or address
 WARNING  Cannot read RPM value of fan cpu: read /sys/class/hwmon/hwmon2/fan1_input: no such device or address
```

A little information.

`fan2go detect`:

```
> asus-isa-0
  Fans     Index  Label    RPM  PWM  Auto
           1      cpu_fan  0    0    true
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  43050

> coretemp-isa-0
  Sensors  Index  Label                       Value
           1      Package id 0 (temp1_input)  45000
           2      Core 0 (temp2_input)        45000
           3      Core 1 (temp3_input)        44000

> acpitz-acpi-0
  Sensors  Index  Label                 Value
           1      hwmon1 (temp1_input)  43000
```

Configuration file:

```
# The path of the database file
dbPath: "/etc/fan2go/fan2go.db"

# Allow the fan initialization sequence to run in parallel for all configured fans
runFanInitializationInParallel: false
# The maximum difference between consecutive RPM measurements to
# consider a fan speed "settled"
maxRpmDiffForSettledFan: 10

# The rate to poll temperature sensors at
tempSensorPollingRate: 200ms
# The number of temp sensor values to keep in a rolling window array
tempRollingWindowSize: 10

# The rate to poll fan RPM input sensors at
rpmPollingRate: 1s
# The number of rpm sensor values to keep in a rolling window array
rpmRollingWindowSize: 10

# The rate to update fan speed targets at
controllerAdjustmentTickRate: 200ms

# A list of fans to control
fans:
  # A user defined ID.
  # Used for logging only
  - id: cpu
    # The type of fan configuration
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: asus-isa-0
      # The index of this fan as displayed by `fan2go detect`
      index: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: yes
    # The curve ID (defined above) that should be used to determine the
    # speed of this fan
    curve: cpu_curve
    # (Optional) Override for the lowest PWM value at which the
    # fan will still be able to start rotating.
    # Note: Settings this to a value that is too small
    #       may damage your fans. Use at your own risk!
    startPwm: 30

#  - id: in_front
#    hwmon:
#      platform: it8620
#      index: 4
#    neverStop: yes
#    curve: case_avg_curve

#  - id: out_back
#    hwmon:
#      platform: it8620
#      index: 5
#    neverStop: yes
#    curve: case_avg_curve

# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration
    hwmon:
      # The controller platform as displayed by `fan2go detect`, f.ex.:
      # "nouveau", "coretemp" or "it8620" etc.
      platform: asus-isa-0
      # The index of this sensor as displayed by `fan2go detect`
      index: 1

#  - id: mainboard
#    hwmon:
#      platform: it8620
#      index: 3

#  - id: sata_ssd
#    hwmon:
#      platform: acpitz
#      index: 1

# A list of control curves which can be utilized by fans
# or other curves
curves:
  # A user defined ID, which is used to reference
  # a curve in a fan configuration (see above)
  - id: cpu_curve
    # The type of curve configuration, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Steps to define a section-wise defined speed curve function
      steps:
        # Sensor value -> Speed (0-255)
        - 40: 0
        - 50: 50
        - 80: 255

#  - id: mainboard_curve
#    linear:
#      sensor: mainboard
      # Sensor input value at which the curve is at minimum speed
#      min: 40
      # Sensor input value at which the curve is at maximum speed
#      max: 80

#  - id: ssd_curve
#    linear:
#      sensor: sata_ssd
#      min: 40
#      max: 70

# - id: case_avg_curve
#    function:
      # Type of aggregation function to use, on of: minimum | maximum | average
#      type: average
      # A list of curve IDs to use
#      curves:
#        - cpu_curve
#        - mainboard_curve
#        - ssd_curve

#statistics:
  # Whether to enable the prometheus exporter or not
#  enabled: false
  # The port to expose the exporter on
#  port: 9000
```

`sudo fan2go config validate`

```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 SUCCESS  Config looks good! :)
```

I am using archlinux and fan2go built from the AUR. Please tell me how to fix this?

markusressel, 2022-04-29:
Hmm looks like fan2go cannot read the RPM of the fan, does the fan connector and the fan itself support it? Can you see RPM values when running `sensors`?

z36r2850o, 2022-04-29:
`sensors`

```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:     2100 RPM
temp1:        +45.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +48.0°C  (high = +87.0°C, crit = +105.0°C)
Core 0:        +48.0°C  (high = +87.0°C, crit = +105.0°C)
Core 1:        +47.0°C  (high = +87.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +45.0°C  (crit = +108.0°C)
```

z36r2850o, 2022-04-30:
Appeared after reboot.

```
> asus-isa-0
  Fans     Index  Label    RPM  PWM  Auto
           1      cpu_fan  85   85   true
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  50050

> coretemp-isa-0
  Sensors  Index  Label                       Value
           1      Package id 0 (temp1_input)  49000
           2      Core 0 (temp2_input)        48000
           3      Core 1 (temp3_input)        49000

> acpitz-acpi-0
  Sensors  Index  Label                 Value
           1      hwmon1 (temp1_input)  50000
```

markusressel, 2022-05-01:
I guess something goes wrong when fan2go tries to set the `pwm_enable` value, but the error is not logged correctly :( I will try to make some adjustments surrounding this.

markusressel, 2022-05-21:
@z36r2850o I have made some changes for #116 as well, can you retest?

markusressel, 2022-05-29:
I am closing this because of inactivity. Feel free to reopen if necessary :)

closed 2022-05-29 by markusressel