markusressel/fan2go #116: ERROR: Unable to set Fan Mode

tmiland, 2022-05-13:
**Describe the bug**
A clear and concise description of what the bug is:

```bash
sudo fan2go
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 WARNING  Cannot read pwm_enable value of in_front
 INFO  Gathering sensor data for in_front...
 WARNING  Cannot read pwm_enable value of cpu_pump
 WARNING  Cannot read pwm_enable value of out_rad1
 WARNING  Cannot read pwm_enable value of out_rad3
 INFO  Gathering sensor data for out_rad1...
 INFO  Gathering sensor data for out_rad3...
 INFO  Gathering sensor data for cpu_pump...
 WARNING  Cannot read pwm_enable value of in_bottom
 INFO  Gathering sensor data for in_bottom...
 WARNING  Cannot read pwm_enable value of out_rad2
 INFO  Gathering sensor data for out_rad2...
 WARNING  Cannot read pwm_enable value of in_back
 INFO  Gathering sensor data for in_back...
 INFO  Loading fan curve data for fan 'out_rad2'...
 INFO  Loading fan curve data for fan 'in_back'...
 INFO  Loading fan curve data for fan 'in_bottom'...
 INFO  Loading fan curve data for fan 'in_front'...
 INFO  Loading fan curve data for fan 'out_rad1'...
 INFO  Loading fan curve data for fan 'out_rad3'...
 INFO  Loading fan curve data for fan 'cpu_pump'...
  ERROR   Unable to set Fan Mode of 'out_rad2' to "1": open /sys/class/hwmon/hwmon3//pwm4_enable: permission denied
  ERROR   Unable to set Fan Mode of 'out_rad2' to "0": open /sys/class/hwmon/hwmon3//pwm4_enable: permission denied
  ERROR   Unable to set Fan Mode of 'out_rad1' to "1": open /sys/class/hwmon/hwmon3//pwm3_enable: permission denied
  ERROR   Unable to set Fan Mode of 'out_rad1' to "0": open /sys/class/hwmon/hwmon3//pwm3_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_front' to "1": open /sys/class/hwmon/hwmon3//pwm8_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_front' to "0": open /sys/class/hwmon/hwmon3//pwm8_enable: permission denied
  ERROR   Unable to set Fan Mode of 'out_rad3' to "1": open /sys/class/hwmon/hwmon3//pwm5_enable: permission denied
  ERROR   Unable to set Fan Mode of 'out_rad3' to "0": open /sys/class/hwmon/hwmon3//pwm5_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_bottom' to "1": open /sys/class/hwmon/hwmon3//pwm6_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_bottom' to "0": open /sys/class/hwmon/hwmon3//pwm6_enable: permission denied
  ERROR   Unable to set Fan Mode of 'cpu_pump' to "1": open /sys/class/hwmon/hwmon3//pwm2_enable: permission denied
  ERROR   Unable to set Fan Mode of 'cpu_pump' to "0": open /sys/class/hwmon/hwmon3//pwm2_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_back' to "1": open /sys/class/hwmon/hwmon3//pwm7_enable: permission denied
  ERROR   Unable to set Fan Mode of 'in_back' to "0": open /sys/class/hwmon/hwmon3//pwm7_enable: permission denied

```

**To Reproduce**
Steps to reproduce the behavior:

```bash
git clone https://github.com/markusressel/fan2go.git
cd fan2go
make build
sudo cp ./bin/fan2go /usr/bin/fan2go
sudo chmod ug+x /usr/bin/fan2go
```

**Desktop:**

- Distro: lsb_release -a
```bash
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
[... 98 lines cut ...]
-r--r--r-- 1 root root 4.0K May 12 09:37 temp2_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp2_min
-r--r--r-- 1 root root 4.0K May 12 09:37 temp3_input
-r--r--r-- 1 root root 4.0K May 12 09:37 temp3_label
-r--r--r-- 1 root root 4.0K May 12 09:37 temp3_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp3_min
-r--r--r-- 1 root root 4.0K May 12 09:37 temp4_input
-r--r--r-- 1 root root 4.0K May 12 09:37 temp4_label
-r--r--r-- 1 root root 4.0K May 12 09:37 temp4_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp4_min
-r--r--r-- 1 root root 4.0K May 12 09:37 temp5_input
-r--r--r-- 1 root root 4.0K May 12 09:37 temp5_label
-r--r--r-- 1 root root 4.0K May 12 09:37 temp5_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp5_min
-r--r--r-- 1 root root 4.0K May 12 09:37 temp6_input
-r--r--r-- 1 root root 4.0K May 12 09:37 temp6_label
-r--r--r-- 1 root root 4.0K May 12 09:37 temp6_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp6_min
-r--r--r-- 1 root root 4.0K May 12 09:37 temp7_input
-r--r--r-- 1 root root 4.0K May 12 09:37 temp7_label
-r--r--r-- 1 root root 4.0K May 12 09:37 temp7_max
-r--r--r-- 1 root root 4.0K May 12 09:37 temp7_min
-rw-r--r-- 1 root root 4.0K May 12 09:37 uevent
 ```
</p></details>

markusressel, 2022-05-15:
Seems like there is a double `/` in the path.
Can you read the file if you try it manually (using `cat` or something similar)?

tmiland, 2022-05-16:
If you see in the truncated info above, the `pwm*_enable` files are missing. 

I'm using this kernel module [nct6687d](https://github.com/Fred78290/nct6687d) on a MSI MPG B550 GAMING PLUS mainboard.

Have yet to figure out why the files are missing...

markusressel, 2022-05-16:
I missed that, my bad.
Can you still control the fans if you manually write values to the write values to the `pwm*` files?
Can you control them using other tools?

tmiland, 2022-05-17:
Yes, using fancontrol works.

markusressel, 2022-05-19:
I took a quick look at the implementation of fancontrol and it seems like they just ignore the `pwm_enable` file and value if it doesn't exist. I will implement a similar approach.

markusressel, 2022-05-19:
Please try again with the latest dev version and let me know :nerd_face:

tmiland, 2022-05-21:
Okay, tested latest dev just now:

```bash
sudo fan2go
███████  █████  ███    ██ ██████   ██████   ██████  
██      ██   ██ ████   ██      ██ ██       ██    ██ 
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██ 
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██ 
██      ██   ██ ██   ████ ███████  ██████   ██████  

 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for in_bottom...
 INFO  Gathering sensor data for out_back...
 INFO  Gathering sensor data for out_top2...
 INFO  Gathering sensor data for out_top1...
 INFO  Gathering sensor data for cpu...
 INFO  Gathering sensor data for gpu...
 INFO  Gathering sensor data for out_top3...
 INFO  Gathering sensor data for in_front...
 INFO  Loading fan curve data for fan 'out_back'...
 INFO  Loading fan curve data for fan 'in_bottom'...
 INFO  Loading fan curve data for fan 'out_top2'...
 INFO  Loading fan curve data for fan 'cpu'...
 INFO  Loading fan curve data for fan 'out_top1'...
 INFO  Loading fan curve data for fan 'gpu'...
 INFO  Loading fan curve data for fan 'out_top3'...
 INFO  Loading fan curve data for fan 'in_front'...
 INFO  Start PWM of gpu: 0
 INFO  Max PWM of gpu: 255
 INFO  Starting controller loop for fan 'gpu'
 INFO  Start PWM of out_top2: 0
 INFO  Max PWM of out_top2: 184
 INFO  Starting controller loop for fan 'out_top2'
 INFO  Start PWM of cpu: 30
 INFO  Max PWM of cpu: 233
 INFO  Starting controller loop for fan 'cpu'
 INFO  Start PWM of in_bottom: 0
 INFO  Max PWM of in_bottom: 10
 INFO  Starting controller loop for fan 'in_bottom'
 INFO  Start PWM of out_back: 0
 INFO  Max PWM of out_back: 159
 INFO  Starting controller loop for fan 'out_back'
 INFO  Start PWM of in_front: 0
 INFO  Max PWM of in_front: 18
 INFO  Starting controller loop for fan 'in_front'
 WARNING  PWM of out_top2 was changed by third party! Last set PWM value was: 37 but is now: 35
 INFO  Start PWM of out_top1: 0
 INFO  Max PWM of out_top1: 234
 INFO  Starting controller loop for fan 'out_top1'
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 32 but is now: 31
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 33 but is now: 31
 INFO  Start PWM of out_top3: 0
 INFO  Max PWM of out_top3: 215
 INFO  Starting controller loop for fan 'out_top3'
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 33 but is now: 31
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 33 but is now: 31
 WARNING  PWM of out_back was changed by third party! Last set PWM value was: 124 but is now: 110
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 35 but is now: 34
 WARNING  PWM of cpu was changed by third party! Last set PWM value was: 36 but is now: 34
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
 WARNING  PWM of out_back was changed by third party! Last set PWM value was: 124 but is now: 110
 WARNING  PWM of in_front was changed by third party! Last set PWM value was: 134 but is now: 125
 WARNING  PWM of out_top1 was changed by third party! Last set PWM value was: 32 but is now: 31
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
 WARNING  PWM of out_back was changed by third party! Last set PWM value was: 124 but is now: 110
 WARNING  PWM of in_front was changed by third party! Last set PWM value was: 134 but is now: 125
 WARNING  PWM of in_bottom was changed by third party! Last set PWM value was: 59 but is now: 40
[... 286 lines cut ...]
 INFO  Trying to restore fan settings for cpu...
 INFO  Fan controller for fan cpu stopped.
 WARNING  PWM of out_top2 was changed by third party! Last set PWM value was: 9 but is now: 10
 INFO  Fan controller for fan in_bottom stopped.
 INFO  Stopping fan controller for fan out_top3...
 INFO  Trying to restore fan settings for out_top3...
 INFO  Fan controller for fan out_back stopped.
 INFO  Stopping fan controller for fan out_top1...
 INFO  Trying to restore fan settings for out_top1...
 INFO  Fan controller for fan out_top3 stopped.
 WARNING  PWM of out_top2 was changed by third party! Last set PWM value was: 9 but is now: 10
 INFO  Fan controller for fan out_top1 stopped.
 WARNING  PWM of out_top2 was changed by third party! Last set PWM value was: 9 but is now: 10
 WARNING  PWM of out_top2 was changed by third party! Last set PWM value was: 9 but is now: 10
 INFO  Stopping fan controller for fan out_top2...
 INFO  Trying to restore fan settings for out_top2...
 INFO  Fan controller for fan out_top2 stopped.
 INFO  Done.
```

Bug is fixed :+1: 

There's alot of log spam though, and there's no third party changing the pwm value (fancontrol is stopped)

Thanks!

markusressel, 2022-05-21:
Cool!

Not sure where the warnings are coming from :thinking: 
On my three personal machines only one prints out a similar warning now and then, but the mainboard really does change it, so its actually correct.

I will close this issue, if you cannot figure out the cause of the warnings and think it is a bug, just open new issue :)

closed 2022-05-21 by markusressel