markusressel/fan2go #119: One of the case fans is not detected

Shished, 2022-05-15:
**Describe the bug**
My PC has 2 case fans - front intake and back outflow but `fan2go detect` only shows front fan (as well as a CPU fan)

```
$ fan2go detect --no-color --no-style
> nct6798-isa-0
  Fans     Index  Label   RPM  PWM  Auto 
           1      hwmon3  0    3    false
           2      hwmon3  0    3    false
  Sensors  Index  Label                                   Value 
           1      SYSTIN (temp1_input)                    31000 
           2      CPUTIN (temp2_input)                    36500 
           3      AUXTIN0 (temp3_input)                   25500 
           4      AUXTIN1 (temp4_input)                   7000  
           5      AUXTIN2 (temp5_input)                   -13000
           6      AUXTIN3 (temp6_input)                   25000 
           7      PECI Agent 0 Calibration (temp7_input)  36500 
           8      PCH_CHIP_CPU_MAX_TEMP (temp8_input)     0     
           9      PCH_CHIP_TEMP (temp9_input)             0     
           10     PCH_CPU_TEMP (temp10_input)             0
```
Where Index #1 is a front fan and #2 is a CPU fan

It is detected by `lm_sensors`. Case and CPU fans are shown as `fan1` and `fan2` respectively, other fan is shown as `fan4`.

**To Reproduce**
Steps to reproduce the behavior:

1. run `fan2go detect`
2. Scroll down to `nct6798`
2. See 2 fans instead of 3

**Expected behavior**
`fan2go detect` shows all available fans and allows to control them.

**Desktop (please complete the following information):**

- Distro: [Arch Linux]
- `uname -a`: `Linux linux 5.17.7-zen1-1-zen #1 ZEN SMP PREEMPT Thu, 12 May 2022 18:55:52 +0000 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0+git with libsensors version 3.6.0+git`
- `fan2go version`: `dev`

markusressel, 2022-05-15:
Can you post the output of `sensors`?

Are there any log messages from fan2go hinting at a problem?

Shished, 2022-05-15:
`fan4` is that fan. fan2go shows no errors
```
nct6798-isa-0290
Adapter: ISA adapter
Vcore:
  in0_input: 0.552
  in0_min: 0.200
  in0_max: 1.504
  in0_alarm: 0.000
  in0_beep: 0.000
PCH Volt:
  in1_input: 1.008
  in1_min: 0.000
  in1_max: 0.000
  in1_alarm: 1.000
  in1_beep: 0.000
AVSB:
  in2_input: 3.424
  in2_min: 3.136
  in2_max: 3.472
  in2_alarm: 0.000
  in2_beep: 0.000
3VCC:
  in3_input: 3.376
  in3_min: 3.136
  in3_max: 3.472
  in3_alarm: 0.000
  in3_beep: 0.000
+12V:
  in4_input: 12.288
  in4_min: 11.424
  in4_max: 12.576
  in4_alarm: 0.000
  in4_beep: 0.000
+5V:
  in6_input: 5.040
  in6_min: 4.760
  in6_max: 5.240
  in6_alarm: 0.000
  in6_beep: 0.000
3.3V:
  in7_input: 3.424
  in7_min: 3.136
  in7_max: 3.472
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 3.200
  in8_min: 3.136
  in8_max: 3.472
  in8_alarm: 0.000
  in8_beep: 0.000
CPU VTT:
  in9_input: 0.536
  in9_min: 0.000
  in9_max: 0.000
  in9_alarm: 1.000
  in9_beep: 0.000
Вентилятор Передний:
  fan1_input: 0.000
  fan1_min: 0.000
  fan1_alarm: 0.000
  fan1_beep: 0.000
  fan1_pulses: 2.000
Вентилятор CPU:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_alarm: 0.000
  fan2_beep: 0.000
  fan2_pulses: 2.000
Вентилятор Задний:
  fan4_input: 840.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 0.000
  fan4_pulses: 2.000
beep_enable:
  beep_enable: 0.000
```

Sensors.d conf only renames this fan
`    label fan4 "Вентилятор Задний"`

markusressel, 2022-05-15:
Hmm interesting, you seem to have `fan1`, `fan2` and `fan4` but `fan3` is missing.
The current code assumes all fan inputs of a chip are increasing by one, and this logic is failing here.
I will have to investigate where I can get the correct index from.

markusressel, 2022-05-16:
@Shished please try again with the latest dev version and let me know if it is fixed.

Shished, 2022-05-16:
Latest update has fixed the problem

```
> nct6798-isa-0
  Fans     Index  Label   RPM  PWM  Auto 
           1      hwmon3  0    21   false
           2      hwmon3  0    21   false
           4      hwmon3  832  178  true
```

Shished, 2022-05-16:
I'm still having problems with it and the app.
fan2go spams log with this messages:

`PWM of back was changed by third party! Last set PWM value was: 201 but is now: 158`

PWM values are different in every line.

markusressel, 2022-05-16:
You mean specifically the fan with index `4` is not controlled correctly?
Or also others?

markusressel, 2022-05-16:
@Shished please try again with the latest version :crossed_fingers:

Shished, 2022-05-17:
All fans are working properly after the latest update. Thank you.

markusressel, 2022-05-17:
Great! Closing this then.
Feel free to open another issue if anything comes up.

PR #122 (2022-05-16): increment index, even if pwm output cant be found
maybe fixes #119
closed 2022-05-17 by markusressel