markusressel/fan2go #140: Not printing correct fan index

thomasdorch, 2022-06-23:
Hi, I am using fan2go on Debian 12 and had no issues getting it installed. I am trying to set up the yaml config but it doesn't seem to be detecting the correct indices for my fans, though every other sensor has a non-zero index aside from the fans.

This is part of the output of `fan2go detect`:

```
> nct6798-isa-0
  Fans     Index  Label   RPM  PWM  Auto 
           0      hwmon7  0    137  false
           0      hwmon7  590  93   false
           0      hwmon7  762  90   false
           0      hwmon7  511  53   false
           0      hwmon7  547  101  false

```

And this is what I get from `sensors`:
```
nct6798-isa-0290
Adapter: ISA adapter
fan1:                        0 RPM  (min =    0 RPM)
fan2:                      604 RPM  (min =    0 RPM)
fan3:                      774 RPM  (min =    0 RPM)
fan4:                      512 RPM  (min =    0 RPM)
fan5:                      541 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
```

Any idea what might be causing this?

Debian 12, kernel 5.18.0-1
Motherboard: Asus ProArt B550

Thank you.

markusressel, 2022-06-24:
I can confirm this on one of my machines, will investigate.

grindzohr, 2022-06-24:
Can confirm this also happens with _nct6792_ using kernel module _nct6775_ if that is of any use.

`fan2go detect`:
```
...
> nct6792-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           0      hwmon2  897   89   false
           0      hwmon2  1161  89   false
           0      hwmon2  1037  89   false
           0      hwmon2  0     127  false
           0      hwmon2  0     127  false
...
```

'sensors':
```
...
nct6792-isa-0290
Adapter: ISA adapter
...
fan1:                   911 RPM  (min =    0 RPM)
fan2:                  1145 RPM  (min =    0 RPM)
fan3:                  1032 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
...
```

Distro: endeavouros
Kernel: 5.18.5-zen1-1-zen
Motherboard: ASRock Fatal1ty B450 Gaming-ITX/ac

markusressel, 2022-06-25:
Should be fixed, please try again :)

closed 2022-06-25 by markusressel