markusressel/fan2go #144: Is it posible to control fan speed depend on two temp sensors?

stalkerg, 2022-07-19:
Subj. 
Very often my CPU is cold, and GPU is hot, but only my water cool system sucks fresh air into the case.
If GPU has a high temp my water cool fans also should increase speeds. Basically, it's should calculate separately, but the final speed should be a max of this two.

nawuko, 2022-07-19:
You can create a custom curve for that, i have the same scenario as you and use a custom curve.

modified readme example:

```yaml
curves:
  - id: case_avg_curve
    function:
      # Type of aggregation function to use, one of: minimum | maximum | average | delta
      type: maximum
      # A list of curve IDs to use
      curves:
        - cpu_curve
        - gpu_curve
```

stalkerg, 2022-07-19:
Thanks! I will try. If something will wrong, I will reopen this ticket.

closed 2022-07-19 by stalkerg