markusressel/fan2go #147: OpenRC init script support.

stalkerg, 2022-07-27:
To support Gentoo, we need OpenRC support as well as systemd. 
This code works perfectly for me:
```
#!/sbin/openrc-run
# Copyright 1999-2019 Gentoo Authors
# Distributed under the terms of the GNU General Public License v2

command="/usr/bin/fan2go"
command_background=true
required_files="/etc/fan2go/fan2go.yaml"
pidfile="/var/run/fan2go.pid"

depend() {
        need localmount
        use lm_sensors
}
```

I also have an ebuild + archive with all dependencies, but it's not finished, and anyway, I should push it into Gentoo.

stalkerg, 2022-07-27:
Also, for Gentoo will be good to have `vendor` folder to avoid creating external dependencies archive. 
https://stackoverflow.com/questions/61547433/what-are-the-benefits-of-having-a-vendor-folder

markusressel, 2022-07-27:
I cannot provide support for Gentoo myself, since I don't use it and cannot invest any time into it, but if you want to maintain a package for it, feel free to do so.

The project is currently still in quite a fragile state, so I don't think committing dependencies is helping much at this point, but its definitely something to keep in mind when the project is in a more mature state.

markusressel, 2022-07-28:
1. I dont disagree. Feel free to help out by opening a PR! :sunglasses:
2. The fact is: I dont use it and wont in the forseeable future. If you want support, add it to your gentoo package, or open a PR to add it here, but dont expect me to fix it if it ever breaks ☺️

stalkerg, 2022-07-28:
Ok, got it! Thanks!

closed 2022-07-28 by stalkerg