markusressel/fan2go #151: Stuck at starting controller loop

diasgst, 2022-08-14:
**Describe the bug**

fan2go initialization would be stuck on starting controller loop indefinitely, no matter how long

**To Reproduce**
Steps to reproduce the behavior:

1. `$ sudo fan2go -v`
2. Wait until initialization reach starting controller loop
3. Starting controller loop is stuck indefinitely

```console
 INFO  Start PWM of gpu: 0
 INFO  Max PWM of gpu: 250
 INFO  Starting controller loop for fan 'gpu'
  DEBUG   Setting Fan PWM of 'gpu' to 23 ...
  DEBUG   Setting Fan PWM of 'gpu' to 22 ...
  DEBUG   Setting Fan PWM of 'gpu' to 21 ...
  DEBUG   Setting Fan PWM of 'gpu' to 20 ...
  DEBUG   Setting Fan PWM of 'gpu' to 19 ...
  DEBUG   Setting Fan PWM of 'gpu' to 18 ...
  DEBUG   Setting Fan PWM of 'gpu' to 17 ...
  DEBUG   Setting Fan PWM of 'gpu' to 16 ...
  DEBUG   Setting Fan PWM of 'gpu' to 15 ...
  DEBUG   Setting Fan PWM of 'gpu' to 14 ...
  DEBUG   Setting Fan PWM of 'gpu' to 13 ...
  DEBUG   Setting Fan PWM of 'gpu' to 12 ...
  DEBUG   Setting Fan PWM of 'gpu' to 11 ...
  DEBUG   Setting Fan PWM of 'gpu' to 10 ...
  DEBUG   Setting Fan PWM of 'gpu' to 9 ...
  DEBUG   Setting Fan PWM of 'gpu' to 8 ...
  DEBUG   Setting Fan PWM of 'gpu' to 7 ...
  DEBUG   Setting Fan PWM of 'gpu' to 6 ...
  DEBUG   Setting Fan PWM of 'gpu' to 5 ...
  DEBUG   Setting Fan PWM of 'gpu' to 4 ...
  DEBUG   Setting Fan PWM of 'gpu' to 3 ...
  DEBUG   Setting Fan PWM of 'gpu' to 2 ...
  DEBUG   Setting Fan PWM of 'gpu' to 1 ...
  DEBUG   Setting Fan PWM of 'gpu' to 0 ...

```
```console
$ fan2go detect --no-color --no-style
> it8688-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           1      hwmon1  1662  117  true 
           2      hwmon1  0     1    false
           3      hwmon1  0     70   true 
  Sensors  Index  Label                 Value
           1      hwmon1 (temp1_input)  50000
           2      hwmon1 (temp2_input)  55000
           3      hwmon1 (temp3_input)  54000
           4      hwmon1 (temp4_input)  22000
           5      hwmon1 (temp5_input)  53000
           6      hwmon1 (temp6_input)  53000

```
```console
$ sensors
it8688-isa-0a40
Adapter: ISA adapter
CPU Vcore:     1.01 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
VCORE SOC:     1.20 V  (min =  +0.00 V, max =  +3.06 V)
CPU VDDP:    912.00 mV (min =  +0.00 V, max =  +3.06 V)
DRAM A/B:      1.40 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.17 V  
CPU_FAN:     1683 RPM  (min =    0 RPM)
SYS_FAN1:       0 RPM  (min =    0 RPM)
SYS_FAN2:       0 RPM  (min =    0 RPM)
temp1:        +50.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +56.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
AMDSI:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +21.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +53.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
temp6:        +53.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor

```

**Desktop:**

- Distro: [Fedora 36]
- `uname -a`: `Linux workstation 5.18.16-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Aug 3 15:44:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0 with libsensors version 3.6.0`
- `fan2go version`: `dev`

markusressel, 2022-08-14:
Why do you think its stuck? Looks like its changing the GPU pwm value, which is exactly what the controller loop is supposed to do (if a change is necessary).

diasgst, 2022-08-14:
Because i accidentally left it running all night, and  it didn't progress any further than that

markusressel, 2022-08-14:
That only means the controller thought that 0 was the best value all the time, which sounds like a configuration issue to me. Have you tried different configurations to force the fan speed to go up instead of down?

diasgst, 2022-08-14:
Here is my current configuration

```console
dbPath: "/etc/fan2go/fan2go.db"
fans:
  - id: gpu
    hwmon:
      platform: it8688-isa-0
      index: 2
    neverStop: false
    maxPwm: 250
    curve: gpu_curve

curves:
  - id: gpu_curve
    # The type of the curve
    linear:
      # The sensor ID to use as a temperature input
      sensor: nvidia_fan
      # Steps to define a section-wise defined speed curve function.
      steps:
       - 40: 0
       - 55: 85
       - 60: 110
       - 65: 145
       - 74: 165
       - 84: 195
       - 90: 245

sensors:
  - id: nvidia_fan
    cmd:
      exec: /usr/bin/nvidia-settings
      args: [ '-q', 'gpucoretemp', '-t' ]

```
I also tried adding `maxRpmDiffForSettledFan: 30` , but it seems to be stuck at setting pwm to 83

markusressel, 2022-08-14:
The `maxRpmDiffForSettledFan` parameter only affects the initial measurement of a fan, its not used after that. The current config layout isn't making this super clear, I will add a comment to clarify this.

About the configuration:  Looks good so far.

Can you post the output of
1. `fan2go sensor --id nvidia_fan`
2. `sudo /usr/bin/nvidia-settings -q gpucoretemp -t`

diasgst, 2022-08-15:
Here is the output as requested

```console
$ fan2go sensor --id nvidia_fan
36
```
```console
$ sudo /usr/bin/nvidia-settings -q gpucoretemp -t
39
```

markusressel, 2022-08-15:
fan2go expects the value to be an integer number in milli-units.

In this case, the value should be `39000` instead of `39` (or `36000` instead of `36`), assuming your GPU isn't at freezing temperatures :cold_face: :sweat_smile: So to make it work you would have to add some calculation to the command, which will multiply it by `1000` before its printed to the console.

I am sorry that the README suggests this command but it doesn't actually work. It was only meant as an example. I never used it myself, because I don't currently own an nvidia GPU. If you come up with a fix that is easy to integrate, I will update the README asap. I will also make it more clear that the value needs to be in milli-units for all of the sensors.

diasgst, 2022-08-17:
Thanks for clearing things out, I have done some tests myself. But it seems like the config file doesn't take literal bash commands, so additional args must be configured for the multiplication of `1000`.

I have successfully run fan2go with the following config:
```console
$ cat /etc/fan2go/fan2go.yaml
dbPath: "/etc/fan2go/fan2go.db"
fans:
  - id: gpu
    hwmon:
      platform: it8628-isa-0
      index: 2
    neverStop: false
    curve: gpu_curve

curves:
  - id: gpu_curve
    linear:
      sensor: nvidia_fan
      steps:
       - 50: 0
       - 55: 85
       - 60: 110
       - 65: 145
       - 70: 175
       - 75: 205
       - 85: 235
       - 87: 250

sensors:
  - id: nvidia_fan
    cmd:
      exec: /usr/bin/bash
      args: [ '/home/diaz/temp.sh' ]
```
I also created a simple bash script that uses arithmetic operation to multiply the monitored output by 1000
```console
$ cat /home/diaz/temp.sh
#!/bin/bash

echo $(( $(/usr/bin/nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader) * 1000 ))
```

I know this isn't the proper way (at least that i know of), but this is a quick and dirty workaround that works for me. Do you have any suggestion for better usability or even without the usage of a bash script?

markusressel, 2022-08-17:
> Do you have any suggestion for better usability or even without the usage of a bash script?

Not really. Using the `cmd` sensor (or fan for that matter) comes with its pros and cons. Initially I didn't want to include this feature at all. This does not mean that it doesn't work, it just means you have to come up with a solution for the commands yourself (which you already did :+1: ).

One thing that might help: you can just append three zeros to the original output and it will represent the same multiplication. Might make the command simple enough to put it into the configuration again without the need for a separate bash script. There is no "correct way" to all of this though.

markusressel, 2022-09-30:
I will close this, feel free to create a new issue if you have other questions.

closed 2022-09-30 by markusressel