markusressel/fan2go #153: Program ignores minPwm value

Shished, 2022-08-30:
**Describe the bug**
The config yaml file has minPwm set to some value but it is ignored by the program, the fan curve starts from PWM 0.

**To Reproduce**
Steps to reproduce the behavior:

1. Remove db file
2. Have minPwm value seto to >0 in yaml file
3. Start `fan2go` and let it initialize the fan curve
4. Run `fan2go curve list`

**Expected behavior**
Fan curve should start from the PWM value set by the minPwm option instead from 0.

**Screenshots**
```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
  ID         Type  
  gpu_curve  Linear

 255 ┤                                                                                               ╭───
 238 ┤                                                                                         ╭─────╯
 221 ┤                                                                                  ╭──────╯
 204 ┤                                                                           ╭──────╯
 187 ┤                                                                     ╭─────╯
 170 ┤                                                              ╭──────╯
 153 ┤                                                        ╭─────╯
 136 ┤                                                 ╭──────╯
 119 ┤                                          ╭──────╯
 102 ┤                                    ╭─────╯
  85 ┤                             ╭──────╯
  68 ┤                       ╭─────╯
  51 ┤                ╭──────╯
  34 ┤         ╭──────╯
  17 ┤   ╭─────╯
   0 ┼───╯
                                                   RPM / PWM

```

**Desktop (please complete the following information):**

- Distro: Arch Linux
- `uname -a`: `Linux linux 5.19.4-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Thu, 25 Aug 2022 17:31:19 +0000 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0+git with libsensors version 3.6.0+git`
- `fan2go version`:  "dev"

**Additional context**

The fan is attached to the GPU but is connected to the header on motherboard.
Contents of  `/etc/fan2go/fan2go.yaml`
```
# The path of the database file
dbPath: "/etc/fan2go/fan2go.db"

# Allow the fan initialization sequence to run in parallel for all configured fans
runFanInitializationInParallel: false
# The maximum difference between consecutive RPM measurements to
# consider a fan speed "settled"
# Note: This parameter is only used for initial analysis of fan curve
#       and has no effect during normal operation
maxRpmDiffForSettledFan: 10

# The rate to poll temperature sensors at
tempSensorPollingRate: 200ms
# The number of temp sensor values to keep in a rolling window array
tempRollingWindowSize: 10

# The rate to poll fan RPM input sensors at
rpmPollingRate: 1s
# The number of rpm sensor values to keep in a rolling window array
rpmRollingWindowSize: 10

# The rate to update fan speed targets at
controllerAdjustmentTickRate: 200ms

fans:
  # A user defined ID.
  # Used for logging only
  - id: gpu
    # The type of fan configuration
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6798-isa-0
      # The index of this fan as displayed by `fan2go detect`
      index: 4
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: false
    # The curve ID (defined above) that should be used to determine the
    # speed of this fan
    curve: gpu_curve
    # (Optional) Override for the lowest PWM value at which the
    # fan is able to maintain rotation if it was spinning previously.
    # (Optional) Override for the lowest PWM value at which the
    # fan will still be able to start rotating.
    # Note: Settings this to a value that is too small
    #       may damage your fans. Use at your own risk!
    # (Optional) Override for the highest PWM value which still yields
    # an increased rotational speed compared to lower values.
    # Note: you can also use this to limit the max speed of a fan.
    minPwm: 84
    # (Optional) Override for the lowest PWM value at which the
    # fan will still be able to start rotating.
    # Note: Settings this to a value that is too small
    #       may damage your fans. Use at your own risk!
    startPwm: 86
    # (Optional) Override for the highest PWM value which still yields
    # an increased rotational speed compared to lower values.
    # Note: you can also use this to limit the max speed of a fan.
    maxPwm: 255
# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: gpu_package
    # The type of sensor configuration
    hwmon:
      # The controller platform as displayed by `fan2go detect`, f.ex.:
      # "nouveau", "coretemp" or "it8620" etc.
      platform: amdgpu-pci-0100
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  # A user defined ID, which is used to reference
  # a curve in a fan configuration (see above)
  - id: gpu_curve
    # The type of curve configuration, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: gpu_package
      # Steps to define a section-wise defined speed curve function
      min: 40
      max: 80
```

markusressel, 2022-08-30:
The measured fan curve is independent from the minPwm value set in the config.
If the fan is never set to a speed below your minPwm, its working as intended. The config is not reflected in the real world curve measurement.

Shished, 2022-08-31:
Just tested it and it shows that the fan starts and stops at about 55 degrees instead of 40.

markusressel, 2022-09-29:
I hope this answers your question, feel free to reopen if not.

closed 2022-09-29 by markusressel

sergeolkhovik, 2023-05-22:
Can you please help me with minPwm and other values as well?
I'm trying to move from fancontrol to fan2go, my FC configuration is this:
```
% cat /etc/fancontrol
# https://wiki.archlinux.org/title/Fan_speed_control
# Configuration file generated by pwmconfig, changes will be lost
INTERVAL=10
DEVPATH=hwmon1=devices/platform/coretemp.0 hwmon2=devices/platform/w83627ehf
DEVNAME=hwmon1=coretemp hwmon2=w83627dhg
#FCTEMPS=hwmon1/pwm2=hwmon0/temp2_input
FCTEMPS=hwmon2/pwm2=/sys/devices/platform/coretemp.0/hwmon/[[:print:]]*/temp2_input
#FCFANS= hwmon1/pwm2=hwmon1/fan2_input
FCFANS= hwmon2/pwm2=/sys/devices/platform/w83627ehf/hwmon/[[:print:]]*/fan2_input
MINTEMP=hwmon2/pwm2=60
MAXTEMP=hwmon2/pwm2=65
MINSTART=hwmon2/pwm2=65
MINSTOP=hwmon2/pwm2=70
MINPWM=hwmon2/pwm2=70
# 2057RPM
MAXPWM=hwmon2/pwm2=94
# FULL
#MAXPWM=hwmon1/device/pwm2=255
```

In other words the CPU fan rotates in range of 65..94 PWM, trying to keep the temperature <60C.
I created this configuration:

```
fans:
  - id: cpu
    hwmon:
      platform: w83627dhg-isa-0
      index: 2
    neverStop: true
    curve: cpu_curve
    minPwm: 65
    maxPwm: 94

sensors:
  - id: cpu_package
    hwmon: 
      platform: w83627dhg-isa-0
      index: 2

curves:
  - id: cpu_curve
    linear:
      sensor: cpu_package
      min: 55
      max: 75
```

Then I remove db file and launch fan2go.

```
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for cpu...
 INFO  Loading fan curve data for fan 'cpu'...
 WARNING  Fan 'cpu' has not yet been analyzed, starting initialization sequence...
 INFO  Computing pwm map...
 INFO  Measuring RPM curve...
```

Surprisingly the fan gets 0 RPM speed:

```
% sudo sensors | grep -E 'fan2|temp2'
fan2:           0 RPM  (min = 1757 RPM, div = 128)  ALARM
temp2:        +58.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
```
May be this is done due to measuring mode, OK.
But fan2go reports another speed:
```
sudo fan2go fan --id cpu speed
37
```
In detect mode it reports:
```
% sudo fan2go detect                 
> w83627dhg-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           1      hwmon2  1654  228  true 
           2      hwmon2  0     57   false
           3      hwmon2  2008  255  false
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  43000
           2      hwmon2 (temp2_input)  64000
           3      hwmon2 (temp3_input)  7500 
```
I see it makes PWM value growing, OK.
57.. 77.. 89.. 122.. (why? the max is 94!) .. 255.
And then, after measurement is done it shows me this message:
WARNING  Suspicious pwm config of fan 'cpu': MinPwm (65) > StartPwm (50)
(strange, I don't have startPwm set in the config, it's commented out, OK, for next run I set it to 70).

But the most important thing is this (in normal mode, after measurement is done):

```
% sudo fan2go detect
> w83627dhg-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           1      hwmon2  1670  228  true 
           2      hwmon2  0     62   false
     
```
Why the PWM value is below 65?

After some time I see it stabilizes at 72 PWM / 60C:
```
% sudo fan2go detect
> w83627dhg-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           1      hwmon2  1654  228  true 
           2      hwmon2  1506  72   false
           3      hwmon2  2008  255  false
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  44000
           2      hwmon2 (temp2_input)  60000
```

Can you please tell me if I made a proper configuration that suits to my requirements?
I hope so, I tried to use pwmMap and steps but got completely confused with that.
I'll try it to load my system and see how fan2go behaves depending on the temperature.

markusressel, 2023-05-22:
> 57.. 77.. 89.. 122.. (why? the max is 94!) .. 255.

Measuring the fan curve has nothing to do with what you set in the config. The fan config is only used for controlling the fan during normal operation. If you set config values, the measurements that are done are ignored and replaced with your manual overrides.

Not sure why you are confused about the measurement, its explained in detail in the README, see: https://github.com/markusressel/fan2go#initialization

> And then, after measurement is done it shows me this message:
WARNING Suspicious pwm config of fan 'cpu': MinPwm (65) > StartPwm (50)
(strange, I don't have startPwm set in the config, it's commented out, OK, for next run I set it to 70).

If you do not specify all values in the config your manual overrides will be mixed with the ones determined by the measurements done during init. If this mixture doesn't make sense, simply override it with your own values.

> Why the PWM value is below 65?

This might be a bug.
The target PWM is applied using a PID controller, and while the calculated target PWM is always inside the minPwm and maxPwm bounds, the value applied by the PID controller is in 0..255, which might make it be able to exceed the expected bounds. If necessary, please open a new issue for this exact scenario.