markusressel/fan2go #161: Problem with fan2go fan init on Dell Inspiron 5370

shbon, 2022-10-01:
**Describe the bug**
Turning off BIOS fan control (AUR: dell-bios-fan-control) and then running sudo fan2go (or sudo fan2go fan init -icpu) for a new configuration displays the following:
```
...
PWM Settings of fan 'cpu': Min 100, Start 100, **Max 0**
```
The fan is turned on (previously 0 RPM, CPU at rest). The RPM is set to 2566 and PWM is 128, but stress testing the CPU indicates that fan2go isn't controlling the fan speed and I'm forced to re-enable BIOS fan control.

**To Reproduce**
```
sudo dell-bios-fan-control 0
sudo fan2go fan reset -icpu
sudo fan2go
stress -c 2 (on another tty)
```
**Expected behavior**
Presumably, the PWM settings should match my configuration:

```
fans
  - id: cpu
    hwmon:
    platform: dell_smm-isa-0
    index: 1
    neverStop: true
    curve: cpu_curve
    minPwm: 75
    startPwm: 100
    maxPwm: 195

sensors:
  - id: cpu_package
    hwmon:
    platform: coretemp-isa-0
    index: 1

curves:
  - id: cpu_curve
    linear:
    sensor: cpu_package
    steps:
     - 30: 0
     - 50: 125
     - 80: 195
...
````

**Desktop:**
- Arch Linux
- Dell Inspiron 5370 / Intel iGPU 8th Gen / Single fan
- fan2go version: latest
- I have the AUR package dell-bios-fan-control installed to disable BIOS control of the fan. It is activated with the command `sudo dell-bios-fan-control 0`

**Additional context**
fan2go _can_ read my temp sensor `fan2go sensor -icpu_package` and fan rpm `fan2go fan rpm -icpu`.
fan2go _can_ set the fan rpm `sudo fan2go fan speed nnn -icpu`, with BIOS fan control disabled.

FanControl (lm-sensors) works OK, but I'm looking to solve the "enumerated hwmon links changing issue" with fan2go. 

Any ideas?

markusressel, 2022-10-01:
Did you wait for the fan curve measurement to finish?
Rerun the main daemon with `--verbose` and post the output.

shbon, 2022-10-01:
Output after doing a "fan reset":
```
sudo fan2go -v --no-color --no-style

INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu...
INFO: Loading fan curve data for fan 'cpu'...
WARNING: Fan 'cpu' has not yet been analyzed, starting initialization sequence...
INFO: Computing pwm map...
DEBUG: Setting Fan PWM of 'cpu' to 255 ...
DEBUG: Setting Fan PWM of 'cpu' to 254 ...
DEBUG: Setting Fan PWM of 'cpu' to 253 ...
DEBUG: Setting Fan PWM of 'cpu' to 252 ...
DEBUG: Setting Fan PWM of 'cpu' to 251 ...
DEBUG: Setting Fan PWM of 'cpu' to 250 ...
DEBUG: Setting Fan PWM of 'cpu' to 249 ...
DEBUG: Setting Fan PWM of 'cpu' to 248 ...
DEBUG: Setting Fan PWM of 'cpu' to 247 ...
DEBUG: Setting Fan PWM of 'cpu' to 246 ...
DEBUG: Setting Fan PWM of 'cpu' to 245 ...
DEBUG: Setting Fan PWM of 'cpu' to 244 ...
DEBUG: Setting Fan PWM of 'cpu' to 243 ...
DEBUG: Setting Fan PWM of 'cpu' to 242 ...
DEBUG: Setting Fan PWM of 'cpu' to 241 ...
DEBUG: Setting Fan PWM of 'cpu' to 240 ...
DEBUG: Setting Fan PWM of 'cpu' to 239 ...
DEBUG: Setting Fan PWM of 'cpu' to 238 ...
DEBUG: Setting Fan PWM of 'cpu' to 237 ...
DEBUG: Setting Fan PWM of 'cpu' to 236 ...
DEBUG: Setting Fan PWM of 'cpu' to 235 ...
DEBUG: Setting Fan PWM of 'cpu' to 234 ...
DEBUG: Setting Fan PWM of 'cpu' to 233 ...
DEBUG: Setting Fan PWM of 'cpu' to 232 ...
DEBUG: Setting Fan PWM of 'cpu' to 231 ...
DEBUG: Setting Fan PWM of 'cpu' to 230 ...
DEBUG: Setting Fan PWM of 'cpu' to 229 ...
DEBUG: Setting Fan PWM of 'cpu' to 228 ...
DEBUG: Setting Fan PWM of 'cpu' to 227 ...
DEBUG: Setting Fan PWM of 'cpu' to 226 ...
DEBUG: Setting Fan PWM of 'cpu' to 225 ...
DEBUG: Setting Fan PWM of 'cpu' to 224 ...
DEBUG: Setting Fan PWM of 'cpu' to 223 ...
DEBUG: Setting Fan PWM of 'cpu' to 222 ...
DEBUG: Setting Fan PWM of 'cpu' to 221 ...
DEBUG: Setting Fan PWM of 'cpu' to 220 ...
DEBUG: Setting Fan PWM of 'cpu' to 219 ...
DEBUG: Setting Fan PWM of 'cpu' to 218 ...
DEBUG: Setting Fan PWM of 'cpu' to 217 ...
DEBUG: Setting Fan PWM of 'cpu' to 216 ...
DEBUG: Setting Fan PWM of 'cpu' to 215 ...
DEBUG: Setting Fan PWM of 'cpu' to 214 ...
DEBUG: Setting Fan PWM of 'cpu' to 213 ...
DEBUG: Setting Fan PWM of 'cpu' to 212 ...
DEBUG: Setting Fan PWM of 'cpu' to 211 ...
DEBUG: Setting Fan PWM of 'cpu' to 210 ...
DEBUG: Setting Fan PWM of 'cpu' to 209 ...
DEBUG: Setting Fan PWM of 'cpu' to 208 ...
DEBUG: Setting Fan PWM of 'cpu' to 207 ...
DEBUG: Setting Fan PWM of 'cpu' to 206 ...
DEBUG: Setting Fan PWM of 'cpu' to 205 ...
DEBUG: Setting Fan PWM of 'cpu' to 204 ...
DEBUG: Setting Fan PWM of 'cpu' to 203 ...
DEBUG: Setting Fan PWM of 'cpu' to 202 ...
DEBUG: Setting Fan PWM of 'cpu' to 201 ...
DEBUG: Setting Fan PWM of 'cpu' to 200 ...
DEBUG: Setting Fan PWM of 'cpu' to 199 ...
DEBUG: Setting Fan PWM of 'cpu' to 198 ...
DEBUG: Setting Fan PWM of 'cpu' to 197 ...
DEBUG: Setting Fan PWM of 'cpu' to 196 ...
DEBUG: Setting Fan PWM of 'cpu' to 195 ...
[... 314 lines cut ...]
sudo fan2go -i cpu fan speed [64-191] -->  PWM 128, RPM 2420 (approx)
sudo fan2go -i cpu fan speed [0-63] -->  PWM 255, RPM 4800 (approx)
```

Output from fan init (verbose, color and style flags don't seem to have any effect):
```
sudo fan2go -v --no-color --no-style -i cpu fan init

[30;46m[30;46m INFO [0m[0m [96m[96mUsing configuration file at: /etc/fan2go/fan2go.yaml[0m[0m
[30;46m[30;46m INFO [0m[0m [96m[96mUsing persistence at: /etc/fan2go/fan2go.db[0m[0m
[30;46m[30;46m INFO [0m[0m [96m[96mDeleting existing data for fan 'cpu'...[0m[0m
[30;46m[30;46m INFO [0m[0m [96m[96mComputing pwm map...[0m[0m
[30;46m[30;46m INFO [0m[0m [96m[96mMeasuring RPM curve...[0m[0m
[30;42m[30;42m SUCCESS [0m[0m [32m[32mDone![0m[0m
[30;46m[30;46m INFO [0m[0m [96m[96mUsing configuration file at: /etc/fan2go/fan2go.yaml[0m[0m
cpu
                
  Min PWM    0  
  Start PWM  100
  Max PWM    0  

 2525 ┼───────────────────────────────────────────────────────────────────────────────────────────────────
                                                    RPM / PWM

```

markusressel, 2022-10-01:
I found an issue with the config overrides for the pwm values not beeing applied correctly, I am working on a fix for that, so thx for making me aware :nerd_face: 

However, I don't think this will help you much because of the "three steps" you discovered.
Is it possible that the lm-sensors driver is only able to set some kind of "fan mode" (like silent, default, boost) using these?
In that case direct control probably isn't possible :cry: 

PS: I think you have a typo in the input range here:
```
sudo fan2go -i cpu fan speed [0-63] -->  PWM 255, RPM 4800 (approx)
```

markusressel, 2022-10-01:
Actually, now that I think about it, it might still work, but only for those three fan speeds.
You will not be able to control it very precisely though.

markusressel, 2022-10-01:
The fan curve measurement looks like nonesense to me. I guess _something_ is still also controlling fan speeds, in a more precise way, even if you change between those 3 speeds though lm-sensors. You might be able to overcome this with the pwm overrides in the config though.

I have merged the fix and you should be able to update immediately and retest.

shbon, 2022-10-01:
Ah, good that you've found something!

I think you're right regarding "fan mode". I used to use i8kutils before the kernel change deprecated the /proc/i8k interface. I think that was the same.

Using a userspace fan control is still beneficial though, as the BIOS fan control on this Dell Inspiron doesn't kick in until the CPU is at ~75 degrees! Fan is **off** before that point...

shbon, 2022-10-01:
Pulled the changes and now fan2go now shows my maxPwd of 195 during initialisation (thanks), **but I still can't get it to react to CPU temperature changes**.

E.g. For the following configuration (and initial conditions cpu temp 30 degrees, BIOS fan control disabled, fan rpm 0)...
```
fans:
  - id: cpu
    hwmon:
      platform: dell_smm-isa-0
      index: 1
    neverStop: false
    curve: cpu_curve
    minPwm: 0
    startPwm: 100
    maxPwm: 195

# A list of sensors to monitor
sensors:
  - id: cpu_package
    hwmon:
      platform: coretemp-isa-0
      index: 1

curves:
  - id: cpu_curve
    linear:
      sensor: cpu_package
      steps:
        # Sensor value -> Speed (0-255)
        - 35: 0
        - 45: 64
        - 65: 195

```
... I was expecting fan2go to set RPM to 0 when entering its "controller loop" (as CPU temp is less than min curve temp). fan2go sets the PWM to 128 on entering its "controller loop" (or maybe the PWM 128 value set _during_ initialisation doesn't get changed???).

Performing a stress test still doesn't trigger the max PWM at 65 degrees...

Running `fan2go sensor -i cpu_package` returns 31000 so it seems the temp sensor **is** configured properly...

Any thoughts?

New log of sudo fan2go -v (for completeness). **Can we get some debug _after_ fan2go enters its "controller loop"?**:
```
INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu...
INFO: Loading fan curve data for fan 'cpu'...
WARNING: Fan 'cpu' has not yet been analyzed, starting initialization sequence...
INFO: Computing pwm map...
DEBUG: Setting Fan PWM of 'cpu' to 255 ...
DEBUG: Setting Fan PWM of 'cpu' to 254 ...
DEBUG: Setting Fan PWM of 'cpu' to 253 ...
DEBUG: Setting Fan PWM of 'cpu' to 252 ...
DEBUG: Setting Fan PWM of 'cpu' to 251 ...
DEBUG: Setting Fan PWM of 'cpu' to 250 ...
DEBUG: Setting Fan PWM of 'cpu' to 249 ...
DEBUG: Setting Fan PWM of 'cpu' to 248 ...
DEBUG: Setting Fan PWM of 'cpu' to 247 ...
DEBUG: Setting Fan PWM of 'cpu' to 246 ...
DEBUG: Setting Fan PWM of 'cpu' to 245 ...
DEBUG: Setting Fan PWM of 'cpu' to 244 ...
DEBUG: Setting Fan PWM of 'cpu' to 243 ...
DEBUG: Setting Fan PWM of 'cpu' to 242 ...
DEBUG: Setting Fan PWM of 'cpu' to 241 ...
DEBUG: Setting Fan PWM of 'cpu' to 240 ...
DEBUG: Setting Fan PWM of 'cpu' to 239 ...
DEBUG: Setting Fan PWM of 'cpu' to 238 ...
DEBUG: Setting Fan PWM of 'cpu' to 237 ...
DEBUG: Setting Fan PWM of 'cpu' to 236 ...
DEBUG: Setting Fan PWM of 'cpu' to 235 ...
DEBUG: Setting Fan PWM of 'cpu' to 234 ...
[... 331 lines cut ...]
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 70 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 71 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 71 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 72 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 72 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 73 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 73 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 74 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 74 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 75 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 75 Actual: 0
DEBUG: Setting Fan PWM of 'cpu' to 76 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 76 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 77 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 78 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 79 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 80 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 81 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 82 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 83 Actual: 128
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 84 Actual: 128
INFO: PWM settings of fan 'cpu': Min 0, Start 100, Max 195
INFO: Starting controller loop for fan 'cpu'

```

shbon, 2022-10-01:
Hmmm, an extra line has now appeared on the debug after an unknown time period (minutes?), which has turned the fan off:

`DEBUG: Setting Fan PWM of 'cpu' to 63...
`

Running a further stress test doesn't illicit any (immediate) response from fan2go...

markusressel, 2022-10-01:
> I was expecting fan2go to set RPM to 0 when entering its "controller loop"

Just to be precise here: fan2go only sets PWM values, it (currently) cannot set target RPM values. 

> fan2go sets the PWM to 128 on entering its "controller loop" (or maybe the PWM 128 value set during initialisation doesn't get changed???).

After computing the "pwm map" fan2go sets the PWM to the `startPwm` value. Since you set a value of `100` in your config, fan2go tries to find the closest value to it that is part of the "pwm map", which results in 128 beeing picked (out of 0, 128 and 255 I guess). I think this was done to make sure the fan is somewhere in the middle of its operating range before continuing, but we can improve this if necessary.

It doesn't look right that the initialization stops at a pwm value of just `84`, when you expect `0`, `128` and `255`. I have added some debug output to print the values of the pwm map, as well as the distinct pwm targets for a given fan (output abbreviated for readability):

```
DEBUG: Distinct PWM value targets of fan gpu: [0 ............... 255]
DEBUG: PWM map of fan 'gpu': map[0:0 1:0 2:0 ...........  254:252 255:255]
```

Please update again and rerun to see what those output.

shbon, 2022-10-01:
Thanks! 
Yes, I meant PWM 0 rather than RPM 0 (my minPwm = 0).

Ah, I thought startPwm was the "safe start" PWM value as per FanControl. i.e. If you're turning the fan on, when its current RPM is 0, use this startPWM value... 

I think I'm giving it enough time. After fan2go stopped the fan (see my post just before yours), I reran my (slow) stress test. It takes about 15 secs to ramp the temp from 30 to 75 degrees, so there should be enough time for a reaction. My YAML config has default values for all the timing settings. I've included my full config for reference at the end.

Pulled the change. New log:
```
INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu...
INFO: Loading fan curve data for fan 'cpu'...
WARNING: Fan 'cpu' has not yet been analyzed, starting initialization sequence...
INFO: Computing pwm map...
DEBUG: Setting Fan PWM of 'cpu' to 255 ...
DEBUG: Setting Fan PWM of 'cpu' to 254 ...
DEBUG: Setting Fan PWM of 'cpu' to 253 ...
DEBUG: Setting Fan PWM of 'cpu' to 252 ...
DEBUG: Setting Fan PWM of 'cpu' to 251 ...
DEBUG: Setting Fan PWM of 'cpu' to 250 ...
DEBUG: Setting Fan PWM of 'cpu' to 249 ...
DEBUG: Setting Fan PWM of 'cpu' to 248 ...
DEBUG: Setting Fan PWM of 'cpu' to 247 ...
DEBUG: Setting Fan PWM of 'cpu' to 246 ...
DEBUG: Setting Fan PWM of 'cpu' to 245 ...
DEBUG: Setting Fan PWM of 'cpu' to 244 ...
DEBUG: Setting Fan PWM of 'cpu' to 243 ...
DEBUG: Setting Fan PWM of 'cpu' to 242 ...
DEBUG: Setting Fan PWM of 'cpu' to 241 ...
DEBUG: Setting Fan PWM of 'cpu' to 240 ...
DEBUG: Setting Fan PWM of 'cpu' to 239 ...
DEBUG: Setting Fan PWM of 'cpu' to 238 ...
DEBUG: Setting Fan PWM of 'cpu' to 237 ...
DEBUG: Setting Fan PWM of 'cpu' to 236 ...
DEBUG: Setting Fan PWM of 'cpu' to 235 ...
DEBUG: Setting Fan PWM of 'cpu' to 234 ...
DEBUG: Setting Fan PWM of 'cpu' to 233 ...
DEBUG: Setting Fan PWM of 'cpu' to 232 ...
DEBUG: Setting Fan PWM of 'cpu' to 231 ...
DEBUG: Setting Fan PWM of 'cpu' to 230 ...
DEBUG: Setting Fan PWM of 'cpu' to 229 ...
DEBUG: Setting Fan PWM of 'cpu' to 228 ...
DEBUG: Setting Fan PWM of 'cpu' to 227 ...
DEBUG: Setting Fan PWM of 'cpu' to 226 ...
DEBUG: Setting Fan PWM of 'cpu' to 225 ...
DEBUG: Setting Fan PWM of 'cpu' to 224 ...
DEBUG: Setting Fan PWM of 'cpu' to 223 ...
DEBUG: Setting Fan PWM of 'cpu' to 222 ...
DEBUG: Setting Fan PWM of 'cpu' to 221 ...
DEBUG: Setting Fan PWM of 'cpu' to 220 ...
DEBUG: Setting Fan PWM of 'cpu' to 219 ...
DEBUG: Setting Fan PWM of 'cpu' to 218 ...
DEBUG: Setting Fan PWM of 'cpu' to 217 ...
DEBUG: Setting Fan PWM of 'cpu' to 216 ...
DEBUG: Setting Fan PWM of 'cpu' to 215 ...
DEBUG: Setting Fan PWM of 'cpu' to 214 ...
DEBUG: Setting Fan PWM of 'cpu' to 213 ...
DEBUG: Setting Fan PWM of 'cpu' to 212 ...
DEBUG: Setting Fan PWM of 'cpu' to 211 ...
DEBUG: Setting Fan PWM of 'cpu' to 210 ...
DEBUG: Setting Fan PWM of 'cpu' to 209 ...
DEBUG: Setting Fan PWM of 'cpu' to 208 ...
DEBUG: Setting Fan PWM of 'cpu' to 207 ...
DEBUG: Setting Fan PWM of 'cpu' to 206 ...
DEBUG: Setting Fan PWM of 'cpu' to 205 ...
DEBUG: Setting Fan PWM of 'cpu' to 204 ...
DEBUG: Setting Fan PWM of 'cpu' to 203 ...
DEBUG: Setting Fan PWM of 'cpu' to 202 ...
DEBUG: Setting Fan PWM of 'cpu' to 201 ...
DEBUG: Setting Fan PWM of 'cpu' to 200 ...
[... 379 lines cut ...]
curves:
  - id: cpu_curve
    linear:
      sensor: cpu_package
      steps:
        # Sensor value -> Speed (0-255)
        - 35: 0
        - 45: 64
        - 65: 195

statistics:
  # Whether to enable the prometheus exporter or not
  enabled: false
  # The port to expose the exporter on
  port: 9000

api:
  # Whether to enable the API or not
  enabled: false
  # The host to listen for connections
  host: localhost
  # The port to listen for connections
  port: 9001

```

shbon, 2022-10-01:
> Hmmm, an extra line has now appeared on the debug after an unknown time period (minutes?), which has turned the fan off:
> 
> `DEBUG: Setting Fan PWM of 'cpu' to 63... `

Regarding the above. **This extra line appears after about 5 minutes.** 

Is that a built-in delay? Or are my temperature polling values being misinterpreted?

```
# The rate to poll temperature sensors at
tempSensorPollingRate: 200ms
# The number of temp sensor values to keep in a rolling window array
tempRollingWindowSize: 10 

```

markusressel, 2022-10-01:
Ok so the distinct values are all over the place, when they should really only contain 3 values.

This is caused by the fact that you seem to get a value of `0` both at the start and the end of the `0-255` spectrum, but not in between, and that my algorithm didn't sort the keys before analyzing the map.

I have done some changes which should improve this, in addition to some tests. With those changes the resulting distinct pwm targets are: `0, 64, 244`, which seem to fit your needs.

**However**, fan2go considering `244` as a "good" value could be debatable, since it also maps to 0, just like all the values between 0 and 63 do, which doesn't make any logical sense.

Is it possible that the "pwm value" changes on its own depending on the speed of the fan and that its not actually a writable value but can only be read? We have observed similar behavior in some GPU fan driver implementations.

> Is that a built-in delay? Or are my temperature polling values being misinterpreted?

There is definitely no built-in delay of 5 minutes anywhere.

The config you posted will result in a **sensor** value that shifts from one value to the next over the course of 10 * 200ms. A sensor value change from 51000 to 50000 will take 2 seconds to complete, but update to values between "start" and "end" 10 times every 200 milliseconds.

markusressel, 2022-10-01:
Btw you can collapse long code blocks, please consider editing your comments for better readability of this whole thread.

See here: https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab

shbon, 2022-10-01:
> This is caused by the fact that you seem to get a value of `0` both at the start and the end of the `0-255` spectrum, but not in between, and that my algorithm didn't sort the keys before analyzing the map.
> 
Why does the initialisation process detect this '0' value at the top end, when I can set the fan PWM to [192-255] with e.g. `sudo fan2go -i cpu fan speed 192` which sets the fan RPM to full speed, which can be read as PWM 255 by `fan2go -i cpu fan speed`?

(Actually, I can't at the moment, as that functionality seems to have been broken in the last release...)
```
fan2go -i cpu fan rpm returns N/A
fan2go -i cpu fan speed returns nothing
sudo fan2go -i cpu fan speed 199 doesn't set the fan PWM anymore

```

> The config you posted will result in a **sensor** value that shifts from one value to the next over the course of 10 * 200ms. A sensor value change from 51000 to 50000 will take 2 seconds to complete, but update to values between "start" and "end" 10 times every 200 milliseconds.

_So there should be a reaction in about 2 seconds, once the temperature crosses my temp curve boundary_. That still doesn't seem to be happening. Although after initialisation, the fan is now shut off straight away rather than after 5 mins (yay), running the stress test (30 to 75 degrees in 15 seconds) still gets no reaction from fan2go...(the fan stays at 0 RPM).

Regarding the new log below, is the PWM map wrong for PWM values >= 192 (the actual value is 128 for 192->243 and 0 for 244->255)? i.e. It looks like the fan will never be set to full speed...
(192 being the lowest "set" value of PWM that results in an "actual" PWM value for 255)

Also not sure about this line. Shouldn't it be [0 64 195] as per config?
`DEBUG: Distinct PWM value targets of fan cpu: [0 64 244]`

New log below (can't get rid of extra white space)

<details>
<summary>New Log (Expand)</summary>

```
INFO: Using configuration file at: /etc/fan2go/fan2go.yaml
INFO: Gathering sensor data for cpu...
INFO: Loading fan curve data for fan 'cpu'...
WARNING: Fan 'cpu' has not yet been analyzed, starting initialization sequence...
INFO: Computing pwm map...
DEBUG: Setting Fan PWM of 'cpu' to 255 ...
DEBUG: Setting Fan PWM of 'cpu' to 254 ...
DEBUG: Setting Fan PWM of 'cpu' to 253 ...
DEBUG: Setting Fan PWM of 'cpu' to 252 ...
DEBUG: Setting Fan PWM of 'cpu' to 251 ...
DEBUG: Setting Fan PWM of 'cpu' to 250 ...
DEBUG: Setting Fan PWM of 'cpu' to 249 ...
DEBUG: Setting Fan PWM of 'cpu' to 248 ...
DEBUG: Setting Fan PWM of 'cpu' to 247 ...
DEBUG: Setting Fan PWM of 'cpu' to 246 ...
DEBUG: Setting Fan PWM of 'cpu' to 245 ...
DEBUG: Setting Fan PWM of 'cpu' to 244 ...
DEBUG: Setting Fan PWM of 'cpu' to 243 ...
DEBUG: Setting Fan PWM of 'cpu' to 242 ...
DEBUG: Setting Fan PWM of 'cpu' to 241 ...
DEBUG: Setting Fan PWM of 'cpu' to 240 ...
DEBUG: Setting Fan PWM of 'cpu' to 239 ...
DEBUG: Setting Fan PWM of 'cpu' to 238 ...
DEBUG: Setting Fan PWM of 'cpu' to 237 ...
DEBUG: Setting Fan PWM of 'cpu' to 236 ...
DEBUG: Setting Fan PWM of 'cpu' to 235 ...
DEBUG: Setting Fan PWM of 'cpu' to 234 ...
DEBUG: Setting Fan PWM of 'cpu' to 233 ...
DEBUG: Setting Fan PWM of 'cpu' to 232 ...
DEBUG: Setting Fan PWM of 'cpu' to 231 ...
DEBUG: Setting Fan PWM of 'cpu' to 230 ...
DEBUG: Setting Fan PWM of 'cpu' to 229 ...
DEBUG: Setting Fan PWM of 'cpu' to 228 ...
DEBUG: Setting Fan PWM of 'cpu' to 227 ...
DEBUG: Setting Fan PWM of 'cpu' to 226 ...
DEBUG: Setting Fan PWM of 'cpu' to 225 ...
DEBUG: Setting Fan PWM of 'cpu' to 224 ...
DEBUG: Setting Fan PWM of 'cpu' to 223 ...
DEBUG: Setting Fan PWM of 'cpu' to 222 ...
DEBUG: Setting Fan PWM of 'cpu' to 221 ...
DEBUG: Setting Fan PWM of 'cpu' to 220 ...
DEBUG: Setting Fan PWM of 'cpu' to 219 ...
[... 225 lines cut ...]
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 2653.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 196.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 165.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 155.000000)...
DEBUG: Waiting for fan cpu to settle (current RPM max diff: 23.000000)...
DEBUG: Fan cpu has settled (current RPM max diff: 6.000000)
DEBUG: Measuring RPM of cpu at PWM 0: 2519
DEBUG: Measured RPM of 2519 at PWM 0 for fan cpu
DEBUG: Setting Fan PWM of 'cpu' to 1 ...
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 1 Actual: 0
DEBUG: Fan cpu: Actual PWM value differs from requested one, skipping. Requested: 2 Actual: 0
DEBUG: Distinct PWM value targets of fan cpu: [0 64 244]
DEBUG: PWM map of fan 'cpu': map[0:0 1:0 2:0 3:0 4:0 5:0 6:0 7:0 8:0 9:0 10:0 11:0 12:0 13:0 14:0 15:0 16:0 17:0 18:0 19:0 20:0 21:0 22:0 23:0 24:0 25:0 26:0 27:0 28:0 29:0 30:0 31:0 32:0 33:0 34:0 35:0 36:0 37:0 38:0 39:0 40:0 41:0 42:0 43:0 44:0 45:0 46:0 47:0 48:0 49:0 50:0 51:0 52:0 53:0 54:0 55:0 56:0 57:0 58:0 59:0 60:0 61:0 62:0 63:0 64:128 65:128 66:128 67:128 68:128 69:128 70:128 7 [... cut ...] 128 206:128 207:128 208:128 209:128 210:128 211:128 212:128 213:128 214:128 215:128 216:128 217:128 218:128 219:128 220:128 221:128 222:128 223:128 224:128 225:128 226:128 227:128 228:128 229:128 230:128 231:128 232:128 233:128 234:128 235:128 236:128 237:128 238:128 239:128 240:128 241:128 242:128 243:128 244:0 245:0 246:0 247:0 248:0 249:0 250:0 251:0 252:0 253:0 254:0 255:0]
INFO: PWM settings of fan 'cpu': Min 0, Start 100, Max 195
INFO: Starting controller loop for fan 'cpu'

```
</details>

markusressel, 2022-10-01:
> Why does the initialisation process detect this '0' value at the top end, when I can set the fan PWM to [192-255] with e.g. `sudo fan2go -i cpu fan speed 192` which sets the fan RPM to full speed, which can be read as PWM 255 by `fan2go -i cpu fan speed`?

Thats a very good question I currently don't have an answer to. I would expect the same thing as you do. It may be due to the "pwm map creation" taking place within only a very short a mount of time, which could lead to weird behavior. This process actually tries to set every possible PWM value in the range 0-255 (as you can see in the log) and "waits" a really short amount of time before reading back what PWM value actually got set afterwards.

If you do this test manually and continuously without fan2go daemon running, using f.ex. a `watch -n 0.2 --color -- sudo fan2go detect `, does the PWM (**not** the RPM) value change "on its own"? Does it fluctuate between values sometimes? Anything unexpected?

> (Actually, I can't at the moment, as that functionality seems to have been broken in the last release...)
> ```
> fan2go -i cpu fan rpm returns N/A
> fan2go -i cpu fan speed returns nothing
> sudo fan2go -i cpu fan speed 199 doesn't set the fan PWM anymore
> ```

There really was nothing changed in the code for that to happen. The commands for reading and setting stuff are completely independent from the logic of the daemon. There is no pwm map calculation f.ex., so maybe the system acts weird because we messed around to much or possibly there was a mistake on your end?

> _So there should be a reaction in about 2 seconds, once the temperature crosses my temp curve boundary_. 

Well, on a "normal" system I would answer with a resounding "yes, it certainly would". However on your system, because of the extremely limited range of distinct PWM values at hand, you would only start to notice a change after the fan control loop decides to increase the fan speed to a level that is closer to `64` than it is to `0`, because there are no other values in between that fan2go considers available, because the PWM map tells it this. It shouldn't take 5 minutes, but it might take more than 15 seconds. It might be an issue that the gap between those supported values is so large, wich might cause weird behavior inside of fan2go. I will have to look into that.

> That still doesn't seem to be happening. Although after initialisation, the fan is now shut off straight away rather than after 5 mins (yay), running the stress test (30 to 75 degrees in 15 seconds) still gets no reaction from fan2go...(the fan stays at 0 RPM).

I think it would be helpful to see statistics not only for the fans, sensors and curves but also the internal state of fan controller itself to better visualize what is happening on the inside while the fan2go daemon is running. I will add some statistics for that, but it will take some time to do that.

> Regarding the new log below:
> 
> 1. Is the PWM map wrong for PWM values >= 192 (the actual value is 128 for 192->243 and 0 for 244->255)? i.e. It looks like the fan will never be set to full speed...
>    (192 being the lowest "set" value of PWM that results in an "actual" PWM value for 255)

It sure doesn't look healthy, yes. I cannot tell what will happen because this is not a "supported" state. I will need to invest more time into this to come up with a solution that can remedy this. In a "perfect" fan this would be a `x -> x, for x in [0..255]` mapping.

> 2. Not sure about this line. Shouldn't it be [0 64 195] as per config?
>    `DEBUG: Distinct PWM value targets of fan cpu: [0 64 244]`

No. While the configuration overrides what the fan control loop should use as PWM boundaries for a specific fan, the distinct pwm value targets are calculated from the pwm map, which in turn is calculcated by trying out all PWM values (as mentioned above). It has nothing to do with what the config contains, only what fan2go sees when it tries to interact with the fan's "PWM knob".

Thx for sticking with this for so long btw, its really helpful and constructive to work on this with you :heart: :nerd_face:

shbon, 2022-10-01:
> If you do this test manually and continuously without fan2go daemon running, using f.ex. a `watch -n 0.2 --color -- sudo fan2go detect `, does the PWM (**not** the RPM) value change "on its own"? Does it fluctuate between values sometimes? Anything unexpected?
> 
Ah, that was interesting (and probably significant)...
If you remember from an earlier post, I wrote:
```
sudo fan2go -i cpu fan speed [0-63] -->  PWM 0
sudo fan2go -i cpu fan speed [64-191] -->  PWM 128, RPM 2420 (approx)
sudo fan2go -i cpu fan speed [192-255] -->  PWM 255, RPM 4800 (approx)
```
Looking at the boundary [191,192], this switches from half fan speed to full fan speed. So with the initial condition of fan at half speed (2400rpm), if you do `sudo fan2go -i cpu fan speed 192`, the fan speed starts ramping up to maximium (4800rpm) _over about 16 seconds_, **but PWM stays at 128 until about 3600rpm, where it "flips" to 255**. i.e It flips at about the half way point. This flip occurs after about 8 seconds. i.e. Linear increase in fan speed.

This would seem to indicate you'd need to wait > 8 seconds to detect this step change in PWM...

The same thing happens going from full speed (rpm) to half speed. i.e. Full speed initial condition, set PWM to 191 --> wait 8 seconds for flip to 128 PWM.

Setting PWM to ranges [64-190] or [193-255] have no effect on RPM or PWM values. (There's no fluctuation).

At the bottom end, it's different.
Initial condition = fan half speed. Set PWM = 63 --> RPM 0 / PWM 0 instantly.
Initial condition = fan off. Set PWM = 64 --> RPM 2400 (very quick ramp + overshoot to 2800 in about 1s then settles to half speed after a couple more seconds) / PWM 128 instantly.

**I think the take home for you is that you need to _watch_ for a stable RPM _before_ reading the PWM value, whenever you set the PWM value during the initialisation...**

> There really was nothing changed in the code for that to happen. The commands for reading and setting stuff are completely independent from the logic of the daemon. There is no pwm map calculation [...]

Sorry, that was noise from me! I ran the commands from the build directory, which picked up the YAML config supplied with the build instead for my /etc/fan2go based config!

> Well, on a "normal" system I would answer with a resounding "yes, it certainly would". However on your system, because of the extremely limited range of distinct PWM values at hand, you would only [...]

That's why I set the values I did [0, 64, 195]. This should provide two linear ranges where the end points are the trigger values for changing PWM to one of 0, 128, 255. (I left a bit of stack in the top value to ensure the PWM 255 will be triggered). It should work, I think.

> It sure doesn't look healthy, yes. I cannot tell what will happen because this is not a "supported" state. I will need to invest more time into this to come up with a solution that can remedy this. [...]
I think it will work OK, once the PWM Map is straightened out, in line with the first part of the message, and my reasoning regarding values [0, 64, 195]. I don't think anything else will need attention. 

> No. While the configuration overrides what the fan control loop should use as PWM boundaries for a specific fan, the distinct pwm value targets are calculated from the pwm map, which in turn is [...]
My understanding is I have a linear temp range of [35-45] with a corresponding PWM range of [0-64]. So reacting to a temp rise from say 35 to 45, you (linearly) ramp PWM from 0 to 64. When the temp hits 45, the PWM hits 64, _which (in this designed case) triggers the fan to half speed_. Similarily for the [45-65] temp range ([65-195] PWM range). You ramp PWM from 65 to 195 as the temp increases from 45 to 65. At 65 degrees, PWM = 195 which triggers full fan speed. (Actually 192 triggers full speed). I'm not quite sure how the value PWM 244 plays into this?  

> Thx for sticking with this for so long btw, its really helpful and constructive to work on this with you heart nerd_face
No worries!

markusressel, 2022-10-01:
> This would seem to indicate you'd need to wait > 8 seconds to detect this step change in PWM...

Ah, that explains why the fast rpm map creation is not working as intended. :+1:

> **I think the take home is that you need to _watch and wait for_ a stable RPM _before_ reading the resulting PWM value in the hwmon area, whenever you set the PWM value in the initialisation procedure...** (Pretty much a "asked for", "wait for stability" "got" function)

For detecting the fan RPM curve fan2go actually already does this, but its obviously failing too since your fan curve is a flatline at 2525 RPM, so I am not sure if it will work for the PWM map creation :thinking: 

The only real fix I can see for this that would work 100% is to allow specifying a custom PWM map in the config and skip the automatic detection entirely. I still don't understand why its detecting a mapping to `0` at the upper end.

> That's why I set the values I did [0, 64, 195]. This should provide two linear ranges where the end points are the trigger values for changing PWM to one of 0, 128, 255. (I left a bit of stack in the top value to ensure the PWM 255 will be triggered). It should work, I think. (Also see my answer to the last but one point).

Yes, I think so too, I just need to cover this scenario with more tests to make sure its working just like the "normal" case does.

Also, you will probably see lots of warnings from fan2go, due to it thinking that an external program is also changing the PWM values, because fan2go constantly compares the "set" PWM value with the one reported by the system currently, and those will only match up after some time. It would be completely fine to ignore those though.

> I think it will work OK, once the PWM Map is straightened out in line with the first part of this post, and my reasoning regarding values [0, 64, 195]. I don't think anything else will need attention.

I hope so too :crossed_fingers: 

> My understanding is I have a linear temp range of [35-45] with a corresponding PWM range of [0-64]. So reacting to a temp rise from say 35 to 45, you (linearly) ramp PWM from 0 to 64 (not physically applying each value?). When the temp hits 45, the PWM hits 64, _which (in this designed case) triggers the fan to half speed_. Similarily for the [45-65] temp range ([65-195] PWM range). You ramp PWM from 65 to 195 as the temp increases from 45 to 65. At 65 degrees, PWM = 195 triggers (if applied) full fan speed. (Actually 192 triggers full speed). I'm not quite sure how the value PWM 244 plays into this?

You're spot on, thats the exact behavior I want to see as well.
In your last log the detected distinct PWM values were logged as this:

```
DEBUG: Distinct PWM value targets of fan cpu: [0 64 244]
```

Thats were I got the 244 from. But the logic you wrote applies to it just as well, the tipping point is just a different one. And of course you would want to specify a value closer to the actual tipping point, like the 195 you mentioned.

I will report back when I have something new to test.

markusressel, 2022-10-09:
I haven't finished the documentation yet, but you can now specify the pwm map in the config, something like this:

```
fans:
  - id: xyz
    ...
    pwmMap:
      0: 0
      64: 128
      195: 255
```

I have done some testing and it behaves as I would expect (and we previously discussed).
Please let me know how it works for you :pray: :nerd_face:

shbon, 2022-10-10:
Thanks for the update. It's controlling the laptop fan nicely now! 

Also runs fine as a systemd service. My only criticism is the journalctl log spam created when moving between half-speed and full-speed and vice versa. You get around 40 _"PWM of cpu was changed by a third party!"_ warning messages per PWM change...

If you remember, I commented that moving from half speed (2400rpm) to full speed (4800rpm) takes about 16 seconds, but the PWM values stays at 128 until about 3600rpm, where it "flips" to 255. So when fan2go "asks for" PWM=255 when current PWM is 128, it will take about 8 secs until the hwmon value shows 255. It's the same when fan2go "asks for" PWM=128, when current PWM is 255.

I realise you've comment about the _"PWM of cpu was changed by a third party!"_ in [https://github.com/markusressel/fan2go/issues/64](url) but can something be done to mitigate for this particular fan type? E.g. If a pwmMap is specified in config don't warn for 'n' seconds or E.g. don't warn again where the PWM 'asked for' and 'got' values are the same as the last "changed by a third party" warning.

Thanks again for the work you've done helping me with this problem. I appreciate it.

markusressel, 2022-10-11:
Great trick with the grep 👍 Glad its working :muscle:

markusressel, 2022-10-11:
I will close this issue. Feel free to open another if something comes up.

closed 2022-10-11 by markusressel