markusressel/fan2go #172: Controlling a water pump

ThisIsTheOnlyUsernameAvailable, 2022-10-24:
Thanks very much for this project -- it is absolutely fantastic.

**Describe the bug**
There seems to be an issue analysing the behaviour of a water pump in a watercooled setup.  Whilst I don't have experience with PID controllers in cooling, I do in autopilots, and my laymans view is that the pump isn't behaving like the software expects.  It never seems to settle at a level acceptable.

I've tried a few alterations of the PID controller without success. 

Has anyone successfully gotten this to work in a similar setup?

**Logs**
`INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 WARNING  Unused sensor configuration: case-1
 WARNING  Unused sensor configuration: case-2
 WARNING  Unused sensor configuration: water
 INFO  Gathering sensor data for back...
 INFO  Gathering sensor data for front...
 INFO  Gathering sensor data for top...
 INFO  Gathering sensor data for pump...
 INFO  Loading fan curve data for fan 'back'...
 INFO  Loading fan curve data for fan 'front'...
 INFO  Loading fan curve data for fan 'top'...
 INFO  Loading fan curve data for fan 'pump'...
 INFO  FanController: Using saved value for pwm map of Fan 'back'
  DEBUG   Distinct PWM value targets of fan back: [0 2 4 7 9 12 15 17 20 22 25 27 30 32 35 37 40 43 45 48 50 53 55 58 60 63 66 68 71 73 76 78 81 83 86 88 91 94 96 99 101 104 106 109 111 114 117 119 122 124 127 129 132 134 137 139 142 145 147 150 152 155 157 160 162 165 168 170 173 175 178 180 183 185 188 190 193 196 198 201 203 206 208 211 213 216 219 221 224 226 229 231 234 236 239 241 244 247 249 252 254]
  DEBUG   PWM map of fan 'back': map[0:0 1:0 2:3 3:3 4:5 5:5 6:5 7:8 8:8 9:10 10:10 11:10 12:13 13:13 14:13 15:15 16:15 17:18 18:18 19:18 20:20 21:20 22:23 23:23 24:23 25:26 26:26 27:28 28:28 29:28 30:31 31:31 32:33 33:33 34:33 35:36 36:36 37:38 38:38 39:38 40:41 41:41 42:41 43:43 44:43 45:46 46:46 47:46 48:48 49:48 50:51 51:51 52:51 53:54 54:54 55:56 56:56 57:56 58:59 59:59 60:61 61:61 62:61 63:64 64:64 65:64 66:66 67:66 68:69 69:69 70:69 71:71 72:71 73:74 74:74 75:74 76:77 77:77 78:79 79:79 80:79 81:82 82:82 83:84 84:84 85:84 86:87 87:87 88:89 89:89 90:89 91:92 92:92 93:92 94:94 95:94 96:97 97:97 98:97 99:99 100:99 101:102 102:102 103:102 104:105 105:105 106:107 107:107 108:107 109:110 110:110 111:112 112:112 113:112 114:115 115:115 116:115 117:117 118:117 119:120 120:120 121:120 122:122 123:122 124:125 125:125 126:125 127:128 128:128 129:130 130:130 131:130 132:133 133:133 134:135 135:135 136:135 137:138 138:138 139:140 140:140 141:140 142:143 143:143 144:143 145:145 146:145 147:148 148:148 149:148 150:150 151:150 152:153 153:153 154:153 155:156 156:156 157:158 158:158 159:158 160:161 161:161 162:163 163:163 164:163 165:166 166:166 167:166 168:168 169:168 170:171 171:171 172:171 173:173 174:173 175:176 176:176 177:176 178:179 179:179 180:181 181:181 182:181 183:184 184:184 185:186 186:186 187:186 188:189 189:189 190:191 191:191 192:191 193:194 194:194 195:194 196:196 197:196 198:199 199:199 200:199 201:201 202:201 203:204 204:204 205:204 206:207 207:207 208:209 209:209 210:209 211:212 212:212 213:214 214:214 215:214 216:217 217:217 218:217 219:219 220:219 221:222 222:222 223:222 224:224 225:224 226:227 227:227 228:227 229:230 230:230 231:232 232:232 233:232 234:235 235:235 236:237 237:237 238:237 239:240 240:240 241:242 242:242 243:242 244:245 245:245 246:245 247:247 248:247 249:250 250:250 251:250 252:252 253:252 254:255 255:255]
 INFO  PWM settings of fan 'back': Min 5, Start 5, Max 99
 INFO  Starting controller loop for fan 'back'
 INFO  FanController: Using saved value for pwm map of Fan 'front'
  DEBUG   Distinct PWM value targets of fan front: [0 2 4 7 9 12 15 17 20 22 25 27 30 32 35 37 40 43 45 48 50 53 55 58 60 63 66 68 71 73 76 78 81 83 86 88 91 94 96 99 101 104 106 109 111 114 117 119 122 124 127 129 132 134 137 139 142 145 147 150 152 155 157 160 162 165 168 170 173 175 178 180 183 185 188 190 193 196 198 201 203 206 208 211 213 216 219 221 224 226 229 231 234 236 239 241 244 247 249 252 254]
  DEBUG   PWM map of fan 'front': map[0:0 1:0 2:3 3:3 4:5 5:5 6:5 7:8 8:8 9:10 10:10 11:10 12:13 13:13 14:13 15:15 16:15 17:18 18:18 19:18 20:20 21:20 22:23 23:23 24:23 25:26 26:26 27:28 28:28 29:28 30:31 31:31 32:33 33:33 34:33 35:36 36:36 37:38 38:38 39:38 40:41 41:41 42:41 43:43 44:43 45:46 46:46 47:46 48:48 49:48 50:51 51:51 52:51 53:54 54:54 55:56 56:56 57:56 58:59 59:59 60:61 61:61 62:61 63:64 64:64 65:64 66:66 67:66 68:69 69:69 70:69 71:71 72:71 73:74 74:74 75:74 76:77 77:77 78:79 79:79 80:79 81:82 82:82 83:84 84:84 85:84 86:87 87:87 88:89 89:89 90:89 91:92 92:92 93:92 94:94 95:94 96:97 97:97 98:97 99:99 100:99 101:102 102:102 103:102 104:105 105:105 106:107 107:107 108:107 109:110 110:110 111:112 112:112 113:112 114:115 115:115 116:115 117:117 118:117 119:120 120:120 121:120 122:122 123:122 124:125 125:125 126:125 127:128 128:128 129:130 130:130 131:130 132:133 133:133 134:135 135:135 136:135 137:138 138:138 139:140 140:140 141:140 142:143 143:143 144:143 145:145 146:145 147:148 148:148 149:148 150:150 151:150 152:153 153:153 154:153 155:156 156:156 157:158 158:158 159:158 160:161 161:161 162:163 163:163 164:163 165:166 166:166 167:166 168:168 169:168 170:171 171:171 172:171 173:173 174:173 175:176 176:176 177:176 178:179 179:179 180:181 181:181 182:181 183:184 184:184 185:186 186:186 187:186 188:189 189:189 190:191 191:191 192:191 193:194 194:194 195:194 196:196 197:196 198:199 199:199 200:199 201:201 202:201 203:204 204:204 205:204 206:207 207:207 208:209 209:209 210:209 211:212 212:212 213:214 214:214 215:214 216:217 217:217 218:217 219:219 220:219 221:222 222:222 223:222 224:224 225:224 226:227 227:227 228:227 229:230 230:230 231:232 232:232 233:232 234:235 235:235 236:237 237:237 238:237 239:240 240:240 241:242 242:242 243:242 244:245 245:245 246:245 247:247 248:247 249:250 250:250 251:250 252:252 253:252 254:255 255:255]
 INFO  PWM settings of fan 'front': Min 5, Start 5, Max 99
 INFO  Starting controller loop for fan 'front'
 INFO  FanController: Using saved value for pwm map of Fan 'top'
  DEBUG   Distinct PWM value targets of fan top: [0 2 4 7 9 12 15 17 20 22 25 27 30 32 35 37 40 43 45 48 50 53 55 58 60 63 66 68 71 73 76 78 81 83 86 88 91 94 96 99 101 104 106 109 111 114 117 119 122 124 127 129 132 134 137 139 142 145 147 150 152 155 157 160 162 165 168 170 173 175 178 180 183 185 188 190 193 196 198 201 203 206 208 211 213 216 219 221 224 226 229 231 234 236 239 241 244 247 249 252 254]
  DEBUG   PWM map of fan 'top': map[0:0 1:0 2:3 3:3 4:5 5:5 6:5 7:8 8:8 9:10 10:10 11:10 12:13 13:13 14:13 15:15 16:15 17:18 18:18 19:18 20:20 21:20 22:23 23:23 24:23 25:26 26:26 27:28 28:28 29:28 30:31 31:31 32:33 33:33 34:33 35:36 36:36 37:38 38:38 39:38 40:41 41:41 42:41 43:43 44:43 45:46 46:46 47:46 48:48 49:48 50:51 51:51 52:51 53:54 54:54 55:56 56:56 57:56 58:59 59:59 60:61 61:61 62:61 63:64 64:64 65:64 66:66 67:66 68:69 69:69 70:69 71:71 72:71 73:74 74:74 75:74 76:77 77:77 78:79 79:79 80:79 81:82 82:82 83:84 84:84 85:84 86:87 87:87 88:89 89:89 90:89 91:92 92:92 93:92 94:94 95:94 96:97 97:97 98:97 99:99 100:99 101:102 102:102 103:102 104:105 105:105 106:107 107:107 108:107 109:110 110:110 111:112 112:112 113:112 114:115 115:115 116:115 117:117 118:117 119:120 120:120 121:120 122:122 123:122 124:125 125:125 126:125 127:128 128:128 129:130 130:130 131:130 132:133 133:133 134:135 135:135 136:135 137:138 138:138 139:140 140:140 141:140 142:143 143:143 144:143 145:145 146:145 147:148 148:148 149:148 150:150 151:150 152:153 153:153 154:153 155:156 156:156 157:158 158:158 159:158 160:161 161:161 162:163 163:163 164:163 165:166 166:166 167:166 168:168 169:168 170:171 171:171 172:171 173:173 174:173 175:176 176:176 177:176 178:179 179:179 180:181 181:181 182:181 183:184 184:184 185:186 186:186 187:186 188:189 189:189 190:191 191:191 192:191 193:194 194:194 195:194 196:196 197:196 198:199 199:199 200:199 201:201 202:201 203:204 204:204 205:204 206:207 207:207 208:209 209:209 210:209 211:212 212:212 213:214 214:214 215:214 216:217 217:217 218:217 219:219 220:219 221:222 222:222 223:222 224:224 225:224 226:227 227:227 228:227 229:230 230:230 231:232 232:232 233:232 234:235 235:235 236:237 237:237 238:237 239:240 240:240 241:242 242:242 243:242 244:245 245:245 246:245 247:247 248:247 249:250 250:250 251:250 252:252 253:252 254:255 255:255]
 INFO  PWM settings of fan 'top': Min 5, Start 5, Max 99
 INFO  Starting controller loop for fan 'top'
[... cut ...]
  DEBUG   Waiting for fan pump to settle (current RPM max diff: 58.000000)...
  DEBUG   Setting Fan PWM of 'back' to 26 ...
  DEBUG   Waiting for fan pump to settle (current RPM max diff: 58.000000)...
  DEBUG   Setting Fan PWM of 'back' to 28 ...
  DEBUG   Setting Fan PWM of 'front' to 28 ...
  DEBUG   Setting Fan PWM of 'top' to 28 ...
  DEBUG   Waiting for fan pump to settle (current RPM max diff: 58.000000)...
  DEBUG   Waiting for fan pump to settle (current RPM max diff: 58.000000)...
^C INFO  Received SIGTERM signal, exiting...
`

markusressel, 2022-10-24:
It looks like the RPM value of the pump is fluctuating within values of about 40 RPM, while the algorithm waits for a condition with a minimum fluctuation of 10 RPM. I thought about exposing this value to the configuration, but in reality the algorithm just isn't... lets say optimal yet :smile: 

Also, analyzing the RPM curve of a pump might not be necessary at all, so an option to skip the analysis part for a particular "fan" might also be an option.

I will relabel this issue as a "feature", since pumps weren't really considered when originally implementing fan2go.

ThisIsTheOnlyUsernameAvailable, 2022-10-25:
I think you undersold your forward-thinking, @markusressel.  You did expose that setting in the configuration file, and it fixed my issue.  For anyone else using a pump, I would recommend using the default setting for all of your fans, and track how far off the pump is from the expected value.  Then you can use the following config option to increase the fluctuation tolerance for the pump.

https://github.com/markusressel/fan2go/blob/master/fan2go.yaml#L10

markusressel, 2022-10-25:
@ThisIsTheOnlyUsernameAvailable Did you remove your comment or did GitHub mess up something?
I saw a notification about a comment from you, but I cannot find it anywhere :eyes:

markusressel, 2022-11-27:
I am closing this because of no activity. Feel free to reopen if necessary.

closed 2022-11-27 by markusressel