markusressel/fan2go #176: Config validation passes, then "curve list" spits out errors

zopieux, 2022-11-05:
**Describe the bug**

<details><summary>This config</summary>

```yaml
curves:
- id: c-disk-2
  linear:
    max: 37
    min: 30
    sensor: s-disk-2
- id: c-disk-3
  linear:
    max: 37
    min: 30
    sensor: s-disk-3
- id: c-disk-6
  linear:
    max: 37
    min: 30
    sensor: s-disk-6
- id: c-disk-7
  linear:
    max: 37
    min: 30
    sensor: s-disk-7
- id: c-core-0
  linear:
    max: 42
    min: 35
    sensor: s-core-0
- id: c-core-1
  linear:
    max: 42
    min: 35
    sensor: s-core-1
- id: c-core-2
  linear:
    max: 42
    min: 35
    sensor: s-core-2
- id: c-core-3
  linear:
    max: 42
    min: 35
    sensor: s-core-3
- curves:
  - c-disk-2
  - c-disk-3
  - c-disk-6
  - c-disk-7
  - c-core-0
  - c-core-1
  - c-core-2
  - c-core-3
  id: c-avg
  type: average
fans:
- curve: c-avg
  hwmon:
    index: 1
    platform: f71869a-isa-0
  id: f-chassis
  neverStop: false
sensors:
- hwmon:
    index: 1
    platform: drivetemp-scsi-2-0
  id: s-disk-2
- hwmon:
    index: 1
    platform: drivetemp-scsi-3-0
  id: s-disk-3
- hwmon:
    index: 1
    platform: drivetemp-scsi-6-0
  id: s-disk-6
- hwmon:
    index: 1
    platform: drivetemp-scsi-7-0
  id: s-disk-7
- hwmon:
    index: 1
    platform: coretemp-isa-0
  id: s-core-0
- hwmon:
    index: 2
    platform: coretemp-isa-0
  id: s-core-1
- hwmon:
    index: 3
    platform: coretemp-isa-0
  id: s-core-2
- hwmon:
    index: 4
    platform: coretemp-isa-0
  id: s-core-3
- hwmon:
    index: 1
    platform: soc_dts0-virtual-0
  id: s-soc-0
- hwmon:
    index: 1
    platform: soc_dts1-virtual-0
  id: s-soc-1
```
</details>

successfully validates when using `fan2go -c … config validate`, yet it clearly is invalid because I forgot to nest `c-avg` inside a `function` key, which only gets caught when running `fan2go -c … curve list`.

**To Reproduce**
Run the commands above using the yaml provided in the zippie.

**Expected behavior**
`config validate` spits out the same error as `curve list` instead of passing successfully.

**Screenshots**
n/a

**Desktop (please complete the following information):**
- `fan2go version`: 0.8.0

**Additional context**

markusressel, 2022-11-06:
Should work now, let me know if it doesn't!

closed 2022-11-06 by markusressel

zopieux, 2022-11-06:
Impressive reaction time, thanks a lot! I can't test right now, but I'll be sure to give it a try when this reaches a future release.

PR #178 (2022-11-06): fix config validation not working then specifying a config via -c
fixes #176