markusressel/fan2go #183: aur package is broken

roshal, 2022-12-04:
[image]

markusressel, 2022-12-04:
Its working for me, did you try a clean build?

roshal, 2022-12-04:
yes

`paru --clean`

`paru --clean --clean`

markusressel, 2022-12-05:
I don't think this is an issue with the package repo, because it has not changed in quite some time.

Seems like paru or your local git configuration has problems cloning the repo. Executing 

```
git clone https://github.com/markusressel/fan2go
```

works as expected. Also clean installing fan2go using `yay` works as expected.

roshal, 2022-12-05:
```gitconfig
[remote "origin"]

fetch = +refs/pull/*/head:refs/tags/pull/*
```

markusressel, 2022-12-05:
Cool, I will close this issue then 🚀

closed 2022-12-05 by markusressel