markusressel/fan2go #184: Fan initialization returning extremely low values on glitchy hardware

Anonymous1157, 2022-12-15:
**Describe the bug**
fan2go is detecting my fans incorrectly during the initialization sequence and sets the PWM values way too low. The offending fans are set to a start PWM of 0 and a max PWM between 30 and 70. The fans can never spin up with these values.

If I delete the db and reinitialize, some of the fans will detect correctly some of the time, a 60mm PWM chipset fan and a 120mm DC (3-pin) exhaust fan.

The fans it has not been able to detect correctly are SilverStone AP181, these are 180mm DC-controlled fans with heavy blades that have a relatively high minimum PWM and will kick and squeal if driven marginally. *They sometimes confuse my fan controller and their speed is not detected correctly until 500 RPM.* (See Screenshots section.) This doesn't affect the built-in UEFI fan controller because it doesn't care about the RPM value, it just reads temperature sensors and sets PWMs or DC voltages.

**To Reproduce**
1. Build a system with an MSI X570 Unify and an old SilverStone case.
2. Start fan2go normally.

**Expected behavior**
The offending fans should have a start PWM near 50 and a max PWM near 250. This would be consistent with the fan curves I was using in my UEFI fan controller. Maybe fan2go should ignore absurd outlier values in the middle of the RPM curve.

**Screenshots**
```
fan-intake-rear
               
  Start PWM  0 
  Max PWM    65

 6734 ┤                         ╭╮
 6285 ┤                         ││
 5836 ┤                         │╰╮
 5387 ┤                         │ │
 4938 ┤                         │ │
 4489 ┤                         │ │
 4040 ┤                         │ │
 3591 ┤                         │ │
 3142 ┤                         │ │
 2693 ┤                        ╭╯ │
 2245 ┤                        │  │
 1796 ┤                        │  │
 1347 ┤                        │  │
  898 ┤                        │  │                          ╭────────────────────────────────────────────
  449 ┤                        │  │      ╭───────────────────╯
    0 ┼────────────────────────╯  ╰──────╯
                                                    RPM / PWM

fan-intake-mid
               
  Start PWM  0 
  Max PWM    93

 26749 ┤                                   ╭╮
 24966 ┤                                   │╰╮
 23182 ┤                                   │ │
 21399 ┤                                   │ │
 19616 ┤                                   │ │
 17833 ┤                                   │ │
 16049 ┤                                   │ │
 14266 ┤                                   │ │
 12483 ┤                                   │ │
 10700 ┤                                   │ │
  8916 ┤                                   │ │
  7133 ┤                         ╭╮        │ │
  5350 ┤                         │╰╮       │ │
  3567 ┤                         │ ╰╮      │ │
  1783 ┤                         │  │      │ │                                          ╭──────────────────
     0 ┼─────────────────────────╯  ╰──────╯ ╰──────────────────────────────────────────╯
                                                     RPM / PWM

fan-intake-front
               
  Start PWM  0 
  Max PWM    69

 4765 ┤                          ╭╮
 4447 ┤                          ││
 4129 ┤                          ││
 3812 ┤                          ││
 3494 ┤                          ││
 3176 ┤                          ││
 2859 ┤                          ││
 2541 ┤                          ││
 2223 ┤                          ││
 1906 ┤                          ││
 1588 ┤                          ││
 1271 ┤                          ││
  953 ┤                          ││                                         ╭─────────────────────────────
  635 ┤                          ││            ╭────────────────────────────╯
  318 ┤                          ││      ╭─────╯
    0 ┼──────────────────────────╯╰──────╯
                                                    RPM / PWM

fan-exhaust
                
  Start PWM  0  
  Max PWM    251

 1236 ┤                                                                                          ╭────────
 1154 ┤                                                                                  ╭───────╯
 1071 ┤                                                                           ╭──────╯
  989 ┤                                                                    ╭──────╯
  906 ┤                                                             ╭──────╯
  824 ┤                                                      ╭──────╯
  742 ┤                                                 ╭────╯
  659 ┤                                             ╭───╯
  577 ┤                                             │
  494 ┤                                             │
  412 ┤                                             │
  330 ┤                                            ╭╯
  247 ┤                                            │
  165 ┤                                            │
   82 ┤                                            │
    0 ┼────────────────────────────────────────────╯
                                                    RPM / PWM

fan-chipset
                
  Start PWM  0  
  Max PWM    255

 5400 ┤                                                                                             ╭─────
 5040 ┤                                                                                  ╭──────────╯
 4680 ┤                                                                         ╭────────╯
 4320 ┤                                                                  ╭──────╯
 3960 ┤                                                         ╭────────╯
 3600 ┤                                                ╭────────╯
 3240 ┤                                          ╭─────╯
 2880 ┤                                   ╭──────╯
 2520 ┤                              ╭────╯
 2160 ┤                         ╭────╯
 1800 ┤                     ╭───╯
 1440 ┤                  ╭──╯
 1080 ┤               ╭──╯
  720 ┤            ╭──╯
  360 ┤            │
    0 ┼────────────╯
                                                    RPM / PWM
```

**Desktop**
- Distro: Gentoo ~amd64
- `uname -a`: `Linux sunset-desktop5 6.0.11-gentoo-dist #1 SMP PREEMPT_DYNAMIC Sun Dec 4 16:29:03 EST 2022 x86_64 AMD Ryzen 7 5800X 8-Core Processor AuthenticAMD GNU/Linux`
- `sensors -v`: `sensors version 3.6.0 with libsensors version 3.6.0`
- `fan2go version`: `dev` (running a community package from GURU which is like Arch's AUR, version 0.6.2)

**Additional context**
This is the point where I would manually configure my start and max PWM points and forget about it, if I had the option.

fan2go looks like it will do what I want, if only I could get it to detect my fans correctly, or just skip this step and use the full PWM range.

Anonymous1157, 2022-12-15:
Oops, this might be a duplicate of #104, I just used different words to describe the same problem!

stalkerg, 2022-12-22:
I need to check, but your Radeon chip probably has the same issue as I fixed here https://github.com/markusressel/fan2go/issues/148 basically, they have only virtual PWM.

markusressel, 2023-05-11:
Thx for the report, however there is very little fan2go can do about this. You can setup a custom pwm map and fan curve to still control the fan, but with such a limited amount of information, it will never be optimal.

Because of that I am closing this issue for now.

closed 2023-05-11 by markusressel

Anonymous1157, 2023-05-19:
There is actually one obvious thing fan2go could do about this. Is it possible to turn off the automatic detection and just work with the full range?

markusressel, 2023-05-19:
Simply provide `minPwm`, `startPwm` and `maxPwm` values via the config to skip the fan curve analysis entirely.