markusressel/fan2go #192: Curve functions: add, substract, multiply, divide

starsforeveryone, 2023-01-14:
Dear developers,

first of all: Thank you for creating this awesome piece of software! (so much better than the old "fancontrol"...)

I have an idea on how to make it even better.

In my case (I mean the actual metal case), because of the inverted mainboard tray and generally the way I've designed the hardware positioning, increasing the GPU airflow adds to the CPU airflow. This means that whenever the GPU core temperature rises above its threshold, the GPU fans start spinning, and pump additional air down into the bottom of the case. This has physical reasons because of the way ASUS designed my graphics card: It doesn't exhaust all the warm air via the back of the card, but a significant portion goes straight through the GPU cooler and exists the card on the other side via an opening in the backplate. This warm GPU exhaust air is then stuck in the lower part of my case, and has to be transported outside by the rear case fan. This creates said situation where the airflow from the GPU is physically added to the already existing airflow (CPU -> rear case fan), which creates a need to make the rear case fan speed up accordingly, in order to exhaust the greater air volume.

I therefore need a curve function "add" that would allow me to add the rpm from the gpu curve to the rear fan curve. So that whenever the GPU fans speed up, the rear fan speeds up as well, in order to get rid of the additional warm air.

Since general mathematical curve functions seem to be useful because they're so versatile / multi-purpose, I suggest implementing "substract", "multiply" and "divide" too. They might be useful to some people with more complex fan structures.

I would have implemented it myself, but sadly, I suck at programming. At least I could narrow it down: The new functions would have to be added to _internal/curves/functional.go_ inside the _switch_ structure from line 33 to 64. There seems to be some array mathematics going on there - this is where the new functions would have to be implemented.

Thank you very much!

markusressel, 2023-01-15:
Hey @starsforeveryone , glad you like the project! :heart: 

Thx for the detailed description :+1: 

In the case of the curve function, addition and substraction of other curve values could be implemented in a "sum" and "difference" function type. I am not sure how intuitive the usage of such functions would be due to the fact that the sum of two curves at 50% will already max out a sum curve based on them. The same is already the case for the `delta` function. I feel like it might be difficult to create useful configurations with those function types but since adding them should be not that hard we can simply give it a try and see if it works out.

I would really like some kind of UI for all of this to make it easier for the user to figure out what is going on, but I can't invest the time into developing it. Maybe some day the API will be used by an ambitious developer :crossed_fingers: 

I don't think multiply and divide doesn't make a whole lot of sense for now, since no use case has come up for them (yet), so there is need to implement anything like that. We can always add it when it becomes necessary.

markusressel, 2023-01-15:
@starsforeveryone I have merged #194, please update fan2go and give it a try :nerd_face:

starsforeveryone, 2023-01-20:
Thank you very much! :)

I tried to apply the sum function and it seems to work flawlessly!

The fact that the sum of two 50% curves will already create a 100% curve is actually exactly what I'd expect from a sum function. It should add curves A + B. If the given PWM values p are p(A) = 110 and p(B) = 145, the sum should be 255. This means the fan would rotate on max, but that's what the sum is.

closed 2023-01-20 by starsforeveryone

markusressel, 2023-01-20:
Perfect 👍 🥳