markusressel/fan2go #196: AMD RX 6800 GPU fan detection yields bad pwm values

starsforeveryone, 2023-01-20:
**Describe the bug**
When trying to use fan2go to control the GPU fan of my ASUS AMD RX 6800, auto detection yields bad pwm values. Fan2go also seems to think that a third party is controlling the fan when, in fact, it's fan2go itself who controls the fan.

**To Reproduce**
Steps to reproduce the behavior:

1. Create profile for fan in yaml config file.
2. Stop fan2go systemd service!
3. Stop any software that controls the GPU fan!
4. Delete fan2go database db file.
5. Start fan2go and let it auto-detect the pwm values of each fan.

Fan curve detection of all other fans works fine, but GPU fan curve detection is stopped when pwm = 102 is reached.
Output: ` INFO  PWM settings of fan 'fan_gpu': Min 0, Start 102, Max 102`

Fan2go then proceeds to ramp down the fan according to the set temperatures, but claims a third party software is controlling the GPU fan. Since no background services are running, this is not the case. You can even clearly see the small ramp-down stepping applied by fan2go - no vendor gpu curve would step down in small increments like that, I suppose.

[image]

**Expected behavior**
Fan2go should automatically test and detect all pwm fan curves, then write it to its database file.

**Desktop (please complete the following information):**
- Distro: Fedora Workstation 37
- `uname -a`: Linux *********-pc 6.2.0-rc4-next-20230120+ #3 SMP PREEMPT_DYNAMIC Fri Jan 20 20:33:24 CET 2023 x86_64 x86_64 x86_64 GNU/Linux (self-compiled kernel)
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: latest github version, self-compiled

markusressel, 2023-01-20:
Probably related to #148 (virtual pwm)

starsforeveryone, 2023-01-22:
I definitely think so. I've read the thread, but it's kinda technical. To summarize, is there a way to add support for correct generation of amdgpu fan curves?

markusressel, 2023-01-22:
Probably not. I think the most realistic way to solve this is adding the ability to define fan curves manually and ignoring the warnings about a third party. Currently the curves are only used for min and max speed, if you specify those in the config the curve analysis doesn't need to be correct. You will still get the warnings though and there is no way to disable them, since they are meant to be a safety feature.

starsforeveryone, 2023-01-22:
> Currently the curves are only used for min and max speed, if you specify those in the config the curve analysis doesn't need to be correct.

I believe this is the best solution/workaround, if it cannot be resolved otherwise.

I tried applying this in my config file, but could not find any reference on how it's done correctly. I only found info about how to set "min" and "max" values for a _curve_, in order to create a simple linear curve. These values are min/max temperatures to define the curve, not min/max pwm values to define the fan's capabilities.

It would be nice if you could give a short piece of advice on how to do it!

markusressel, 2023-01-22:
Please have a look at the [example config](https://github.com/markusressel/fan2go/blob/24c41d20a666f82a949756d40d7509fa1094b930/fan2go.yaml#L45).

In your case, specifying the `minPwm` (and optionally `startPwm`) and `maxPwm` value for each fan that was not analyzed correctly should help.

I have added an "advanced options" section to the README about this.

starsforeveryone, 2023-01-23:
Ha! Now it works! Awesome! 😎 👍

```
Workaround for the amdgpu problem:
- Disable all fan control softwares. Disable fan2go service, if you had it enabled: sudo systemctl stop fan2go.service
- Stop fan: sudo fan2go fan speed 0 -i YOUR_FAN_NAME
- Increase fan speed until all of the GPU fans barely start: sudo fan2go fan speed SPEED -i YOUR_FAN_NAME. Write this number down, then add some extra margin for safety to this value, to ensure the fans will always start reliably.
- Define "minPWM" (the value you determined) and "maxPWM" (usually 255) in the config file, as described in the README file. This has to be done in the "fan" section, since these are properties of the fan.
- Reinitialize the fan: sudo fan2go fan init -i YOUR_FAN_NAME
- Test the whole thing via: sudo fan2go
- (This step can be jumped, if you already had the service enabled): For automatic start of fan2go at boot, enable the fan2go service: sudo systemctl enable fan2go.service
- If testing showed that everything works as intended, start the fan2go service: sudo systemctl start fan2go.service
```

Thank you very much for extending the README file. This will be very helpful for beginners!

I very much hope that somebody will one day create a GUI for fan2go. The features and stability are amazing. Great work!

Lastly, maybe we could let fan2go output a hint / message for the user whenever it detects a "third party change" of an amdgpu-class fan. Then the user knows it could be a bug due to incomplete support of AMD graphic cards, with a link to instructions on how to use a workaround. Would make fan2go more user friendly, in my opinion?

closed 2023-01-23 by starsforeveryone

markusressel, 2023-01-23:
It's not a lack of support for AMD cards, it's their weird implementation that makes it impossible. I can add an FAQ entry about this, but I think we need to figure this out on a case by case basis.

starsforeveryone, 2023-01-23:
*takes notes*

Wishlist:
- Software developer who writes a GUI for fan2go
- Software developer from AMD who writes a commit for supporting their most common RX 6000 and RX 7000 cards 😆

Okay, and in the meantime, we have a workaround now! :) Again, thank you very much!

stalkerg, 2023-05-22:
> It's not a lack of support for AMD cards, it's their weird implementation that makes it impossible. 

It's not weird, 0-100 or 0-255 it's very common options, the main problem here what `sensors` API expects only 0-255, and not clear anywhere what to do if you do not have so many PWM levels on hardware/firmware. I can expect 1-128, 1-64 as well. Probably it can be a good proposal to provide a range for each control from the driver. 
Most important at least 50% of linux sensors emulate 0-255 and underneath support different ranges or behavior. 
Anyway, it need heavy work in the email list to improve such a situation.