markusressel/fan2go #198: High idle power draw / Increase in wakeup calls per second

starsforeveryone, 2023-01-24:
**Describe the bug**
When fan2go is active, it seems to increase the tick_sched_timer count by a factor of 4. This considerably increases the system's idle power consumption.

**To Reproduce**
Steps to reproduce the behavior:

Create a config with 200ms polling rates:
```
tempSensorPollingRate: 200ms
rpmPollingRate: 200ms
controllerAdjustmentTickRate: 200ms
```

Compare to an identical config with 2 second (2000ms) rates:
```
tempSensorPollingRate: 2s
rpmPollingRate: 2s
controllerAdjustmentTickRate: 2s
```

**Expected behavior**
When using "powertop" to look at system wakeup calls, the amount of system wakeup calls per second should be as low as possible. I remember it being as low as 40 wakeup calls per second on an old HP laptop on Debian back in 2009. I'm already shocked at it being 200-500 per second on modern distributions. The more wakeup calls, the higher the power draw. fan2go should have as low of a footprint on this as possible.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
- Distro: Fedora Workstation 37
- `uname -a`: Linux xxxxxxxxxx-pc 6.2.0 Fri Jan 20 20:33:24 CET 2023 x86_64 x86_64 x86_64 GNU/Linux (self-compiled kernel using standard Fedora kernel config)
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: latest github version (self-compiled)

**Additional context**
The only thing that was changed between the three screenshots was the fact that fan2go service was started (200ms, 2s timing rates) or not started (without fan control).

Fan2go config file (with .txt extension to make it uploadable here): 
[fan2go.yaml.txt](https://github.com/markusressel/fan2go/files/10492100/fan2go.yaml.txt)

Oh, and each time the hard drive temperature sensor gets polled, the hard drive clicks a little bit. No issue while in idle (hard drive spinning, but controller in power save mode), but clearly audible while active. Its two Seagate IronWolf Pro drives. Considering adding an external temp sensor to my mainboard to avoid this issue.

markusressel, 2023-01-25:
Hey @starsforeveryone, thx for investigating!

It would surely be great if we could reduce the battery impact of fan2go. I have not done much to optimize yet since I only use it on desktop machines where other tasks cause way more power consumption in comparison to fan2go which made it unattractive to improve it. If you find something that can be improved without compromising the existing functionality feel free to open a PR for this 👍 🤓

starsforeveryone, 2023-01-25:
I'm not a programmer, so sadly, I cannot contribute anything in the form of commits.

I did however do some testing. Here's what I found:

- Fan2go creates several background processes, not just one. I seem to have at least one or two more than I have fans.
- One of the background processes seems to be causing much more wakeup calls per second than all the others combined.
- Fan2go polls sensors and sends instructions every couple of hundred milliseconds. This increases response time, but at the cost of creating a lot of background work.
- All three related settings (tempSensorPollingRate, rpmPollingRate, controllerAdjustmentTickRate) seem to increase the number of wakeup calls per second quite equally, if set to equal values.
- Even if all three settings are set to absurdly high intervals (in order to reduce background activity), the amount of wakeup calls that fan2go causes (~100/sec) is still higher than my amdgpu driver (~60/sec) at idle. Which is a statement for itself. 🤭

My ideas would therefore be:

1. Go over the entire source code to reduce the amount of background work that is done by the code.
2. Reduce the amount of fan2go instances that are created.

starsforeveryone, 2023-01-25:
I'm not a programmer, so sadly, I cannot contribute anything in the form of commits.

I did however do some testing. Here's what I found:

- Fan2go creates several background processes, not just one. I seem to have at least one or two more than I have fans.
- One of the background processes seems to be causing much more wakeup calls per second than all the others combined.
- Fan2go polls sensors and sends instructions every couple of hundred milliseconds. This increases response time, but at the cost of creating a lot of background work.
- All three related settings (tempSensorPollingRate, rpmPollingRate, controllerAdjustmentTickRate) seem to increase the number of wakeup calls per second quite equally, if set to equal values.
- Increasing these numbers can cause the fans to not start reliably anymore. This seems to be caused by the fans getting too low of a PWM signal, therefore rotating slower than expected. _controllerAdjustmentTickRate_ seems to be the culprit here.
- Even if all three settings are set to absurdly high intervals (in order to reduce background activity), the amount of wakeup calls that fan2go causes (~100/sec) is still higher than my amdgpu driver (~60/sec) at idle. Which is a statement for itself. 🤭

My ideas would therefore be:
1. Go over the entire source code to reduce the amount of background work that is done by the code. Tedious, I know, but worth it!
2. Reduce the amount of fan2go instances that are created to the necessary minimum. Maybe there's some overhead here that can be cut down.
3. Set sensor/rpm polling and fan adjustment values to something much larger - more like seconds instead of milliseconds. Coolers nowadays are large enough to buffer that.
4. Then, adjust the standard fan behaviour in a way that the fans will still keep spinning as expected. Maybe there is a way to still give them enough PWM input without setting _controllerAdjustmentTickRate_ too high.

I recently caught the OpenRGB project causing a ton of wakeup calls per second, thus increasing power consumption by more than 10W in idle! (https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2989)
Their recommended solution was to close OpenRGB when not using animations. Very efficient. 🙄 Glad I'm not using animations and can set the whole thing to static lighting which only needs to be set once.

I'm also on a desktop PC, but with rising energy prices, every watt counts.

markusressel, 2023-01-27:
Since I wanted to take a look at [parca](https://github.com/parca-dev/parca) anyway I took a quick look at what takes the most time when running fan2go:

[image]

As you can see (with a simple configuration of two sensors and fans) simply reading and writing the files makes up almost all of the processing time. This leads me to believe that there is little we can improve about the code other than reducing the number of checks & updates per "time unit", which is already possible using the configuration options (and you also demonstrated this in your previous comment).

Otherwise it might be possible to reduce the wakeup count by aggregating fan pwm updates, since currently a controller instance is used for each fan, each of them running in a separate goroutine. This can probably lead to multiple wakeups when the fans are updated in very close successions instead of updating them all at once, leading to only a single wakup. Implementing this is far from trivial though since it would essentially require some kind of synchronization of all fan controllers across goroutines. This would very likely cause new edge cases that would need to be addressed in a way that suits all users. So even though this is a possible route to go, its a massive change of how fan2go works internally and I am not fond about increasing the code (and cognitive) complexity to _maybe_ save wakups in that matter.

starsforeveryone, 2023-01-27:
Thanks for investigating! That absolutely makes sense. We should reduce wakeup calls via lengthening the intervals _tempSensorPollingRate_, _rpmPollingRate_, and _controllerAdjustmentTickRate_. This is easy to do and yields the best results.

markusressel, 2023-05-11:
Since there is currently no idea on how we can improve this further, I am closing this issue for now. If anything new comes up, feel free to open a new issue. Thx!

closed 2023-05-11 by markusressel