markusressel/fan2go #211: Can't attach empty fan curve data to fan

Sir-Photch, 2023-03-10:
**Describe the bug**
I get the following output, even though I can confirm PWM control is working:

```console
$ doas fan2go
███████  █████  ███    ██ ██████   ██████   ██████
██      ██   ██ ████   ██      ██ ██       ██    ██
█████   ███████ ██ ██  ██  █████  ██   ███ ██    ██
██      ██   ██ ██  ██ ██ ██      ██    ██ ██    ██
██      ██   ██ ██   ████ ███████  ██████   ██████

 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
 INFO  Gathering sensor data for cpu...
 INFO  Gathering sensor data for gpu...
 INFO  Gathering sensor data for case...
 INFO  Loading fan curve data for fan 'gpu'...
 INFO  Loading fan curve data for fan 'case'...
 INFO  Loading fan curve data for fan 'cpu'...
 INFO  FanController: Using saved value for pwm map of Fan 'gpu'
 INFO  PWM settings of fan 'gpu': Min 0, Start 102, Max 102
 INFO  Starting controller loop for fan 'gpu'
 WARNING  Fan 'cpu' has not yet been analyzed, starting initialization sequence...
 INFO  FanController: Using saved value for pwm map of Fan 'cpu'
 INFO  Measuring RPM curve...
 WARNING  Fan 'case' has not yet been analyzed, starting initialization sequence...
 INFO  FanController: Using saved value for pwm map of Fan 'case'
 INFO  Measuring RPM curve...
  ERROR   Cant attach empty fan curve data to fan case
  ERROR   Failed to attach fan curve data to fan case: invalid argument
 INFO  Fan controller for fan case stopped.
 WARNING  Cannot send notification, unable to detect user of current display session
panic: invalid argument

goroutine 7 [running]:
github.com/markusressel/fan2go/internal.RunDaemon.func7()
        github.com/markusressel/fan2go/internal/backend.go:138 +0x17e
github.com/oklog/run.(*Group).Run.func1({0xc0003014c0?, 0xc0002b1938?})
        github.com/oklog/run@v1.1.0/group.go:38 +0x2f
created by github.com/oklog/run.(*Group).Run
        github.com/oklog/run@v1.1.0/group.go:37 +0x69
```

**To Reproduce**

fan2go.yaml
```yaml
sensors:
  - id: cpu-Tctl
    hwmon:
      platform: k10temp-pci-00c3
      index: 1

  - id: gpu-edge
    hwmon:
      platform: amdgpu-pci-02d00
      index: 1

fans:
  - id: cpu
    hwmon:
      platform: nct6687-isa-0a20
      index: 1
    neverStop: false
    curve: cpu-curve

  - id: gpu
    hwmon:
      platform: amdgpu-pci-02d00
      index: 1
    neverStop: false
    curve: gpu-curve

  - id: case
    hwmon:
      platform: nct6687-isa-0a20
      index: 6
    neverStop: true
    curve: case-curve

curves:
  - id: cpu-curve
    linear:
      sensor: cpu-Tctl
      steps:
        - 40: 0
        - 55: 50
        - 75: 155
        - 90: 255

  - id: gpu-curve
    linear:
      sensor: gpu-edge
      steps:
        - 50: 0
        - 80: 100
        - 90: 175
        - 100: 255

  - id: case-curve
    function:
      type: maximum
      curves:
        - cpu-curve
        - gpu-curve
```

```console
$ fan2go detect
> k10temp-pci-00c3
  Sensors  Index  Label                Value
           1      Tctl (temp1_input)   33875
           2      Tccd1 (temp3_input)  30750

> amdgpu-pci-02d00
  Fans     Index  Label   RPM   PWM  Auto
           1      hwmon0  1440  99   false
  Sensors  Index  Label                   Value
           1      edge (temp1_input)      38000
           2      junction (temp2_input)  40000
           3      mem (temp3_input)       42000

> nct6687-isa-0a20
  Fans     Index  Label          RPM  PWM  Auto
           1      CPU Fan        572  4    false
           2      Pump Fan       0    255  false
           3      System Fan #1  0    153  false
           4      System Fan #2  0    153  false
           5      System Fan #3  0    153  false
           6      System Fan #4  477  16   false
           7      System Fan #5  0    153  false
           8      System Fan #6  0    153  false
  Sensors  Index  Label                     Value
           1      CPU (temp1_input)         33000
           2      System (temp2_input)      34000
           3      VRM MOS (temp3_input)     39500
           4      PCH (temp4_input)         32000
           5      CPU Socket (temp5_input)  31000
           6      PCIe x1 (temp6_input)     30500
           7      M2_1 (temp7_input)        0

> nvme-pci-0100
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  34850
```

**Expected behavior**
fan2go to finish (or not to panic)

**Desktop (please complete the following information):**

- Distro: Arch Linux
- `uname -a`: `Linux archpc 6.2.2-zen2-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 08 Mar 2023 04:07:25 +0000 x86_64 GNU/Linux`
- `sensors -v`: `sensors version 3.6.0+git with libsensors version 3.6.0+git`
- `fan2go version`: `fan2go-git 0.8.0.r64.g52765ac-1`

Is this somehow caused by my config-setup?

markusressel, 2023-03-11:
Hmm either fan2go cannot measure the fan curve, or the saved data for the curve got somehow corrupted. Did you try deleting the db (not the config) and start over?

If fan2go cannot measure the fan curve, try to manually set min- start- and maxPwm values in the config to skip the measurement.

Sir-Photch, 2023-03-11:
I have deleted the database previously, yes. I have tried again and this time it seemed to be measuring, but it couldn't manage to finish after about 30mins, that's why I aborted that.

> If fan2go cannot measure the fan curve, try to manually set min- start- and maxPwm values in the config to skip the measurement.

How do I determine those values manually? I have some values by `fancontrol`, can I just use them?

```console
MINSTART=hwmon3/pwm1=150 hwmon0/pwm1=105 hwmon3/pwm6=150
MINSTOP=hwmon3/pwm1=0 hwmon0/pwm1=75 hwmon3/pwm6=0
```

markusressel, 2023-03-11:
Yes, the values determined by fancontrol also work for fan2go.

Not sure why the analysis is failing on fan2go when it seems to work on fancontrol.

markusressel, 2023-09-10:
Since I cannot investigate this without access to your exact hardware I will close this issue. Feel free to open a new issue if anything comes up. :heart:

closed 2023-09-10 by markusressel