markusressel/fan2go #216: Add ability to generate a config

zt64, 2023-03-25:
**Is your feature request related to a problem? Please describe.**
Currently the configuration is quite daunting to a newcomer like me. I feel it would be a welcome improvement to add the ability for fan2go to generate the config file.

**Describe the solution you'd like**
The detect command could maybe have an argument that generates a matching config file for it, or another command.

markusressel, 2023-03-25:
Hey @zt64, thx for reporting! :heart: 

Its a bit ironic that one of the reasons to build fan2go when I initially set out to do it was to get rid of the insane config fancontrol has :sweat_smile: But I totally get that the configuration of fan2go also has become quite complex, since it has grown a lot over time. The fact that the "example" config contains examples for _everything_ is probably also not helping, although this is intentional to aid in documentation, which makes it look a lot more complex than it actually is. Also, I want the documentation in the README to be a good enough guide to assist with the initial setup, but I can also see that even the README is probably daunting at first sight.

I am not sure if generating a config is the best solution for this, since there is pretty much no way to generate a complete configuration without user input. For user input to be possible fan2go would have to include some kind of "scripted process" that asks the user a bunch of stuff and creates the configuration in the end. However, thats the way fancontrol does this, and I am really not a big fan.

As I currently see it, what fan2go needs to get traction with a wider audience - where users doesn't want to fiddle with config files - is a GUI frontend. The terminal is great and I definitely don't want to get rid of it though. Because my time is limited, and I like to keep things modular and separate, thats why fan2go already provides a (currently pretty basic) REST API, which was meant to provide a way to other members of the community to provide frontends for fan2go. This would even allow for different frontends to be developed independently, f.ex. a TUI frontend, a JavaScript frontend and a ChatGPT frontend (who knows these days :laughing:).

My goal with this is to keep fan2go as the "backend tool", providing only minimal CLI tools to interact with it (and maybe even these should become a CLI "frontend" at some point) and therefore limit the scope of this repository at a manageble level.

What is currently missing is someone from the community, who is willing to get this ball rolling by starting to write a frontend. Sure, the current API provided by fan2go is nowhere near complete enough to create a frontend from it, and we probably have to redesign how fan2go stores and applies its configuration at some point when the API can provide "write operations", but I am happy to assist with this on the fan2go side.

zt64, 2023-03-25:
I agree with the points you make, a GUI for sure would be a good feature. fan2go is a lot nicer than fancontrol I feel and I'm glad that I found it. Maybe when the REST API is more complete I might be interested in making a GUI. Thank you for fan2go and for the quick reply

markusressel, 2025-05-08:
Hey @zt64 , since I won't be implementing this, I will close this issue for now.
If a "real" GUI is created by someone, that wants to modify the configuration via API calls, we can open a new issue to implement required API endpoints.
Thx!

closed 2025-05-08 by markusressel