markusressel/fan2go #220: Consider relicense (AGPL is problematic)

jfroy, 2023-04-09:
The AGPL is problematic for businesses (difficulty to legally comply is the main cited reason). Would you be open to relicensing under the GPL (or even a more permissive license e.g. MIT, Apache 2)? Every contributor would of course need to agree. I am not trying to start a flame war and this issue should be closed the moment it turns into one.

markusressel, 2023-04-09:
Hey @jfroy thx for opening an issue :heart: 

fan2go is a daemon that runs locally on a device, so the AGPL should pose the same requirements as the GPL would. I only used AGPL because I wanted a restrictive license and didn't want to think about edge cases. So I don't really see the a reason to switch from AGPL to GPL as it wouldn't really change much (I think).

Switching to something more permissive is another story, and this will not happen. fan2go is a pet project and meant to be used by private individuals. The license is restrictive on purpose to ensure any progress is floating back to the community (in some way or another) and not kept private. Contributing to fan2go should not be a problem, even for companies. If a company wants to use fan2go, its also free to do so, but modifications have to be made public. Maybe I am missing something but I really don't understand why companies fear restrictive licenses so much - but its not relevant to me here.

Still, I am curious why you think a license change would benefit the project?

markusressel, 2023-04-25:
As I have no intention to change the license, I willl close this issue.

closed 2023-04-25 by markusressel