markusressel/fan2go #224: Validation failed: Fan cpu: invalid index, must be >= 1

sergeolkhovik, 2023-05-11:
**Describe the bug**
Hi! Trying to jump off of fancontrol and found this project interesting. Trying to setup some basic configuration, created config file and then validation fails:

```
% sudo fan2go config validate  
 INFO  Using configuration file at: /etc/fan2go/fan2go.yaml
  ERROR   Validation failed: Fan cpu: invalid index, must be >= 1
```

Unfortunately can't find anything about this error (simply google with 'fan2go "invalid index"' gives no results).
Adding more info below.
Can anyone please help me what's wrong with the configuration in order to go on?

**To Reproduce**
Here's detect output:

```
% sudo fan2go detect          
> w83627dhg-isa-0
  Fans     Index  Label   RPM   PWM  Auto 
           1      hwmon2  1607  140  true 
           2      hwmon2  1308  70   false
           3      hwmon2  1125  255  false
  Sensors  Index  Label                 Value
           1      hwmon2 (temp1_input)  42000
           2      hwmon2 (temp2_input)  45000
           3      hwmon2 (temp3_input)  9000 

> nouveau-pci-0100
  Sensors  Index  Label                 Value
           1      hwmon0 (temp1_input)  44000

> coretemp-isa-0
  Sensors  Index  Label                 Value
           1      Core 0 (temp2_input)  49000
           2      Core 1 (temp3_input)  46000
           3      Core 2 (temp4_input)  51000
           4      Core 3 (temp5_input)  48000
```

and this is my configuration file (just compiled it based on the readme):

```
% cat /etc/fan2go/fan2go.yaml                                                                            
# A list of fans to control                                                                              
fans:                                                                                                    
  # A user defined ID.
  # Used for logging only
  - id: cpu  
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is                                                          
      # connected to this fan (see sensor.platform below)                                                
      platform: w83627dhg-isa-0
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 2
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM        
      pwmChannel: 2
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true                              
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve                                                                                     
        # (Optional) Override for the lowest PWM value at which the
    # fan is able to maintain rotation if it was spinning previously.
    minPwm: 70           
    # (Optional) Override for the lowest PWM value at which the    
    # fan will still be able to start rotating.
    # Note: Settings this to a value that is too small
    #       may damage your fans. Use at your own risk!
    startPwm: 65
    # (Optional) Override for the highest PWM value which still yields
    # an increased rotational speed compared to lower values.
    # Note: you can also use this to limit the max speed of a fan.
    maxPwm: 94
    # (Optional) Override for the PWM map used internally by fan2go for
    # mapping the "normal" 0-255 value range to values supported by this fan.
    # This can be used to compensate for a very limited set of supported values
    # (f.ex. off, low, high). If not set manually, the map will be computed
    # automatically by fan2go during fan initialization.
    pwmMap:
      0: 0
      64: 128
      192: 255

# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: w83627dhg-isa-0
      # The index of this sensor as displayed by `fan2go detect`
      index: 2

# A list of control curves which can be utilized by fans
# or other curves
curves:
  # A user defined ID, which is used to reference
  # a curve in a fan configuration (see above)
  - id: cpu_curve
    # The type of curve configuration, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Steps to define a section-wise defined speed curve function
      steps:
        # Sensor value -> Speed (0-255)
        - 40: 0
        - 50: 50
        - 80: 255
```

I'm not sure but looks like it doesn't see platform and other settings in fans / hwmon section, the same error happens if I even put xplatorm instead of platform.
And I don't know how to debug this?

**Desktop (please complete the following information):**

- Distro: Ubuntu
- `uname -a`: Linux htpc 6.2.0-20-generic # 20-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr  6 07:48:48 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: dev (but I remember I downloaded 0.8.0, file is 18942624 26.10.2022 fan2go-linux-amd64)

markusressel, 2023-05-11:
The config has changed since 0.8.0, update to latest and it should work 😊

sergeolkhovik, 2023-05-11:
Ah.. I checked sources of 0.8.0 and see this in the yaml file:

```
# A list of fans to control
fans:
  # A user defined ID.
  # Used for logging only
  - id: cpu
    # The type of fan configuration
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6798-isa-0
      # The index of this fan as displayed by `fan2go detect`
      index: 1

```

OK, it's getting clear, thanks!
Will try tomorrow or on weekend.

markusressel, 2023-05-16:
@sergeolkhovik I will close this for now, feel free to reopen if its still not working.
If you encounter another problem you cannot solve yourself, feel free to open another issue about it.
Thx!

closed 2023-05-16 by markusressel