markusressel/fan2go #227: How does the fan curve set the speed according to the temperature?

bluehysteric, 2023-05-15:
How does the fan curve set the speed according to the temperature?

markusressel, 2023-05-16:
Hey @bluehysteric !

Please have a look at the README, if thats not enough, please specify what exactly it is you are trying to understand.

bluehysteric, 2023-05-17:
> Hey @bluehysteric !
> 
> Please have a look at the README, if thats not enough, please specify what exactly it is you are trying to understand.

Sorry, README.md didn't make much sense to me. I want the fan to control the speed according to the temperature of a certain sensor, for example, set the cpu temperature to 30 50 60 degrees Celsius when the fan speed is 40% 60% 80% respectively, such rules fan2go can do?
Looking forward to your answer。

markusressel, 2023-05-17:
Why is the "Curves" Section in the README not enough?
https://github.com/markusressel/fan2go#curves

> Under `curves:` you need to define a list of fan speed curves, which represent the speed of a fan based on one or more temperature sensors.

bluehysteric, 2023-05-18:
> 为什么 README 中的“曲线”部分不够用？ https://github.com/markusressel/fan2go#curves
> 
> > `curves:`您需要在下面定义一个风扇速度曲线列表，这些曲线表示基于一个或多个温度传感器的风扇速度。
> 
> > [image][image]GitHub**[fan2go, 一个简单的守护进程，提供基于温度传感器的动态风扇速度控制，下载fan2go的源码_GitHub_帮酷](https://github.com/markusressel/fan2go)**一个简单的守护进程，提供基于温度传感器的动态风扇速度控制。- GitHub - markusressel/fan2go: 一个简单的守护进程，提供基于温度传感器的动态风扇速度控制。

I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature
`# A list of fans to control
fans:
  - id: CpuFan
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 2
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 2
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 2
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Air intake
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 3
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 3
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 3
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Out of the wind
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 1
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 1
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    #curve: cpu_curve
    maxRpm: 100
# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 2
  - id: system_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  - id: cpu_curve
    # The type of the curve, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Sensor input value at which the curve is at minimum speed
      #min: 30
      # Sensor input value at which the curve is at maximum speed
      #max: 80
      rpm:
        # Sensor value -> Speed (in pwm)
        - 30: 40
        - 50: 70
        - 60: 90`

bluehysteric, 2023-05-18:
> Why is the "Curves" Section in the README not enough? https://github.com/markusressel/fan2go#curves
> 
> > Under `curves:` you need to define a list of fan speed curves, which represent the speed of a fan based on one or more temperature sensors.
> 
> > [image][image] GitHub**[GitHub - markusressel/fan2go: A simple daemon providing dynamic fan speed control based on temperature sensors.](https://github.com/markusressel/fan2go)**A simple daemon providing dynamic fan speed control based on temperature sensors. - GitHub - markusressel/fan2go: A simple daemon providing dynamic fan speed control based on temperature sensors.

I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature

> # A list of fans to control
fans:
  - id: CpuFan
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 2
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 2
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 2
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Air intake
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 3
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 3
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 3
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Out of the wind
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 1
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 1
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    #curve: cpu_curve
    maxRpm: 100
# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 2
  - id: system_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  - id: cpu_curve
    # The type of the curve, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Sensor input value at which the curve is at minimum speed
      #min: 30
      # Sensor input value at which the curve is at maximum speed
      #max: 80
      rpm:
        # Sensor value -> Speed (in pwm)
        - 30: 40
        - 50: 70
        - 60: 90

bluehysteric, 2023-05-18:
> Why is the "Curves" Section in the README not enough? https://github.com/markusressel/fan2go#curves

I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature
`
I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature
`

bluehysteric, 2023-05-18:
> Why is the "Curves" Section in the README not enough? https://github.com/markusressel/fan2go#curves

I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature

`
# A list of fans to control
fans:
  - id: CpuFan
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 2
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 2
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 2
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Air intake
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 3
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 3
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 3
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Out of the wind
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 1
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 1
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    #curve: cpu_curve
    maxRpm: 100
# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 2
  - id: system_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  - id: cpu_curve
    # The type of the curve, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Sensor input value at which the curve is at minimum speed
      #min: 30
      # Sensor input value at which the curve is at maximum speed
      #max: 80
      rpm:
        # Sensor value -> Speed (in pwm)
        - 30: 40
        - 50: 70
        - 60: 90
`

bluehysteric, 2023-05-18:
```
> Hey @bluehysteric !
> 
> Please have a look at the README, if thats not enough, please specify what exactly it is you are trying to understand.
```

bluehysteric, 2023-05-18:
> Hey @bluehysteric !
> 
> Please have a look at the README, if thats not enough, please specify what exactly it is you are trying to understand.

I think it should be adequate, I just don't quite understand it. Here is my configuration, can you help me see what is wrong, the configuration works fine, but the fans are all running at 100%. I don't quite understand the pwm so I need to be adjusting the rpm based on temperature

```
# A list of fans to control
fans:
  - id: CpuFan
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 2
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 2
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 2
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Air intake
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 3
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 3
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 3
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    curve: cpu_curve
  - id: Out of the wind
    # The type of fan configuration, one of: hwmon | file
    hwmon:
      # The platform of the controller which is
      # connected to this fan (see sensor.platform below)
      platform: nct6792-isa-0290
      # The index of this fan as displayed by `fan2go detect`
      index: 1
      # The channel of this fan's RPM sensor as displayed by `fan2go detect`
      rpmChannel: 1
      # The pwm channel that controls this fan; fan2go defaults to same channel number as fan RPM
      pwmChannel: 1
    # Indicates whether this fan should never stop rotating, regardless of
    # how low the curve value is
    neverStop: true
    # The curve ID that should be used to determine the
    # speed of this fan
    #curve: cpu_curve
    maxRpm: 100
# A list of sensors to monitor
sensors:
  # A user defined ID, which is used to reference
  # a sensor in a curve configuration (see below)
  - id: cpu_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 2
  - id: system_package
    # The type of sensor configuration, one of: hwmon | file | cmd
    hwmon:
      # A regex matching a controller platform displayed by `fan2go detect`, f.ex.:
      # "coretemp", "it8620", "corsaircpro-*" etc.
      platform: nct6792-isa-0290
      # The index of this sensor as displayed by `fan2go detect`
      index: 1
curves:
  - id: cpu_curve
    # The type of the curve, one of: linear | function
    linear:
      # The sensor ID to use as a temperature input
      sensor: cpu_package
      # Sensor input value at which the curve is at minimum speed
      #min: 30
      # Sensor input value at which the curve is at maximum speed
      #max: 80
      rpm:
        # Sensor value -> Speed (in pwm)
        - 30: 40
        - 50: 70
        - 60: 90
```

markusressel, 2023-05-18:
There is no key "rpm" fir the linear curve configuration, you probably want to use "steps". Please try the `validate` command to see wether your configuration is valid or not.

Currently it's not possible in fan2go to configure a curve based on RPM, it's always PWM. PWM is basically the speed in percent but from 0-255. So to get 80 % speed you do 0.8 * 255. To specify the range fan2go can operate the fan in you can further limit the range (of 0-255) by specifying the minPwm and maxPwm config options on the fan object based on observations. 0 will then correspond to the minPwm value and 255 to the maxPwm value.

bluehysteric, 2023-05-18:
> There is no key "rpm" for the linear curve configuration, you probably want to use "steps". Please try the `validate` command to see wether your configuration is valid or not.
> 
> Currently it's not possible in fan2go to configure a curve based on RPM, it's always PWM. PWM is basically the speed in percent but from 0-255. So to get 80 % speed you do 0.8 * 255. To specify the range fan2go can operate the fan in you can further limit the range (of 0-255) by specifying the minPwm and maxPwm config options on the fan object based on observations. 0 will then correspond to the minPwm value and 255 to the maxPwm value.

Okay, got it.

closed 2023-05-23 by markusressel