markusressel/fan2go #231: Unable to run as configured

bluehysteric, 2023-05-22:
Information about the run:
```
May 22 17:28:28 009 fan2go[661]: INFO: Loading fan curve data for fan 'Air intake'...
May 22 17:28:28 009 fan2go[661]: INFO: FanController: Using saved value for pwm map of Fan 'Out of the wind'
May 22 17:28:28 009 fan2go[661]: INFO: PWM settings of fan 'Out of the wind': Min 0, Start 0, Max 11
May 22 17:28:28 009 fan2go[661]: INFO: Starting controller loop for fan 'Out of the wind'
May 22 17:28:28 009 fan2go[661]: INFO: FanController: Using saved value for pwm map of Fan 'CpuFan'
May 22 17:28:28 009 fan2go[661]: INFO: PWM settings of fan 'CpuFan': Min 0, Start 0, Max 254
May 22 17:28:28 009 fan2go[661]: INFO: Starting controller loop for fan 'CpuFan'
May 22 17:28:29 009 fan2go[661]: INFO: FanController: Using saved value for pwm map of Fan 'Air intake'
May 22 17:28:29 009 fan2go[661]: INFO: PWM settings of fan 'Air intake': Min 0, Start 0, Max 28
May 22 17:28:29 009 fan2go[661]: INFO: Starting controller loop for fan 'Air intake'
```
yaml：
```
# A list of fans to control
fans:
  - id: Out of the wind
    hwmon:
      platform: nct6792-isa-0290
      #pwmChannel: 1
      #rpmChannel: 1
      index: 1
    neverStop: true
    curve: OutFan
  - id: CpuFan
    hwmon:
      platform: nct6792-isa-0290
      #pwmChannel: 2
      #rpmChannel: 2
      index: 2
      
    neverStop: true
    curve: CpuFan
  - id: Air intake
    hwmon:
      platform: nct6792-isa-0290
      #pwmChannel: 3
      #rpmChannel: 3
      index: 3
    neverStop: true
    curve: AirFan
sensors:
  - id: CPUTIN
    hwmon:
      platform: nct6792-isa-0290
      index: 2

curves:
  - id: CpuFan
    linear:
      sensor: CPUTIN
      steps:
        - 30: 77 
        - 45: 115
        - 55: 140
        - 60: 153
        - 75: 255
  - id: AirFan
    linear:
      sensor: CPUTIN
      steps:
        - 45: 128
        - 57: 204
        - 65: 230
        - 70: 250
  - id: OutFan
    linear:
      sensor: CPUTIN
      steps:
        - 30: 150
        - 43: 255

api:
  # Whether to enable the API or not
  enabled: false
  # The host to listen for connections
  host: localhost
  # The port to listen for connections
  port: 9002
```

markusressel, 2023-05-22:
There is no error or problem described anywhere... ?

bluehysteric, 2023-05-23:
> There is no error or problem described anywhere... ?
Yes, there is no error message
```
❯ journalctl -u fan2go -f
May 23 09:33:32 009 fan2go[676]: INFO: Loading fan curve data for fan 'Out of the wind'...
May 23 09:33:32 009 fan2go[676]: INFO: FanController: Using saved value for pwm map of Fan 'CpuFan'
May 23 09:33:32 009 fan2go[676]: INFO: PWM settings of fan 'CpuFan': Min 0, Start 0, Max 254
May 23 09:33:32 009 fan2go[676]: INFO: Starting controller loop for fan 'CpuFan'
May 23 09:33:32 009 fan2go[676]: INFO: FanController: Using saved value for pwm map of Fan 'Air intake'
May 23 09:33:32 009 fan2go[676]: INFO: PWM settings of fan 'Air intake': Min 0, Start 0, Max 28
May 23 09:33:32 009 fan2go[676]: INFO: Starting controller loop for fan 'Air intake'
May 23 09:33:32 009 fan2go[676]: INFO: FanController: Using saved value for pwm map of Fan 'Out of the wind'
May 23 09:33:32 009 fan2go[676]: INFO: PWM settings of fan 'Out of the wind': Min 0, Start 0, Max 11
May 23 09:33:32 009 fan2go[676]: INFO: Starting controller loop for fan 'Out of the wind'
```

bluehysteric, 2023-05-23:
> So... what do you expect me to do then? If there is no error, and no problem as well, I will close this issue.

Fan speed can not run according to my profile, the fan is always only about 300rpm

closed 2023-09-10 by markusressel