markusressel/fan2go #237: Fan stopped even neverStop is set to true

sergeolkhovik, 2023-06-10:
**Describe the bug**
This is in addition to https://github.com/markusressel/fan2go/issues/153#issuecomment-1557646367, I did more tests and have a verbose log for reviewing.

So, now I have a configuration of 3 fans, 3 sensors (pretty old HTPC based on Asus P5K SE and Intel(R) Xeon(R) CPU E5440 @ 2.83GHz), the main one is CPU and it has 'neverStop: true' option set. I remove DB file and run this very nice fan2go tool in verbose mode, it runs calibration (and CPU fan gets stopped during this phase which is OK, I understand that), but then starts normal mode and during this phase I see the fan is stopped. In the logs I see:

```
 INFO  FanController: Using saved value for pwm map of Fan 'cpu'
  DEBUG   Distinct PWM value targets of fan cpu: [0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56>
  DEBUG   PWM map of fan 'cpu': map[0:0 1:1 2:2 3:3 4:4 5:5 6:6 7:7 8:8 9:9 10:10 11:11 12:12 13:13 14:14 15:15 16:16 17:17 18:18 19:19 20:20 21:21 22:22 23:23 24:24 25:25 26:26 27:27 28:28 29:29 30:30 31:31 32>
 INFO  PWM settings of fan 'cpu': Min 65, Start 70, Max 94
 INFO  Starting controller loop for fan 'cpu'
  DEBUG   Measuring RPM of gpu at PWM 222: 1670
  DEBUG   Measured RPM of 1670 at PWM 222 for fan gpu
  DEBUG   Setting Fan PWM of 'gpu' to 223 ...
  DEBUG   Setting Fan PWM of 'cpu' to 250 ...
  DEBUG   Setting Fan PWM of 'cpu' to 240 ...
  DEBUG   Setting Fan PWM of 'cpu' to 230 ...
  DEBUG   Setting Fan PWM of 'cpu' to 220 ...
  DEBUG   Setting Fan PWM of 'cpu' to 212 ...
  DEBUG   Setting Fan PWM of 'cpu' to 204 ...
  DEBUG   Setting Fan PWM of 'cpu' to 196 ...
  DEBUG   Setting Fan PWM of 'cpu' to 188 ...
  DEBUG   Setting Fan PWM of 'cpu' to 180 ...
  DEBUG   Measuring RPM of gpu at PWM 223: 1670
  DEBUG   Measured RPM of 1670 at PWM 223 for fan gpu
  DEBUG   Setting Fan PWM of 'gpu' to 224 ...
  DEBUG   Setting Fan PWM of 'cpu' to 174 ...
  DEBUG   Setting Fan PWM of 'cpu' to 168 ...
  DEBUG   Setting Fan PWM of 'cpu' to 162 ...
  DEBUG   Setting Fan PWM of 'cpu' to 156 ...
  DEBUG   Setting Fan PWM of 'cpu' to 150 ...
  DEBUG   Setting Fan PWM of 'cpu' to 144 ...
  DEBUG   Setting Fan PWM of 'cpu' to 138 ...
  DEBUG   Setting Fan PWM of 'cpu' to 134 ...
  DEBUG   Setting Fan PWM of 'cpu' to 130 ...
  DEBUG   Setting Fan PWM of 'cpu' to 126 ...
  DEBUG   Measuring RPM of gpu at PWM 224: 1670
  DEBUG   Measured RPM of 1670 at PWM 224 for fan gpu
  DEBUG   Setting Fan PWM of 'gpu' to 225 ...
  DEBUG   Setting Fan PWM of 'cpu' to 122 ...
  DEBUG   Setting Fan PWM of 'cpu' to 118 ...
  DEBUG   Setting Fan PWM of 'cpu' to 114 ...
  DEBUG   Setting Fan PWM of 'cpu' to 110 ...
  DEBUG   Setting Fan PWM of 'cpu' to 106 ...
  DEBUG   Setting Fan PWM of 'cpu' to 102 ...
  DEBUG   Setting Fan PWM of 'cpu' to 98 ...
  DEBUG   Setting Fan PWM of 'cpu' to 96 ...
  DEBUG   Setting Fan PWM of 'cpu' to 94 ...
  DEBUG   Setting Fan PWM of 'cpu' to 92 ...
  DEBUG   Measuring RPM of gpu at PWM 225: 1687
  DEBUG   Measured RPM of 1687 at PWM 225 for fan gpu
  DEBUG   Setting Fan PWM of 'gpu' to 226 ...
  DEBUG   Setting Fan PWM of 'cpu' to 90 ...
  DEBUG   Setting Fan PWM of 'cpu' to 88 ...
  DEBUG   Setting Fan PWM of 'cpu' to 86 ...
  DEBUG   Setting Fan PWM of 'cpu' to 84 ...
  DEBUG   Setting Fan PWM of 'cpu' to 82 ...
  DEBUG   Setting Fan PWM of 'cpu' to 80 ...
  DEBUG   Setting Fan PWM of 'cpu' to 78 ...
  DEBUG   Setting Fan PWM of 'cpu' to 76 ...
  DEBUG   Setting Fan PWM of 'cpu' to 74 ...
  DEBUG   Setting Fan PWM of 'cpu' to 72 ...
  DEBUG   Measuring RPM of gpu at PWM 226: 1687
  DEBUG   Measured RPM of 1687 at PWM 226 for fan gpu
  DEBUG   Setting Fan PWM of 'gpu' to 227 ...
  DEBUG   Setting Fan PWM of 'cpu' to 70 ...
  DEBUG   Setting Fan PWM of 'cpu' to 68 ...
  DEBUG   Setting Fan PWM of 'cpu' to 66 ...
  DEBUG   Setting Fan PWM of 'cpu' to 64 ...
  DEBUG   Setting Fan PWM of 'cpu' to 62 ...
  DEBUG   Setting Fan PWM of 'cpu' to 61 ...
```

while I have this setting for CPU fan;
```
    minPwm: 65
```

**Desktop (please complete the following information):**

- Distro: Ubuntu 23.04
- `uname -a`: Linux htpc 5.19.0-41-generic #42-Ubuntu SMP PREEMPT_DYNAMIC Mon Apr 17 19:17:03 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux (I had to downgrade for wakeonlan)
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: dev (18942624, 26.10.2022)

**Additional context**
Both config and full log of calibration and normal run attached.
[fan2go.zip](https://github.com/markusressel/fan2go/files/11713425/fan2go.zip)

PS: anyway the system works much stable than with fancontrol, I'm very happy that I found your tool!

markusressel, 2023-10-01:
Is this only an issue on startup? 
Does the fan actually spin down and come to a full stop during this?

hostlund, 2024-09-07:
I had the same issue, it seems that the min and start values changed after i deleted the database file:

Before DB delete
[image]

And after
[image]

I did setup fan2go on kernel 6.10 and everything worked as it should, but I'm now running 6.11.