markusressel/fan2go #239: curve input for linear

haudiweg, 2023-06-18:
I want to controll my fans with a temp difference (difference from air input and water temp) combined with linear (slow speed on 2C difference /full speed on 20C diference)

pls allow curves as sensors

i use water cooling in my pcs and have water, air input and air output temperatur sensors

markusressel, 2023-06-18:
This is already possible. Take a look at the `difference` function for a curve.

haudiweg, 2023-06-18:
were is the doku for the difference function?
(i know about the difference function but not how to use the output of difference function to feed a linear step function with it)

haudiweg, 2023-08-20:
Ok looked at the code,
Forked repo and added the feature in.
But im not sure if the math is right (i never programmed in go)

markusressel, 2023-09-10:
Closing this for now. If you come up with a proposal on how to integrate this properly upstream, feel free to create a new issue. :heart:

closed 2023-09-10 by markusressel