markusressel/fan2go #244: wrong data with cmd fan

piotr-sikora-v, 2023-07-20:
I set everything, and when checking is good... but when I try to run nothing is good ;)
Check:
```
fan2go  -c fan2go-min.yaml fan --id cmd_fan speed
50
fan2go  -c fan2go-min.yaml fan --id cmd_fan rpm
RPM: 9000
```
And start daemon 
```
fan2go -c etc/fan2go/fan2go-min.yaml 

 INFO  Using configuration file at: /etc/fan2go/fan2go-min.yaml
 INFO  Gathering sensor data for cmd_fan...
 INFO  Loading fan curve data for fan 'cmd_fan'...
 INFO  FanController: Using saved value for pwm map of Fan 'cmd_fan'
 INFO  PWM settings of fan 'cmd_fan': Min 0, Start 1, Max 255
 INFO  Starting controller loop for fan 'cmd_fan'
 WARNING  WARNING: Increasing minPWM of cmd_fan from 0 to 1, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 1 to 2, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 2 to 3, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 3 to 4, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 4 to 5, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 5 to 6, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 6 to 7, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 7 to 8, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 8 to 9, which is supposed to never stop, but RPM is 0
 WARNING  WARNING: Increasing minPWM of cmd_fan from 9 to 10, which is supposed to never stop, but RPM is 0
...
  ERROR   CRITICAL: Fan cmd_fan avg. RPM is 0, even at PWM value 255
  ERROR   CRITICAL: Fan cmd_fan avg. RPM is 0, even at PWM value 255
  ERROR   CRITICAL: Fan cmd_fan avg. RPM is 0, even at PWM value 255
  ERROR   CRITICAL: Fan cmd_fan avg. RPM is 0, even at PWM value 255
  ERROR   CRITICAL: Fan cmd_fan avg. RPM is 0, even at PWM value 255
```

BTW for `cmd_fan` min value is set 10 and start to 30

rpm read is bash script that do echo 9000 and nothing elese.

markusressel, 2023-10-22:
Please post the contents of `/etc/fan2go/fan2go-min.yaml` as well as all other relevant files, like f.ex. the bash script.

markusressel, 2024-04-05:
Closing due to inactivity. Feel free to reopen if you have something new.

closed 2024-04-05 by markusressel