markusressel/fan2go #245: I didn't see the channel column

skill7899, 2023-07-24:
**Describe the bug**
A clear and concise description of what the bug is.
when i run fan2go detect，I didn't see the channel column

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

root@pve:~# fan2go detect
> coretemp-isa-0
  Sensors  Index  Label                       Value
           1      Package id 0 (temp1_input)  49000
           2      Core 0 (temp2_input)        46000
           3      Core 4 (temp3_input)        43000
           4      Core 8 (temp4_input)        45000
           5      Core 12 (temp5_input)       45000
           6      Core 16 (temp6_input)       47000
           7      Core 20 (temp7_input)       45000
           8      Core 24 (temp8_input)       47000
           9      Core 28 (temp9_input)       46000
           10     Core 32 (temp10_input)      45000
           11     Core 33 (temp11_input)      45000
           12     Core 34 (temp12_input)      45000
           13     Core 35 (temp13_input)      45000
           14     Core 36 (temp14_input)      48000
           15     Core 37 (temp15_input)      48000
           16     Core 38 (temp16_input)      48000
           17     Core 39 (temp17_input)      48000

> nvme-pci-0400
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  42850
           2      Sensor 1 (temp2_input)   42850
           3      Sensor 2 (temp3_input)   37850
           4      Sensor 3 (temp4_input)   43850

> acpitz-acpi-0
  Sensors  Index  Label                 Value
           1      hwmon0 (temp1_input)  27800

> nct6798-isa-0
  Fans     Index  Label   RPM   PWM  Auto
           1      hwmon5  4500  198  true
           2      hwmon5  1687  198  true
           3      hwmon5  0     153  false
           4      hwmon5  0     153  false
           5      hwmon5  0     153  false
           7      hwmon5  0     255  false
  Sensors  Index  Label                                Value
           1      SYSTIN (temp1_input)                 39000
           2      CPUTIN (temp2_input)                 40500
           3      AUXTIN0 (temp3_input)                34500
           4      AUXTIN1 (temp4_input)                35000
           5      AUXTIN2 (temp5_input)                36000
           6      AUXTIN3 (temp6_input)                -38000
           7      PECI Agent 0 (temp7_input)           49500
           8      PCH_CHIP_CPU_MAX_TEMP (temp8_input)  0
           9      PCH_CHIP_TEMP (temp9_input)          0
           10     PCH_CPU_TEMP (temp10_input)          0

> nvme-pci-0300
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  45850
           2      Sensor 1 (temp2_input)   45850

> nvme-pci-0200
  Sensors  Index  Label                    Value
           1      Composite (temp1_input)  39850
           2      Sensor 1 (temp2_input)   39850
           3      Sensor 2 (temp3_input)   39850

**Desktop (please complete the following information):**

- Distro: [e.g. Arch Linux]
- `uname -a`: Linux pve 6.2.16-4-pve-relaxablermrr #1 SMP PREEMPT_DYNAMIC PVE 6.2.16-5 (2023-07-14T17:53Z) x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0 with libsensors version 3.6.0
- `fan2go version`: dev

**Additional context**
Add any other context about the problem here.

mks-h, 2023-09-10:
I didn't get it either, when installing by the first set of instructions in the "Manual" section. This is because it downloads the latest stable release, which doesn't feature the column in code. So, I had to compile it myself (the second set of instructions in the "Manual" section) — now it works.

markusressel, 2023-09-10:
@mks-h Thx for clearing that up, I thought the releases are currently not that relevant since the package will always compile the latest dev state, but I guess I was wrong. I will create a new release to remedy this asap.

markusressel, 2023-09-10:
New release is out, see: https://github.com/markusressel/fan2go/releases/tag/0.8.1

closed 2023-09-10 by markusressel

mks-h, 2023-09-10:
Great, thanks!