markusressel/fan2go #246: Help needed with sensor values

SteveSysadmin, 2023-08-03:
My GPU temperature sensor is configured in fan2go.yaml like this:
```
sensors:
  - id: gpu_sensor
    cmd:
      exec: /usr/bin/nvidia-smi
      args: [ "--query-gpu=temperature.gpu", "--format=csv,noheader,nounits" ]
```
and I can query the sensor like this:

`fan2go sensor --id gpu_sensor`

The value is in 2 digit format, eg 35. Is this an acceptable input value? I noticed in "fan2go detect" the values are all 5 digits.

Please tell me if this is ok or not.

SteveSysadmin, 2023-08-03:
So fan2go definitely wants 5-digit temperature sensor values.

I jury rigged it in fan2go/internal/sensors/cmd.go  line 41

`return (temp * 1000), nil`

recompiled and it works. Happy days!

markusressel, 2023-08-05:
Correct. See https://github.com/markusressel/fan2go#cmd-1

closed 2023-08-05 by markusressel