markusressel/fan2go #250: Panic runing config  validate

edt-xx, 2023-08-16:
Panic runing config  validate

Build using yay -S fan2go-git
run: fan2go config validate

[ed@grover fan2go]$ sudo fan2go config -v validate
  FATAL   Error reading config file, While parsing config: yaml: line 30: did not find expected '-' indicator
panic: 

goroutine 1 [running]:
github.com/pterm/pterm.checkFatal(...)
        github.com/pterm/pterm@v0.12.65/prefix_printer.go:351
github.com/pterm/pterm.(*PrefixPrinter).Printfln(0x55d6ad0eade0, {0x55d6acab1948, 0x1d}, {0xc0000f3c88, 0x1, 0x1})
        github.com/pterm/pterm@v0.12.65/prefix_printer.go:293 +0x1ed
github.com/markusressel/fan2go/internal/ui.Fatal({0x55d6acab1948, 0x1d}, {0xc0000f3c88, 0x1, 0x1})
        github.com/markusressel/fan2go/internal/ui/logging.go:52 +0x8b
github.com/markusressel/fan2go/internal/configuration.ReadInConfig()
        github.com/markusressel/fan2go/internal/configuration/config.go:111 +0x5e
github.com/markusressel/fan2go/internal/configuration.DetectAndReadConfigFile(...)
        github.com/markusressel/fan2go/internal/configuration/config.go:103
github.com/markusressel/fan2go/cmd/config.glob..func1(0x55d6ad0ee900?, {0xc00021a350?, 0x1?, 0x1?})
        github.com/markusressel/fan2go/cmd/config/validate.go:16 +0x1e
github.com/spf13/cobra.(*Command).execute(0x55d6ad0ee900, {0xc00021a330, 0x1, 0x1})
        github.com/spf13/cobra@v1.7.0/command.go:940 +0x862
github.com/spf13/cobra.(*Command).ExecuteC(0x55d6ad0ee060)
        github.com/spf13/cobra@v1.7.0/command.go:1068 +0x3bc
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.7.0/command.go:992
github.com/markusressel/fan2go/cmd.Execute()
        github.com/markusressel/fan2go/cmd/root.go:88 +0x115
main.main()
        ./main.go:25 +0x17
[ed@grover fan2go]$

**Expected behavior**
Errors in the config should be displayed correctly, there should not be a panic.

**Desktop (please complete the following information):**

- Distro: Arch Linux
- uname -a:  Linux grover 6.4.10-1-stable-git-00055-gdbf67f639df8-dirty #1 SMP PREEMPT_DYNAMIC Sun, 13 Aug 2023 - 12:53:09 +0000 x86_64 GNU/Linux
                      kernel built with the 6.5 hwmon pull to support nct6799-isa-0290
- sensors -v: sensors version 3.6.0+git with libsensors version 3.6.0+git
- fan2go version: dev (0.8.0.r151.g7c86ff0-1)

Add any other context about the problem here.

I've tried go-beta (same errors as above) an gcc-go (gets a compile error).  
My fan2go.yaml and output of detect included in setup.zit

[setup.zip](https://github.com/markusressel/fan2go/files/12363300/setup.zip)

edt-xx, 2023-08-17:
Figured out what is happening.   Looks like viper wants indents of 4 spaces, 2 will not work.  I might be a good idea to update the sample yaml to use 4 spaces.

markusressel, 2023-08-18:
Hey @edt-xx,

I am using 2 spaces in all of my fan2go configs on all of my devices (currently three) without issues, so you probably had a different issue with your configuration file.

As for the panic: Yep thats a bit ugly. Feel free to open a PR to clean this up :)

PR #256 (2023-09-10): less verbose CLI output
fixes #250
closed 2023-09-10 by markusressel