markusressel/fan2go #258: Confusing error message. Need help setting up.

jhillest, 2023-09-27:
**Describe the bug**
Thank you for creating fan2go!
It seems to be a great alternative to fancontrol. The configuration file in yaml-format is much easier to read.

I'm struggeling to set up fan2go.
I get the error message "must have one of index or rpmChannel, must be >= 1"
I'm confused as to what I'm doing wrong. Please advice.

**To Reproduce**
Run "sudo fan2go fan init --id cpu

**Expected behavior**
A non fatal result upon completion of the fan initialization sequence.

**Desktop (please complete the following information):**

- Distro: [e.g. Arch Linux]
- `uname -a`: Linux 6.5.3-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 13 Sep 2023 08:37:16 +0000 x86_64 GNU/Linux
- `sensors -v`: sensors version 3.6.0+git with libsensors version 3.6.0+git
- `fan2go version`: 0.8.1.r9.g048db61-1 (from the AUR)

**Additional context**
Here's my fan2go.yaml:

dbPath: "/etc/fan2go/fan2go.db"
runFanInitializationInParallel: false
maxRpmDiffForSettledFan: 20
fanResponseDelay: 2
tempSensorPollingRate: 200ms
tempRollingWindowSize: 10
rpmPollingRate: 1s
rpmRollingWindowSize: 10
controllerAdjustmentTickRate: 200ms
# Channel 1: Exhaust
# Channel 2: CPU
# Channel 3: Intake
fans:
  - id: cpu
    hwmon:
      platform: nct6791-isa-0290
      rpmChannel: 2
    neverStop: false
    curve: PECI_Agent
    minPwm: 32
    startPwm: 64
    maxPwm: 255
  - id: exhaust
    hwmon:
      platform: nct6791-isa-0290
      rpmChannel: 1
    neverStop: true
    curve: case_avg_curve
    minPwm: 100
    startPwm: 150
  - id: intake
    hwmon:
      platform: nct6791-isa-0290
      rpmChannel: 3
    neverStop: true
    curve: case_avg_curve
    minPwm: 60
    startPwm: 80

sensors:
  - id: PECI_Agent
    hwmon:
      platform: nct6791-isa-0290
      index: 1
  - id: CPU_socket
    hwmon:
      platform: nct6791-isa-0290
      index: 3

curves:
  - id: PECI_Agent
    linear:
      sensor: PECI_Agent
      min: 48
      max: 65
  - id: CPU_socket_curve
    linear:
      sensor: CPU_socket
      min: 40
      max: 80
  - id: case_avg_curve
    function:
      type: average
      curves:
        - PECI_Agent
        - CPU_socket_curve

statistics:
  enabled: false
  port: 9000
api:
  enabled: false
  host: localhost
  port: 9001
profiling:
  enabled: false
  host: localhost
  port: 6060

markusressel, 2023-09-27:
Probably a duplicate of #230
There seems to be an issue with running the `fan init` command. As a workaround try running the daemon instead, which will also initialize the fans on "first sight". If you can run the `validate` command without issues, the daemon will run fine.

markusressel, 2023-09-27:
Please try again with latest dev.

closed 2023-09-27 by markusressel

jhillest, 2023-09-29:
I can now run the `validate` command without issues.

PR #259 (2023-09-27): reset Configuration data before unmarshalling the config
fixes #230
fixes #258