markusressel/fan2go #261: Provide a GUI.

RokeJulianLockhart, 2023-10-01:
**Is your feature request related to a problem? Please describe.**
Modifying configuration files and invoking the command via the terminal is rarely as intuitive as performing similar actions with a GUI.

**Describe the solution you'd like**
Provide a GUI.

**Describe alternatives you've considered**
A TUI.

**Additional context**
I suggest wxWidgets, delegating to Qt. It necessitates the least manually written code for the most automatically generated native code for the most platforms (it delegates to Qt and WinUI, etcetera).

markusressel, 2023-10-01:
Hey @RokeJulianLockhart 🤓 

Ihx for the suggestion, I share your thought. However, I do want to keep fan2go itself as only the daemon that runs systemwide. The reson for this mainly time. I want to keep the scope of this package as small as possible. To still make it possible for other people to expand the functionality there already is a simple API included in fan2go. See: https://github.com/markusressel/fan2go#api

I have played myself with the idea of writing a TUI, but I haven't gotten around to it.

markusressel, 2023-10-07:
I have spent some time tinkering with a TUI for fan2go.
Feel free to check it out: https://github.com/markusressel/fan2go-tui

markusressel, 2023-10-09:
I will close this issue for now. If you have suggestions for how to improve fan2go-tui feel free to create a new issue in that repository. If you would like to see a GUI application, feel free to create one and - as stated before - I will be happy to extend the API if needed. I do not have the resources to create and maintain one myself. The TUI will probably also stay relatively basic in the foreseeable future.

closed 2023-10-09 by markusressel